kaloz/mwlwifi #300: Broadcast traffic fails for 5Ghz iPhone - HomeKit control lost

iancd, 2018-06-06:
Creating a new issue for problem symptoms previously described in issue #36. Following advice from @yuhhaurlin to test using arping I am now able to reproduce problem. Previously I rebooted the router to solve the problem. Please let me know if there are other troubleshoot steps I should take.
#### Hardware
* Link-sys WRT1200 with dual 2G/5G networks (identical SSIDs for roaming)
* Lede SNAPSHOT r6952
* IOT HomeKit devices on Ethernet and on 2.4G Network (multiple vendors)
* iPhone 6S (iOS 11.4). Problems found on multiple iPhones in my LAN.
Problem: iPhones randomly stops being able to control IOT devices connected via Ethernet and on 2.4G WIFI network.  Homekit IOT Devices will all show 'no response' in control centre on the iPhone. This problem happens at different times but normally 24+hours after router is restarted.
#### Troubleshooting steps:
In the example session below, the iPhone (statically assigned IP 192.168.5.100) is positioned 1m away from router and screen unlocked (not sleeping). iPhone is connected to 5G and has no problems routing to the Ethernet WAN.
1. Ping from Ethernet LAN computer > iPhone. Ping is ok
2. ARPing from Ethernet LAN computer > iPhone. Arping times out - no response
```
~ ping 192.168.5.100
PING 192.168.5.100 (192.168.5.100): 56 data bytes
64 bytes from 192.168.5.100: icmp_seq=0 ttl=64 time=3.225 ms
64 bytes from 192.168.5.100: icmp_seq=1 ttl=64 time=3.182 ms
^C
~ arping 192.168.5.100
ARPING 192.168.5.100
Timeout
Timeout
^C
```
3. Disable the 5G network which forces iPhone to roam to 2G network
4. Repeat Arping Ethernet LAN computer > iPhone. Arping is ok
iPhone roams to 2G **HomeKit functionality** is restored like magic. Arping succeeds

```
# arping 192.168.5.100
ARPING 192.168.5.100
60 bytes from <MAC ADDRESS REDACTED> (192.168.5.100): index=0 time=147.961 msec
60 bytes from <MAC ADDRESS REDACTED> (192.168.5.100): index=1 time=171.399 msec
^C
```

#### /etc/wireless/config
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
	option country 'GB'
	option txpower '17'
	option htmode 'VHT40'
	option channel '40'
	option legacy_rates '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option macaddr '<REDACTED>'
	option ssid '<REDACTED>'
	option wps_pushbutton '0'
	option key '<REDACTED>'
	option encryption 'psk2+ccmp'
	option wpa_disable_eapol_key_retries '1'
	option disassoc_low_ack '0'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'GB'
	option log_level '3'
	option channel '2'
	option legacy_rates '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option macaddr '<REDACTED>'
	option wps_pushbutton '0'
	option ssid '<REDACTED>'
	option encryption 'psk2+ccmp'
	option key '<REDACTED>'
	option wpa_disable_eapol_key_retries '1'
	option disassoc_low_ack '0'
```
#### Version
```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20180330
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: <REDACTED>
2g: disable
5g: enable
antenna: 2 2
irq number: 48
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1100000
iobase1: e1280000
tx limit: 768
rx limit: 64
qe trigger number: 21875530
```

yuhhaurlin, 2018-06-06:
How about arping to other devices?

iancd, 2018-06-06:
@yuhhaurlin From my Ethernet LAN test computer I tested arping to other devices both ethernet and the IOT devices on the 2.4G network and they were fine. Only the iPhone client timed out with arping - at the time of testing it was the only device on the 5G network, although during times I have observed the same HomeKit symptoms on multiple iPhones simultaneously.

yuhhaurlin, 2018-06-06:
Can you connect other devices to 5g and check arpping to these devices?

iancd, 2018-06-06:
@yuhhaurlin Yeah I can do that next time it fails and report back. I'll test bother another identical iPhone and a non 80211ac 5G laptop. When HomeKit fails the symptoms show in the same way for all iOS devices but good idea to test.

bbogey, 2018-06-06:
I tested to ping ipad air 2
        
```
$ ping 192.168.1.221
PING 192.168.1.221 (192.168.1.221) 56(84) bytes of data.
64 bytes from 192.168.1.221: icmp_seq=1 ttl=64 time=150 ms
64 bytes from 192.168.1.221: icmp_seq=2 ttl=64 time=172 ms
64 bytes from 192.168.1.221: icmp_seq=3 ttl=64 time=195 ms
64 bytes from 192.168.1.221: icmp_seq=4 ttl=64 time=115 ms
64 bytes from 192.168.1.221: icmp_seq=5 ttl=64 time=138 ms
64 bytes from 192.168.1.221: icmp_seq=6 ttl=64 time=161 ms
64 bytes from 192.168.1.221: icmp_seq=7 ttl=64 time=184 ms
64 bytes from 192.168.1.221: icmp_seq=8 ttl=64 time=207 ms
64 bytes from 192.168.1.221: icmp_seq=9 ttl=64 time=2.44 ms
64 bytes from 192.168.1.221: icmp_seq=10 ttl=64 time=149 ms
64 bytes from 192.168.1.221: icmp_seq=11 ttl=64 time=173 ms
64 bytes from 192.168.1.221: icmp_seq=12 ttl=64 time=196 ms
^C
--- 192.168.1.221 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 11013ms
rtt min/avg/max/mdev = 2.448/153.922/207.083/52.234 ms

$ arping -I p2p1 192.168.1.221
ARPING 192.168.1.221 from 192.168.1.33 p2p1
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  3.699ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  119.709ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  3.260ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  167.490ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  192.074ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  113.651ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  137.192ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  161.016ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  3.282ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  209.267ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  130.760ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  154.458ms
Unicast reply from 192.168.1.221 [24:A0:74:4A:1F:E4]  178.591ms
^CSent 15 probes (3 broadcast(s))
Received 13 response(s)

root@lede:~# iwinfo wlan0-2 assoclist
24:A0:74:4A:1F:E4  -66 dBm / -90 dBm (SNR 24)  35570 ms ago
        RX: 24.0 MBit/s                                 1717 Pkts.
        TX: 702.0 MBit/s, VHT-MCS 8, 80MHz, VHT-NSS 2       698 Pkts.
        expected throughput: unknown

root@lede:~# iw dev wlan0-2 station dump
Station 24:a0:74:4a:1f:e4 (on wlan0-2)
        inactive time:  830 ms
        rx bytes:       198284
        rx packets:     2089
        tx bytes:       294642
        tx packets:     759
        tx retries:     0
        tx failed:      0
        rx drop misc:   0
        signal:         -63 dBm
        signal avg:     -60 dBm
        tx bitrate:     702.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 2
        rx bitrate:     24.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 14135 seconds
root@lede:~# iwinfo wlan0-2 assoclist
24:A0:74:4A:1F:E4  -64 dBm / -89 dBm (SNR 25)  880 ms ago
        RX: 24.0 MBit/s                                 2112 Pkts.
        TX: 702.0 MBit/s, VHT-MCS 8, 80MHz, VHT-NSS 2       759 Pkts.
        expected throughput: unknown

root@lede:~#
```

iancd, 2018-06-06:
I have just lost homekit control again. This time I added Macbook to the 5G network and retested Arping to iPhone. Macbook arpings/pings OK and iPhone homekit control is not working.

```
ARPING 192.168.5.122
60 bytes from <REDACTED> (192.168.5.122): index=0 time=137.330 msec
60 bytes from <REDACTED> (192.168.5.122): index=1 time=160.232 msec

ARPING 192.168.5.100
Timeout
Timeout
```

```
iwinfo wlan0 assoclist
Macbook 192.168.5.122  -56 dBm / -92 dBm (SNR 36)  9080 ms ago
	RX: 48.0 MBit/s                                 5763 Pkts.
	TX: 300.0 MBit/s, MCS 15, 40MHz                 8039 Pkts.
	expected throughput: unknown

iphone 192.168.5.100  -47 dBm / -92 dBm (SNR 45)  34390 ms ago
	RX: 400.0 MBit/s, VHT-MCS 9, 40MHz, VHT-NSS 2       132 Pkts.
	TX: 400.0 MBit/s, VHT-MCS 9, 40MHz, VHT-NSS 2        90 Pkts.
	expected throughput: unknown
```
Errors in logs..
```
Wed Jun  6 15:44:28 2018 daemon.info hostapd: wlan0: STA <REDACTED> IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jun  6 15:44:44 2018 user.warn igmpproxy[5291]: MRT_ADD_MEMBERSHIP failed; Errno(19): No such device
```

iancd, 2018-06-13:
Closing this after more testing I no longer think is related to mwlwifi.  I was able to reproduce the problem with an aberrant HomeKit device which somehow stops the iPhones from returning arp who-has packets.

closed 2018-06-13 by iancd