kaloz/mwlwifi #302: MCS Values 8 thru 15 no longer working on 10.3.8.0-20180614 88W8864

c0manche, 2018-06-15:
Just tried the latest driver 10.3.8.0-20180614 on Lede I have noticed MCS 8 thru 15 are no longer available when I go back to an older driver they are working.

Older driver iw phy0 info reports

HT TX/RX MCS rate indexes supported: 0-23, 32

New driver reports

HT TX/RX MCS rate indexes supported: 0-7, 16-23, 32

BrainSlayer, 2018-06-15:
i expect this will fix it (@yuhhaurlin , please check this)

--- core.c      (Revision 4105)
+++ core.c      (Arbeitskopie)
@@ -442,7 +442,7 @@
        band->ht_cap.ampdu_density = IEEE80211_HT_MPDU_DENSITY_4;

        band->ht_cap.mcs.rx_mask[0] = 0xff;
-       if (priv->antenna_rx == ANTENNA_RX_2)
+       if (priv->antenna_rx != ANTENNA_RX_1)
                band->ht_cap.mcs.rx_mask[1] = 0xff;
        if (priv->antenna_rx == ANTENNA_RX_4_AUTO)
                band->ht_cap.mcs.rx_mask[2] = 0xff;

yuhhaurlin, 2018-06-15:
@BrainSlayer Thanks for your help. Our QA also finds this issue, I will check and fix it.

BrainSlayer, 2018-06-15:
@yuhhaurlin 
just checked wrt1900ac (the old v1 one)
this driver is completelly broken. 
    4.132999] ieee80211 phy0: priv->pcmd_buf = ceb9c000  priv->pphys_cmd_buf = 0eb9c000
[    4.176965] ieee80211 phy0: fw download start
[    4.294691] ieee80211 phy0: FwSize = 118788 downloaded Size = 118788 curr_iteration 65508
[    5.142367] ieee80211 phy0: fw download complete
[    5.142409] ieee80211 phy0: device node: /soc/pcie-controller/pcie@2,0
[    5.149728] ieee80211 phy0: pcmd = ceb9c000
[    5.167428] ieee80211 phy0: firmware version: 0x702091b
[    5.191110] ieee80211 phy0: firmware region code: 0
[    5.218593] ------------[ cut here ]------------
[    5.223336] WARNING: CPU: 1 PID: 969 at /home/seg/DEV/mvebu/src/router/private/compat-wireless-2017-09-03/net/mac80211/../wireless/core.c:612 _1296+0x420/0x8b8 [mac80211]
[    5.238626] Modules linked in: mwlwifi(+) mac80211 compat lzma_decompress(P) lzma_compress(P) mii tmp421 pwm_fan leds_pca963x leds_tlc591xx orion_wdt
[    5.252242] CPU: 1 PID: 969 Comm: insmod Tainted: P                4.9.108 #158
[    5.259580] Hardware name: Marvell Armada 370/XP (Device Tree)
[    5.265464] [<c010ee38>] (unwind_backtrace) from [<c010ac5c>] (show_stack+0x10/0x14)
[    5.273256] [<c010ac5c>] (show_stack) from [<c02b5670>] (dump_stack+0x88/0x9c)
[    5.280507] [<c02b5670>] (dump_stack) from [<c0122d34>] (__warn+0xe4/0x100)
[    5.287505] [<c0122d34>] (__warn) from [<c0122e00>] (warn_slowpath_null+0x20/0x28)
[    5.295169] [<c0122e00>] (warn_slowpath_null) from [<bf073580>] (_1296+0x420/0x8b8 [mac80211])
[    5.303885] [<bf073580>] (_1296 [mac80211]) from [<bf02d060>] (_670+0x75c/0xa5c [mac80211])
[    5.312329] [<bf02d060>] (_670 [mac80211]) from [<bf0cc45c>] (_305+0xa4c/0xcc4 [mwlwifi])
[    5.320575] [<bf0cc45c>] (_305 [mwlwifi]) from [<bf0d6c44>] (_360+0x294/0x2e8 [mwlwifi])
[    5.328730] [<bf0d6c44>] (_360 [mwlwifi]) from [<c02fc1bc>] (pci_device_probe+0x74/0xc8)
[    5.336867] [<c02fc1bc>] (pci_device_probe) from [<c033bac8>] (really_probe+0xfc/0x25c)
[    5.344914] [<c033bac8>] (really_probe) from [<c033bcb4>] (__driver_attach+0x8c/0xb0)
[    5.352791] [<c033bcb4>] (__driver_attach) from [<c033a0ec>] (bus_for_each_dev+0x4c/0xa0)
[    5.360999] [<c033a0ec>] (bus_for_each_dev) from [<c033b194>] (bus_add_driver+0xe8/0x200)
[    5.369217] [<c033b194>] (bus_add_driver) from [<c033c25c>] (driver_register+0xb8/0xf4)
[    5.377265] [<c033c25c>] (driver_register) from [<c0101830>] (do_one_initcall+0xb8/0x164)
[    5.385488] [<c0101830>] (do_one_initcall) from [<c01983bc>] (do_init_module+0x60/0x1c4)
[    5.393623] [<c01983bc>] (do_init_module) from [<c0187bac>] (load_module+0x1c28/0x1f14)
[    5.401655] [<c0187bac>] (load_module) from [<c0187fc4>] (SyS_init_module+0x12c/0x160)
[    5.409612] [<c0187fc4>] (SyS_init_module) from [<c0107600>] (ret_fast_syscall+0x0/0x48)
[    5.417834] ---[ end trace c2499d4053cb8a77 ]---
[    5.422511] ieee80211 phy0: fail to register device
[    5.427420] ieee80211 phy0: unmapped+free'd 0 0xce968080 0xe968080 4096

yuhhaurlin, 2018-06-15:
Our QA has tested this version of code for 88W8864, 88W8964 and 88W8997. They only found throughput issue with HT rate. The version of backports used by  updated OpenWrt repository is 2017-11-01, I can't check what the warning is? Can you show me the code of line 612 of core.c?

BrainSlayer, 2018-06-15:
the warning is within wiphy_register. there seem to be a problem while if_comb checks. maybe related to the new mesh interface type. 
since i use the git version from mwlwifi, the line 612 is "}"
so i dont know with which code you're working, but its clearly not the one here posted in github. the line 612 request makes no real sense

BrainSlayer, 2018-06-15:
removing the mesh interface type from if_comb / ap_if_limits fixes the issue.

BrainSlayer, 2018-06-15:
this fixes the issue for older chipsets. the other fix i posted. fixes the mcs rate issue

--- core.c      (Revision 4106)
+++ core.c      (Arbeitskopie)
@@ -98,12 +98,31 @@
        { .bitrate = 540, .hw_value = 108, },
 };

-static const struct ieee80211_iface_limit ap_if_limits[] = {
+static const struct ieee80211_iface_limit ap_if_limits_kf2[] = {
        { .max = SYSADPT_NUM_OF_AP, .types = BIT(NL80211_IFTYPE_AP) },
        { .max = SYSADPT_NUM_OF_MESH, .types = BIT(NL80211_IFTYPE_MESH_POINT) },
        { .max = SYSADPT_NUM_OF_CLIENT, .types = BIT(NL80211_IFTYPE_STATION) },
 };

+static const struct ieee80211_iface_combination ap_if_comb_kf2 = {
+       .limits = ap_if_limits_kf2,
+       .n_limits = ARRAY_SIZE(ap_if_limits_kf2),
+       .max_interfaces = SYSADPT_NUM_OF_AP,
+       .num_different_channels = 1,
+       .radar_detect_widths =  BIT(NL80211_CHAN_WIDTH_20_NOHT) |
+                               BIT(NL80211_CHAN_WIDTH_20) |
+                               BIT(NL80211_CHAN_WIDTH_40) |
+                               BIT(NL80211_CHAN_WIDTH_80) |
+                               BIT(NL80211_CHAN_WIDTH_160),
+};
+
+
+
+static const struct ieee80211_iface_limit ap_if_limits[] = {
+       { .max = SYSADPT_NUM_OF_AP, .types = BIT(NL80211_IFTYPE_AP) },
+       { .max = SYSADPT_NUM_OF_CLIENT, .types = BIT(NL80211_IFTYPE_STATION) },
+};
+
 static const struct ieee80211_iface_combination ap_if_comb = {
        .limits = ap_if_limits,
        .n_limits = ARRAY_SIZE(ap_if_limits),
@@ -828,9 +847,14 @@

        hw->wiphy->interface_modes = 0;
        hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_AP);
-       hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_MESH_POINT);
        hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_STATION);
-       hw->wiphy->iface_combinations = &ap_if_comb;
+
+       if (priv->chip_type == MWL8997) {
+               hw->wiphy->interface_modes |= BIT(NL80211_IFTYPE_MESH_POINT);
+               hw->wiphy->iface_combinations = &ap_if_comb_kf2;
+       } else {
+               hw->wiphy->iface_combinations = &ap_if_comb;
+       }
        hw->wiphy->n_iface_combinations = 1;

BrainSlayer, 2018-06-15:
ah wait. the answer is more simple

#ifdef CONFIG_MAC80211_MESH is the solution. you need to exclude the interface type from the ap_if_limits array in case this option is not set. otherwise the driver will not work on configuration without having mesh enabled

BrainSlayer, 2018-06-15:
--- core.c      (Revision 4106)
+++ core.c      (Arbeitskopie)
@@ -100,7 +100,9 @@

 static const struct ieee80211_iface_limit ap_if_limits[] = {
        { .max = SYSADPT_NUM_OF_AP, .types = BIT(NL80211_IFTYPE_AP) },
+#ifdef CONFIG_MAC80211_MESH
        { .max = SYSADPT_NUM_OF_MESH, .types = BIT(NL80211_IFTYPE_MESH_POINT) },
+#endif
        { .max = SYSADPT_NUM_OF_CLIENT, .types = BIT(NL80211_IFTYPE_STATION) },
 };

yuhhaurlin, 2018-06-15:
1. You should define mesh for backports package.
2. Mcs rate is fixed, I will update here later.

yuhhaurlin, 2018-06-15:
Fixed by 10.3.8.0-20180615.

closed 2018-06-15 by yuhhaurlin

danielpetre, 2018-06-28:
Hello, i am not sure the driver is fixed yet, 10.3.8.0 still crashes on latest openwrt compiled on 27 june while 10.3.4.0 still works.

yuhhaurlin, 2018-06-28:
@danielpetre What crash problem you encountered?

danielpetre, 2018-06-28:
Same wireless/core.c:612 crash, unfortunately did not saved the log and now i am on previous driver version which works okay.

yuhhaurlin, 2018-06-28:
Please give me log. Is it due to mesh is not configured?

danielpetre, 2018-06-28:
I do not use mesh, don't have the mesh option compiled.

yuhhaurlin, 2018-06-28:
Please check https://github.com/kaloz/mwlwifi/commit/c2f82f1ac18ed7fb48be1eb8d678a3cf32192069.