kaloz/mwlwifi #305: iwinfo scan returning bad signal levels

lantis1008, 2018-07-01:
Hi,

Is `iwinfo wlan1 scan` supposed to work or has it not been supported?
Signal level is always  reported as -256dBm

```
root@Gargoyle:~# iwinfo wlan1 scan
Cell 01 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 1
          Signal: -256 dBm  Quality: 70/70
          Encryption: mixed WPA/WPA2 PSK (TKIP, CCMP)

Cell 02 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 1
          Signal: -256 dBm  Quality: 70/70
          Encryption: none

Cell 03 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 1
          Signal: -256 dBm  Quality: 70/70
          Encryption: none

Cell 04 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 1
          Signal: -256 dBm  Quality: 70/70
          Encryption: WPA2 PSK (CCMP)

Cell 05 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 3
          Signal: -256 dBm  Quality: 70/70
          Encryption: none

Cell 06 - Address: xx:xx:xx:xx:xx:xx
          ESSID: "xxx"
          Mode: Master  Channel: 6
          Signal: -256 dBm  Quality: 70/70
          Encryption: none
```

yuhhaurlin, 2018-07-02:
This is WRT3200ACM?

lantis1008, 2018-07-02:
No WRT32X
https://www.linksys.com/us/p/P-WRT32X/

Should be very similar hardware to WRT3200ACM

yuhhaurlin, 2018-07-02:
Please use firmware for WRT32X.

yuhhaurlin, 2018-07-02:
You can report problem related to WRT3200ACM here. Thanks.

lantis1008, 2018-07-02:
I am using WRT32X firmware. It is officially supported by Openwrt now. You will not support it or investigate issue?

yuhhaurlin, 2018-07-02:
As I know, it uses proprietary driver.

yuhhaurlin, 2018-07-02:
So you mean it can use mwlwifi now?

lantis1008, 2018-07-02:
Yes it uses mwlwifi. The hardware is almost exactly the same as WRT3200ACM, but there are some minor differences. 
I wonder if these differences are causing the issues, and had hoped you might be able to investigate?

Are you able to get one of these devices for testing internally?

yuhhaurlin, 2018-07-02:
Due to it dose not use mwlwifi, I will not test it.

lantis1008, 2018-07-02:
It DOES use mwlwifi

yuhhaurlin, 2018-07-02:
WRT32X should use proprietary driver.

bbogey, 2018-07-02:
It uses mwlwifi (88W8964)

https://wikidevi.com/wiki/Linksys_WRT32X

yuhhaurlin, 2018-07-02:
I will check it later.

yuhhaurlin, 2018-07-02:
Yes. The merged OpenWrt repository (I got it on 20180601) can build image for WRT32X. I will check it later.

bbogey, 2018-07-02:
Yes it pretty recent https://github.com/openwrt/openwrt/commit/bfbdeeb3de3da31f7e5f9bd429e079c2d839644a

@lantis1008 you are using gargoyle?

lantis1008, 2018-07-02:
@bbogey yes. For all intents and purposes it should be identical for this issue, but will happily test Openwrt to check. 
I had intended to do so this evening (although yesterday I finally put this router into full time production as my gateway. Might have been a bit premature).
I’m seeing hostapd issues as well which may be related but I am not with the device currently.

BrainSlayer, 2018-07-02:
the WRT32X is identical to the WRT3200ACM. just enclosure is different. no hw difference. interesting point in the posted informations.
power table loaded from dts: yes
firmware region code: 0x0

powertable loading from dts is not supported for 8964. so it seems there is a problem with the powertable data on that chipset ans the region code doesnt look correct as well

yuhhaurlin, 2018-07-02:
@BrainSlayer Yes. That is what I suspect, I will get WRT32X to check if it has the same problem.

lantis1008, 2018-07-02:
For reference, my device is an AU region code.

BrainSlayer, 2018-07-02:
@yuhhaurlin : i get one soon as well. so i can check if i have the same behaviour on that device

yuhhaurlin, 2018-07-02:
@BrainSlayer Thanks.

eduperez, 2018-07-02:
On a WRT3200ACM, running LEDE 17.01.4 plus mwlwifi c2f82f1ac18ed7fb48be1eb8d678a3cf32192069, I see the same behaviour: running "`iw wlan_2g scan`" or "`iw wlan_5g scan`", both report "`signal: 0.00 dBm`" for all stations found.

lantis1008, 2018-07-02:
@eduperez what about 
`cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info`
?

Is it reporting info ok?

eduperez, 2018-07-02:
```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180615
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:bc:34:e2
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000007
radio: enable
iobase0: e0f00000
iobase1: e1180000
tx limit: 1024
rx limit: 16384
```

lantis1008, 2018-07-02:
Ok so that issue may only be WRT32X, but iw scan may be others. 
Thanks for your input

BrainSlayer, 2018-07-02:
i can reproduce it with dd-wrt as well using the iw utility. so its not a WRT32X specific issue

ValCher1961, 2018-07-02:
@lantis1008.
you applied a patch - regfree?

lantis1008, 2018-07-02:
@ValCher1961 
No this one is clean. The WRT32X is sold in my country so I have no need to use that patch on this device yet.

ValCher1961, 2018-07-02:
Means in EEPROM there are no regional definitions, it's interesting.

lantis1008, 2018-07-05:
@yuhhaurlin now that it has been reproduced on WRT3200ACM, does this mean you are able to investigate sooner?
I understand you are busy with mu-mimo

yuhhaurlin, 2018-07-05:
It the issue is major problem (system crash), otherwise I will check it later. Thanks.

lantis1008, 2018-08-01:
Any update to this one please?

lantis1008, 2018-08-15:
@BrainSlayer I heard you recently received one of these. Did you check the same behaviour for power table load reporting?

BrainSlayer, 2018-08-15:
i'll check tomorrow if i can see this issue on it (with dd-wrt)

BrainSlayer, 2018-08-16:
@lantis1008 , @yuhhaurlin i can reproduce the issue on the WRT32X. according to mwlwifi the powertable isnt loaded from dts but the mwldriver returns bogus signal values. they are always 0. i also can't see any reason why

[    3.295052] ieee80211 phy0: priv->pcmd_buf = de4fc000  priv->pphys_cmd_buf = 1e4fc000
[    3.349402] ieee80211 phy0: fw download start
[    3.349407] ieee80211 phy0: ddr init: download start
[    3.349784] ieee80211 phy0: ddr init: download complete
[    3.487441] ieee80211 phy0: FwSize = 450732 downloaded Size = 450732 curr_iteration 65521
[    6.357283] ieee80211 phy0: fw download complete
[    6.357316] ieee80211 phy0: device node: /soc/pcie-controller/pcie@1,0
[    6.373677] ieee80211 phy0: pcmd = de4fc000
[    6.390302] ieee80211 phy0: firmware version: 0x9030206
[    6.750289] ieee80211 phy0: firmware region code: 10
[    6.782009] ieee80211 phy0: 2G disabled, 5G enabled
[    6.786917] ieee80211 phy0: 4 TX antennas, 4 RX antennas
[    6.792352] pci 0000:00:02.0: enabling device (0140 -> 0142)
[    6.798133] ieee80211 phy1: priv->pcmd_buf = da364000  priv->pphys_cmd_buf = 1a364000
[    6.804748] ieee80211 phy1: fw download start
[    6.804754] ieee80211 phy1: ddr init: download start
[    6.805135] ieee80211 phy1: ddr init: download complete
[    6.945831] ieee80211 phy1: FwSize = 450732 downloaded Size = 450732 curr_iteration 65526
[    8.141280] ieee80211 phy1: fw download complete
[    8.141304] ieee80211 phy1: device node: /soc/pcie-controller/pcie@2,0
[    8.157614] ieee80211 phy1: pcmd = da364000
[    8.175309] ieee80211 phy1: firmware version: 0x9030206
[    8.536283] ieee80211 phy1: firmware region code: 10
[    8.559994] ieee80211 phy1: 2G enabled, 5G disabled
[    8.564919] ieee80211 phy1: 4 TX antennas, 4 RX antennas

root@DD-WRT:/sys/kernel/debug/ieee80211/phy0/mwlwifi# cat info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180716
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:c6:7a:5a
2g: disable
5g: enable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2018-08-16:
@BrainSlayer Thanks for your information, I will check it later.

BrainSlayer, 2018-08-16:
@yuhhaurlin i will check some ideas for the cause in between and will post if i find something

lantis1008, 2018-08-16:
It is interesting that yours shows not loaded from DTS whereas mine does. 
And that your region code works Ok but mine does not.

BrainSlayer, 2018-08-16:
@lantis1008 is your dts modified or plain openwrt?

lantis1008, 2018-08-16:
Should just be plain. There’s a possibility that I corrupted one of the MTD partitions during my dev work. But my understanding is that these types of settings are hidden away including an EEPROM?

lantis1008, 2018-08-16:
Do you think Openwrt DTS is not right?

BrainSlayer, 2018-08-16:
@lantis1008 its stored in eeprom and not mtd. correct. (hopefully, not that linksys really did something special here) about the dts. i know the openwrt dts, i just was asking if you modified it or if its plain vanilla from current openwrt tree

lantis1008, 2018-08-16:
No unmodified. I tried to compare openwrt dts against ddwrt but syntax is slightly different. i think is ok though.

BrainSlayer, 2018-08-16:
@yuhhaurlin i found a different in the preprogrammed powertables. wrt3200acm is programmed for the possible channels by chipset. the wrt32x is always preprogrammed for 2.4 and 5 ghz on both chipsets. this is a example here. and to be honest. the preprogrammed powertable looks bogus and invalid
power table loaded from dts: no
firmware region code: 0x10
number of channel: 39
  1 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  2 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  3 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  4 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  5 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  6 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  7 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  8 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  9 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 10 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 11 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 12 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 13 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 14 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 36 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 40 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 44 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 48 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 52 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 56 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 60 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 64 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
100 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
104 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
108 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
112 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
116 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
120 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
124 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
128 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
132 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
136 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
140 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
144 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
149 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
153 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
157 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
161 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
165 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0

BrainSlayer, 2018-08-16:
@lantis1008 : dd-wrt has slight modifications which is architecture specific like partition layout and leds etc. but the main parts should be identical

BrainSlayer, 2018-08-16:
.dts overloading of powertables is maybe possible .the data we need is included in the linksys firmware i extracted, but mwlwifi does not support pwltables from dts for that specific chipset. the code for dts loading is only made for older chipsets which do have less parameters. so mwlwifi needs to be enhanced for supporting loading of 8964 based parameters

lantis1008, 2018-08-16:
```
[    8.899933] ieee80211 phy0: priv->pcmd_buf = dde2c000  priv->pphys_cmd_buf = 1de2c000
[    8.979160] ieee80211 phy0: fw download start
[    8.979167] ieee80211 phy0: ddr init: download start
[    8.979545] ieee80211 phy0: ddr init: download complete
[    9.123221] ieee80211 phy0: FwSize = 450732 downloaded Size = 450732 curr_iteration 65521
[   11.985361] ieee80211 phy0: fw download complete
[   11.985425] ieee80211 phy0: device node: /soc/pcie/pcie@1,0
[   12.001277] ieee80211 phy0: pcmd = dde2c000
[   12.018205] ieee80211 phy0: firmware version: 0x9030206
[   12.041624] ieee80211 phy0: firmware region code: 0
[   12.069774] ieee80211 phy0: 2G disabled, 5G enabled
[   12.074684] ieee80211 phy0: 4 TX antennas, 4 RX antennas
[   12.080176] pci 0000:00:02.0: enabling device (0140 -> 0142)
[   12.085999] ieee80211 phy1: priv->pcmd_buf = de234000  priv->pphys_cmd_buf = 1e234000
[   12.092967] ieee80211 phy1: fw download start
[   12.092972] ieee80211 phy1: ddr init: download start
[   12.093356] ieee80211 phy1: ddr init: download complete
[   12.235797] ieee80211 phy1: FwSize = 450732 downloaded Size = 450732 curr_iteration 65527
[   13.429132] ieee80211 phy1: fw download complete
[   13.429161] ieee80211 phy1: device node: /soc/pcie/pcie@2,0
[   13.444314] ieee80211 phy1: pcmd = de234000
[   13.462356] ieee80211 phy1: firmware version: 0x9030206
[   13.485145] ieee80211 phy1: firmware region code: 0
[   13.508674] ieee80211 phy1: 2G enabled, 5G disabled
[   13.513590] ieee80211 phy1: 4 TX antennas, 4 RX antennas
```

Mine seems to fail to find the region code and any channels.
I wonder why the wifi even works at all for me??

yuhhaurlin, 2018-08-16:
For 8964, there should be no power table in DTS.

lantis1008, 2018-08-16:
This seems like serious issue. Are you able to prioritise it sooner than “later”?

yuhhaurlin, 2018-08-16:
Why this is serious?

lantis1008, 2018-08-16:
If device is potentially not loading power table correctly, maybe it is not behaving as it should?
I don’t know, I’m just trying to ask you to look at it with a bit of priority. I get that you’re busy, but you seem really disengaged on this particular issue. Like it does not matter?

yuhhaurlin, 2018-08-17:
8964 doesn’t need to load power table.

BrainSlayer, 2018-08-17:
@yuhhaurlin: this is serious since this makes the device unoperable to roam in client mode since all signal levels are bad. noise is zero for most channels too in any way. consider that this affects ap mode too since hostapd has the feature to kick clients with bad signal levels and since noise is zero too for most channels this will kick out the clients as well

yuhhaurlin, 2018-08-17:
@BrainSlayer Thanks for your information. I will check and fix it.

BrainSlayer, 2018-08-17:
@yuhhaurlin btw. the original wrt32x from linksys does overload the power tables on 8964 for CN, AU and AH regions. so your comment is simply wrong. see that code snipped from original firmware

        case "$CERT" in
                CN)
                        if [ -f /etc/config/wireless ]; then
                                uci set wireless.@wifi-device[0].regioncode=0x91
                                uci set wireless.@wifi-device[1].regioncode=0x91
                                uci commit wireless
                        fi
                        iwpriv wdev0 setcmd "loadpwrgrpstbl /etc/mrvl/145.ini"
                        iwpriv wdev1 setcmd "loadpwrgrpstbl /etc/mrvl/145.ini"
                ;;
                AU)
                        if [ -f /etc/config/wireless ]; then
                                uci set wireless.@wifi-device[0].regioncode=0x81
                                uci set wireless.@wifi-device[1].regioncode=0x81
                                uci commit wireless
                        fi
                        iwpriv wdev0 setcmd "loadpwrgrpstbl /etc/mrvl/129.ini"
                        iwpriv wdev1 setcmd "loadpwrgrpstbl /etc/mrvl/129.ini"
                ;;
                AH)
                        if [ -f /etc/config/wireless ]; then
                                uci set wireless.@wifi-device[0].regioncode=0x90
                                uci set wireless.@wifi-device[1].regioncode=0x90
                                uci commit wireless
                        fi
                        iwpriv wdev0 setcmd "loadpwrgrpstbl /etc/mrvl/144.ini"
                        iwpriv wdev1 setcmd "loadpwrgrpstbl /etc/mrvl/144.ini"
                ;;
        esac

BrainSlayer, 2018-08-17:
but that doesnt affect US of course and the current bug is visible on US devices so far. i havent got my hands on a EU model yet for checking

yuhhaurlin, 2018-08-17:
I think it is related to rssi of received packets. It is not related to power table. However, I will check it.

lantis1008, 2018-08-17:
Mine is AU model @brainslayer. 
Are you saying that I should adjust DTS to provide a power table for the above listed regions?
Of course, driver modification also required to go along with it. 

Would it need to be new format table like you list above or one of th existing tables from earlier device (like 1900AC) would do? Or that depends entirely on what yuhhaulin decides to do for the solution.

BrainSlayer, 2018-08-17:
i agree, but this shouldnt affect the noise level which has a problem too.  but hard to say whats wrong here

yuhhaurlin, 2018-08-17:
For current version of firmware of 8964, it won't allow you to download power table if power table is programmed in EEPROM. This issue should be related to rssi of received packets. If rssi of client is all right, I think I need to trace iwinfo to see what happened.

[21 more replies]

closed 2018-08-24 by yuhhaurlin