kaloz/mwlwifi #306: kmod-mwlwifi 06-15-2018 Slow 5G wifi performance vs mwlwifi dated 03-30-2018

alirz1, 2018-07-02:
I been compiling my own custom builds for my Wrt1900ac Mamba for a while. Recently switched to LEDE from the older CC branch of Openwrt.
Uptil last month i was running LEDE build r6874 that had mwlwifi driver dated 03-30-2018. I had decent 5G wifi speeds on that build. I would average at around 30-35MB/sec.
Recently upgraded to a newer build (r7290) few days ago and i notice that wifi was slow. Did quite a of bit of testing and at best 5g wifi speed was 12MB/sec. I then tested today's build of r7360, same slow wifi.
I then downgraded back to r6874 and wifi speed was fast again.
The only thing i see that the mwlwifi driver and the kmod-mwlwifi module was changed on Jun 15th.

I also notice that in r7290 and r7360 kernel debug lugging seems to be turned ON bases on what I see in the system log? Could that be the cause of the slow performance?

yuhhaurlin, 2018-07-02:
Can you show me the test result of iperf? This version of driver has been tested by our QA for all supported chips.

alirz1, 2018-07-02:
As I mentioned, I think it's not the driver really. Because I put the mwlwifi driver bin file from March on my router running the June build and the wifi speed was still slow. I think it could be the know mwlwifi module? But I don't know.
Does iperf need to be installed on the router?

yuhhaurlin, 2018-07-02:
You can build in iperf via “make menuconfig”. For client, you can google to know how to install it.

alirz1, 2018-07-03:
Ok i tested both builds and ran iperf3 on the router. The client device for this was my samsung s8+ connected to the 5G wifi of the router. If i'm reading the results right, iperf does NOT seem to show any significant speed issue. However i can repeatedly reproduce my issue from two devices (Dell XPS laptop and an Android 8 device, samsung s8.) in real life tests

For my "real life" speed test, i connect ONLY 1 device at a time to the 5G network of the router and transfer download a file over WEBDAV from my NAS. In both the cases, laptop or the phone, the file transfer speed is almost 3 times slower on the new LEDE build.
Note that both builds are exactly the same in regards to what packages are included etc. The only difference i see in them is the new mwlwifi stuff in the june build.
Thank you

Anyways, here are iperf result.

New Build 06-28-2018 (r7360)

Connecting to host 10.0.0.1, port 5201
[  8] local 10.0.0.3 port 59288 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  8]   0.00-1.07   sec  30.0 MBytes   236 Mbits/sec                  
[  8]   1.07-2.00   sec  28.0 MBytes   251 Mbits/sec                  
[  8]   2.00-3.04   sec  19.7 MBytes   159 Mbits/sec                  
[  8]   3.04-4.02   sec  16.7 MBytes   143 Mbits/sec                  
[  8]   4.02-5.01   sec  20.0 MBytes   170 Mbits/sec                  
[  8]   5.01-6.00   sec  33.0 MBytes   278 Mbits/sec                  
[  8]   6.00-7.02   sec  34.0 MBytes   280 Mbits/sec                  
[  8]   7.02-8.00   sec  33.9 MBytes   291 Mbits/sec                  
[  8]   8.00-9.01   sec  42.8 MBytes   357 Mbits/sec                  
[  8]   9.01-10.01  sec  28.7 MBytes   241 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  8]   0.00-10.01  sec   287 MBytes   240 Mbits/sec                  sender
[  8]   0.00-10.01  sec   286 MBytes   240 Mbits/sec                  receiver

iperf Done.

--------------------------------------------------------------------------------------

Old Build 06-14-2018 (r6874)

Connecting to host 10.0.0.1, port 5201
[  8] local 10.0.0.3 port 59202 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  8]   0.00-1.00   sec  16.4 MBytes   137 Mbits/sec                  
[  8]   1.00-2.02   sec  34.7 MBytes   287 Mbits/sec                  
[  8]   2.02-3.02   sec  36.7 MBytes   306 Mbits/sec                  
[  8]   3.02-4.00   sec  29.6 MBytes   254 Mbits/sec                  
[  8]   4.00-5.02   sec  39.4 MBytes   326 Mbits/sec                  
[  8]   5.02-6.01   sec  34.8 MBytes   294 Mbits/sec                  
[  8]   6.01-7.01   sec  35.6 MBytes   297 Mbits/sec                  
[  8]   7.01-8.01   sec  34.7 MBytes   292 Mbits/sec                  
[  8]   8.01-9.04   sec  34.7 MBytes   283 Mbits/sec                  
[  8]   9.04-10.00  sec  35.9 MBytes   315 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  8]   0.00-10.00  sec   332 MBytes   279 Mbits/sec                  sender
[  8]   0.00-10.00  sec   332 MBytes   278 Mbits/sec                  receiver

iperf Done.

yuhhaurlin, 2018-07-03:
Sorry. From the iperf test result, I can't find there is significant performance differences between these two mwlwifi driver. Thanks.

closed 2018-07-03 by yuhhaurlin

bbogey, 2018-07-03:
@alirz1

You get webdav over ssl? Maybe hw acceleration cryptodev not enabled properly? Monitor cpu / sirq at the same time and do some openssl speed tests.

alirz1, 2018-07-03:
Nope, it's unencrypted webdav.

alirz1, 2018-07-03:
So I recompiled a new build today (r7382) but edited the makefile so that the build utilitized the March 30 comit for the mwlwifi. So as I suspected my wifi speed is still fast.
So there is definitely some issue with wifi thoroughput with the June mwlwifi driver regardless what iperf tests show.

yuhhaurlin, 2018-07-03:
Can you load other pre-built image to do test?

alirz1, 2018-07-04:
The only two sources that i know of, of a pre built image i.e an image not built by me are:
https://downloads.lede-project.org/releases/17.01.4/targets/mvebu/generic/
or
https://davidc502sis.dynamic-dns.net/

i can try one of those.

P.S. ill mention this again. The images that i'm compiling seem to show that kernel debugging is turned on and from what i see it all seems to be for the wlan interface. Could this debugging be causing the slow speeds?
in the system log i see lines like: Tue Jul 3 12:04:37 2018 kern.debug kernel: [ 177.475670] ieee80211 phy1: change: 0x100

yuhhaurlin, 2018-07-04:
Can you get the full log for me? If only the message you showed,  it is normal and it should not affect throughput.

yuhhaurlin, 2018-07-04:
BTW, what is the speedtest for these two version of drivers?

yuhhaurlin, 2018-07-04:
You mean test result of speedtest?

yuhhaurlin, 2018-07-04:
I suggest that you can do iperf test between LAN and your wireless client.

alirz1, 2018-07-04:
I guess you didn't see the full post above. I already did the iperf tests and they don't show any speed issues for some reason.

bbogey, 2018-07-04:
Is it correct direction? You can do download test with -R

alirz1, 2018-07-04:
yes it is.

alirz1, 2018-07-04:
Speedtest will show my WAN to LAN traffic. My WAN bandwidth is 120mbps. So a speedtest app test would somewhat be irrelevant to this issue. My issue is with LAN to wifi throughput.

yuhhaurlin, 2018-07-04:
Originally I think you test between router and client. If you already test between LAN and client, I have no comment now.

alirz1, 2018-07-07:
If it helps you in finding any problem and if you would like to do a real life 5g wifi speed test, not just iperf. I' ve compiled two images for the wrt1900ac off the trunk today(july 7th 2016).
The builds are identical except that one uses mwlwifi drive from March and the other uses mwlwifi drive from June.
https://drive.google.com/open?id=1S1QyTzrPid53y18u7f7AWNCQ-jKWHNvQ

alirz1, 2018-07-07:
@yuhhaurlin Can this be re-opened or tested by your side?

yuhhaurlin, 2018-07-07:
No. Iperf is good and I think speedtest is good too.

yuhhaurlin, 2018-07-07:
BTW, how about other pre-build image?

alirz1, 2018-07-07:
If you can give me a link for a prebuild image that you will like me to test. I can. Seems.like I'm the only one willing to test.

yuhhaurlin, 2018-07-07:
You already found two, right?

yuhhaurlin, 2018-07-07:
If iperf and speedtest is all right, I don’t know what can I do.