kaloz/mwlwifi #307: Very slow Wifi 5GHz, Intel 8260 (2x2), bad wifi performance in general !!!

Kherby, 2018-07-04:
Today i've noticed that websites were loading very slow when i was connected with my WRT3200acm.
Iperf test shows again that something is wrong with the throughput.... 
This happens every day and i have to reconnect Wifi or restart the Interface to fix it.
Also the general 5GHz Wifi performance is bad most of the time, same with 2.4GHz.
The throughput is not stable for some reason, even when i'm very close to the Router...

```
650.0 Mbit/s, 80MHz, VHT-MCS 7, VHT-NSS 2, Short GI
468.0 Mbit/s, 80MHz, VHT-MCS 5, VHT-NSS 2

root@OpenWRT:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20180330
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384
```

Iperf test:
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.155 -t 60s
Connecting to host 192.168.1.155, port 5201
[  4] local 192.168.1.199 port 60031 connected to 192.168.1.155 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   256 KBytes  2.09 Mbits/sec
[  4]   1.00-2.00   sec   256 KBytes  2.10 Mbits/sec
[  4]   2.00-3.00   sec   256 KBytes  2.10 Mbits/sec
[  4]   3.00-4.00   sec   384 KBytes  3.14 Mbits/sec
[  4]   4.00-5.00   sec   384 KBytes  3.15 Mbits/sec
[  4]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec
[  4]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec
[  4]  10.00-11.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  11.00-12.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  12.00-13.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  13.00-14.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  14.00-15.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  15.00-16.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  16.00-17.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  17.00-18.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  18.00-19.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  19.00-20.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  20.00-21.00  sec   128 KBytes  1.05 Mbits/sec
[  4]  21.00-22.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  22.00-23.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  23.00-24.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  24.00-25.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  25.00-26.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  26.00-27.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  27.00-28.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  28.00-29.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  29.00-30.00  sec  0.00 Bytes  0.00 bits/sec
[... 317 lines cut ...]
[  4]  42.00-43.00  sec  30.1 MBytes   252 Mbits/sec
[  4]  43.00-44.00  sec  30.0 MBytes   252 Mbits/sec
[  4]  44.00-45.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  45.00-46.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  46.00-47.00  sec  30.8 MBytes   258 Mbits/sec
[  4]  47.00-48.00  sec  26.8 MBytes   224 Mbits/sec
[  4]  48.00-49.00  sec  32.5 MBytes   273 Mbits/sec
[  4]  49.00-50.00  sec  25.6 MBytes   215 Mbits/sec
[  4]  50.00-51.00  sec  25.4 MBytes   213 Mbits/sec
[  4]  51.00-52.00  sec  23.4 MBytes   196 Mbits/sec
[  4]  52.00-53.00  sec  30.0 MBytes   252 Mbits/sec
[  4]  53.00-54.00  sec  28.5 MBytes   239 Mbits/sec
[  4]  54.00-55.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  55.00-56.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  56.00-57.00  sec  25.0 MBytes   210 Mbits/sec
[  4]  57.00-58.00  sec  31.5 MBytes   264 Mbits/sec
[  4]  58.00-59.00  sec  26.5 MBytes   222 Mbits/sec
[  4]  59.00-60.00  sec  30.1 MBytes   253 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  1.84 GBytes   263 Mbits/sec                  sender
[  4]   0.00-60.00  sec  1.84 GBytes   263 Mbits/sec                  receiver

iperf Done.
```

yuhhaurlin, 2018-07-04:
How about the test result with stock firmware?

ryan-dev, 2018-07-04:
I just want to chime in here due to I also have the WRT3200ACM with the
four high gain antennas from Linksys (aka Belkin).

Might I add @yuhhaurlin is not a developer for OpenWrt but for Marvell
itself (some folks get that mixed up) as this github repo is by the
maintainer (kaloz) for the Marvel mvebu target at OpenWrt.

I have had similar issues however you have left out lots of crucial
information.

Contrary to some information online the Linksys high gain antennas are not
intended for multi level dwellings (though it does work just fine installed
in the basement mounted on wall with antennas pointed upwards).

One major drawback for the WRT3200ACM series for 5ghz DFS channels is lots
of RF interference and actual radar signals.

Exactly what WiFi channels are you testing on, is there lots of
interference on current or nearby channels plus are all your tests on the
same channel?

>

yuhhaurlin, 2018-07-04:
If you have the same issue with stock firmware, you can check with Belkin. How about other clients?

Kherby, 2018-07-04:
I've tested a Dell Notebook with Qualcomm QCA61x4A 802.11ac (2x2) Adapter and the throughput seems to be better and way more stable than with my Intel 8260 (2x2) Adapter....

```
866.7 Mbit/s, 80MHz, VHT-MCS 9, VHT-NSS 2, Short GI
866.7 Mbit/s, 80MHz, VHT-MCS 9, VHT-NSS 2, Short GI

```
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.155 -t 30s
Connecting to host 192.168.1.155, port 5201
[  4] local 192.168.1.130 port 50244 connected to 192.168.1.155 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec  19.2 MBytes   160 Mbits/sec
[  4]   1.01-2.01   sec  17.5 MBytes   147 Mbits/sec
[  4]   2.01-3.00   sec  17.9 MBytes   151 Mbits/sec
[  4]   3.00-4.00   sec  17.9 MBytes   150 Mbits/sec
[  4]   4.00-5.01   sec  18.8 MBytes   157 Mbits/sec
[  4]   5.01-6.00   sec  18.8 MBytes   158 Mbits/sec
[  4]   6.00-7.01   sec  18.0 MBytes   150 Mbits/sec
[  4]   7.01-8.00   sec  20.4 MBytes   172 Mbits/sec
[  4]   8.00-9.00   sec  19.5 MBytes   164 Mbits/sec
[  4]   9.00-10.00  sec  18.9 MBytes   158 Mbits/sec
[  4]  10.00-11.00  sec  19.2 MBytes   161 Mbits/sec
[  4]  11.00-12.01  sec  15.8 MBytes   132 Mbits/sec
[  4]  12.01-13.00  sec  16.9 MBytes   142 Mbits/sec
[  4]  13.00-14.00  sec  20.4 MBytes   171 Mbits/sec
[  4]  14.00-15.00  sec  19.6 MBytes   165 Mbits/sec
[  4]  15.00-16.01  sec  19.0 MBytes   159 Mbits/sec
[  4]  16.01-17.00  sec  20.5 MBytes   173 Mbits/sec
[  4]  17.00-18.00  sec  21.4 MBytes   179 Mbits/sec
[  4]  18.00-19.00  sec  16.4 MBytes   138 Mbits/sec
[  4]  19.00-20.00  sec  16.5 MBytes   138 Mbits/sec
[  4]  20.00-21.00  sec  18.4 MBytes   155 Mbits/sec
[  4]  21.00-22.01  sec  16.2 MBytes   136 Mbits/sec
[  4]  22.01-23.00  sec  16.5 MBytes   139 Mbits/sec
[  4]  23.00-24.00  sec  15.5 MBytes   130 Mbits/sec
[  4]  24.00-25.00  sec  15.4 MBytes   129 Mbits/sec
[  4]  25.00-26.00  sec  16.0 MBytes   134 Mbits/sec
[  4]  26.00-27.00  sec  15.4 MBytes   129 Mbits/sec
[  4]  27.00-28.00  sec  17.0 MBytes   143 Mbits/sec
[  4]  28.00-29.00  sec  16.8 MBytes   140 Mbits/sec
[  4]  29.00-30.01  sec  16.4 MBytes   137 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-30.01  sec   536 MBytes   150 Mbits/sec                  sender
[  4]   0.00-30.01  sec   536 MBytes   150 Mbits/sec                  receiver

iperf Done.
```

1m away from the Router:
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.155 -t 30s
Connecting to host 192.168.1.155, port 5201
[  4] local 192.168.1.130 port 50304 connected to 192.168.1.155 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  20.4 MBytes   170 Mbits/sec
[  4]   1.00-2.01   sec  21.6 MBytes   181 Mbits/sec
[  4]   2.01-3.00   sec  21.1 MBytes   178 Mbits/sec
[  4]   3.00-4.00   sec  20.8 MBytes   174 Mbits/sec
[  4]   4.00-5.00   sec  21.6 MBytes   181 Mbits/sec
[  4]   5.00-6.01   sec  19.6 MBytes   164 Mbits/sec
[  4]   6.01-7.00   sec  20.6 MBytes   174 Mbits/sec
[  4]   7.00-8.00   sec  27.1 MBytes   227 Mbits/sec
[  4]   8.00-9.00   sec  26.6 MBytes   223 Mbits/sec
[  4]   9.00-10.00  sec  26.8 MBytes   225 Mbits/sec
[  4]  10.00-11.00  sec  27.0 MBytes   226 Mbits/sec
[  4]  11.00-12.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  12.00-13.00  sec  26.8 MBytes   225 Mbits/sec
[  4]  13.00-14.01  sec  27.4 MBytes   229 Mbits/sec
[  4]  14.01-15.01  sec  25.9 MBytes   217 Mbits/sec
[  4]  15.01-16.00  sec  26.8 MBytes   226 Mbits/sec
[  4]  16.00-17.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  17.00-18.00  sec  27.2 MBytes   228 Mbits/sec
[  4]  18.00-19.00  sec  27.2 MBytes   229 Mbits/sec
[  4]  19.00-20.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  20.00-21.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  21.00-22.00  sec  27.9 MBytes   234 Mbits/sec
[  4]  22.00-23.00  sec  27.2 MBytes   228 Mbits/sec
[  4]  23.00-24.00  sec  27.8 MBytes   233 Mbits/sec
[  4]  24.00-25.01  sec  27.5 MBytes   229 Mbits/sec
[  4]  25.01-26.00  sec  27.5 MBytes   232 Mbits/sec
[  4]  26.00-27.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  27.00-28.01  sec  26.9 MBytes   224 Mbits/sec
[  4]  28.01-29.00  sec  27.5 MBytes   232 Mbits/sec
[  4]  29.00-30.01  sec  27.6 MBytes   231 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-30.01  sec   772 MBytes   216 Mbits/sec                  sender
[  4]   0.00-30.01  sec   772 MBytes   216 Mbits/sec                  receiver

iperf Done.
```

The Intel 8260 doesn't seem to be a bad adapter but for some reason it doesn't work well with the Marvel Wifi chip or the used driver...
With the Qualcomm Adapter it seems to be fine and there are no evil speed drops to 30-40mbps like with my Intel 8260 adapter...
I'm using driver version: 20.60.07 (09.05.2018) at the moment.
Should i test an older Intel driver ? 

@ryan-dev: All test done via channel 100(DFS-Auto) @27dBm and at the same place...
I bought the high gain antennas and i only use them on a horizontal plane and installed them poitining updwards...
There are some other 5GHz AP's in my area but not that much and none of them are near Channel 100 and as i said i dont have any problem with my DIR-860L using OpenWRT and the Intel 8260 Adapter.
I've also tested channel 36 and 52 @20dBm and it doesn't make any difference, same drops, ~same throughput.

Marvel and Intel 8260 doesn't seem to go well together and I don't think my Intel adapter is faulty as i only have these problems with the WRT3200acm.
As requested im going to test my Intel Adapter with the stock firmware but im pretty sure that it won't change the problem at all...

Is there anything else i can do ? I really want to fix the problem and it maybe could help others as well !

Kherby, 2018-07-04:
Here is another test @ channel 36 with the Intel AC8260 adapter:

```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.555 -t 60s
Connecting to host 192.168.1.155, port 5201
[  4] local 192.168.1.199 port 49366 connected to 192.168.1.155 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  16.2 MBytes   136 Mbits/sec
[  4]   1.00-2.00   sec  17.5 MBytes   147 Mbits/sec
[  4]   2.00-3.00   sec  17.1 MBytes   144 Mbits/sec
[  4]   3.00-4.00   sec  15.6 MBytes   131 Mbits/sec
[  4]   4.00-5.00   sec  9.12 MBytes  76.5 Mbits/sec
[  4]   5.00-6.00   sec  3.50 MBytes  29.3 Mbits/sec
[  4]   6.00-7.00   sec  4.38 MBytes  36.7 Mbits/sec
[  4]   7.00-8.00   sec  4.12 MBytes  34.6 Mbits/sec
[  4]   8.00-9.00   sec  13.1 MBytes   110 Mbits/sec
[  4]   9.00-10.00  sec  15.4 MBytes   129 Mbits/sec
[  4]  10.00-11.00  sec  17.5 MBytes   147 Mbits/sec
[  4]  11.00-12.00  sec  17.9 MBytes   150 Mbits/sec
[  4]  12.00-13.00  sec  11.1 MBytes  93.3 Mbits/sec
[  4]  13.00-14.00  sec  12.4 MBytes   104 Mbits/sec
[  4]  14.00-15.00  sec  4.38 MBytes  36.7 Mbits/sec
[  4]  15.00-16.00  sec  4.50 MBytes  37.8 Mbits/sec
[  4]  16.00-17.00  sec  8.75 MBytes  73.4 Mbits/sec
[  4]  17.00-18.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  18.00-19.00  sec  17.0 MBytes   143 Mbits/sec
[  4]  19.00-20.00  sec  13.6 MBytes   114 Mbits/sec
[  4]  20.00-21.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  21.00-22.00  sec  11.0 MBytes  92.3 Mbits/sec
[  4]  22.00-23.00  sec  3.38 MBytes  28.3 Mbits/sec
[  4]  23.00-24.00  sec  3.75 MBytes  31.4 Mbits/sec
[  4]  24.00-25.00  sec  3.75 MBytes  31.5 Mbits/sec
[  4]  25.00-26.00  sec  20.9 MBytes   175 Mbits/sec
[  4]  26.00-27.00  sec  22.0 MBytes   185 Mbits/sec
[  4]  27.00-28.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  28.00-29.00  sec  22.4 MBytes   188 Mbits/sec
[  4]  29.00-30.00  sec  22.1 MBytes   185 Mbits/sec
[  4]  30.00-31.00  sec  10.9 MBytes  91.2 Mbits/sec
[  4]  31.00-32.00  sec  6.00 MBytes  50.4 Mbits/sec
[  4]  32.00-33.00  sec  8.00 MBytes  67.1 Mbits/sec
[  4]  33.00-34.00  sec  8.12 MBytes  68.2 Mbits/sec
[  4]  34.00-35.00  sec  7.38 MBytes  61.9 Mbits/sec
[  4]  35.00-36.00  sec  1.75 MBytes  14.7 Mbits/sec
[  4]  36.00-37.00  sec  1.88 MBytes  15.7 Mbits/sec
[  4]  37.00-38.00  sec  2.38 MBytes  19.9 Mbits/sec
[  4]  38.00-39.00  sec  3.12 MBytes  26.2 Mbits/sec
[  4]  39.00-40.00  sec  9.25 MBytes  77.7 Mbits/sec
[  4]  40.00-41.00  sec  10.2 MBytes  85.9 Mbits/sec
[  4]  41.00-42.00  sec  3.62 MBytes  30.4 Mbits/sec
[  4]  42.00-43.00  sec  10.9 MBytes  91.2 Mbits/sec
[  4]  43.00-44.00  sec  4.50 MBytes  37.8 Mbits/sec
[  4]  44.00-45.00  sec  2.12 MBytes  17.8 Mbits/sec
[  4]  45.00-46.00  sec   640 KBytes  5.25 Mbits/sec
[  4]  46.00-47.00  sec  4.38 MBytes  36.7 Mbits/sec
[  4]  47.00-48.00  sec   896 KBytes  7.34 Mbits/sec
[  4]  48.00-49.00  sec  1.12 MBytes  9.44 Mbits/sec
[  4]  49.00-50.00  sec  4.50 MBytes  37.7 Mbits/sec
[  4]  50.00-51.00  sec  2.25 MBytes  18.9 Mbits/sec
[  4]  51.00-52.00  sec  5.88 MBytes  49.2 Mbits/sec
[  4]  52.00-53.00  sec  2.25 MBytes  18.9 Mbits/sec
[  4]  53.00-54.00  sec  2.50 MBytes  21.0 Mbits/sec
[  4]  54.00-55.00  sec  5.88 MBytes  49.3 Mbits/sec
[  4]  55.00-56.00  sec  8.00 MBytes  67.1 Mbits/sec
[  4]  56.00-57.00  sec  3.75 MBytes  31.4 Mbits/sec
[  4]  57.00-58.00  sec  5.88 MBytes  49.3 Mbits/sec
[  4]  58.00-59.00  sec  5.00 MBytes  41.9 Mbits/sec
[  4]  59.00-60.00  sec  4.38 MBytes  36.6 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec   535 MBytes  74.8 Mbits/sec                  sender
[  4]   0.00-60.00  sec   535 MBytes  74.8 Mbits/sec                  receiver

iperf Done.

```

yuhhaurlin, 2018-07-04:
Please test with stock firmware. I will get intel client to check it later. Thanks.

Kherby, 2018-07-05:
Here are the results with the stock firmware... 
Firmware Version: 1.0.6.186168

5GHz @ channel 36 @ 80MHz, right next to the router:
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.199 -t 60s
Connecting to host 192.168.1.199, port 5201
[  4] local 192.168.1.130 port 50872 connected to 192.168.1.199 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  36.0 MBytes   301 Mbits/sec
[  4]   1.00-2.00   sec  34.5 MBytes   290 Mbits/sec
[  4]   2.00-3.00   sec  36.1 MBytes   303 Mbits/sec
[  4]   3.00-4.00   sec  34.5 MBytes   289 Mbits/sec
[  4]   4.00-5.00   sec  34.4 MBytes   289 Mbits/sec
[  4]   5.00-6.00   sec  34.9 MBytes   292 Mbits/sec
[  4]   6.00-7.00   sec  34.6 MBytes   291 Mbits/sec
[  4]   7.00-8.00   sec  35.2 MBytes   295 Mbits/sec
[  4]   8.00-9.00   sec  34.1 MBytes   287 Mbits/sec
[  4]   9.00-10.00  sec  34.5 MBytes   289 Mbits/sec
[  4]  10.00-11.00  sec  34.5 MBytes   289 Mbits/sec
[  4]  11.00-12.00  sec  33.2 MBytes   279 Mbits/sec
[  4]  12.00-13.00  sec  34.8 MBytes   291 Mbits/sec
[  4]  13.00-14.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  14.00-15.00  sec  35.0 MBytes   294 Mbits/sec
[  4]  15.00-16.00  sec  35.0 MBytes   293 Mbits/sec
[  4]  16.00-17.00  sec  34.1 MBytes   287 Mbits/sec
[  4]  17.00-18.00  sec  34.8 MBytes   291 Mbits/sec
[  4]  18.00-19.00  sec  35.1 MBytes   294 Mbits/sec
[  4]  19.00-20.00  sec  34.5 MBytes   290 Mbits/sec
[  4]  20.00-21.00  sec  34.8 MBytes   292 Mbits/sec
[  4]  21.00-22.00  sec  35.0 MBytes   293 Mbits/sec
[  4]  22.00-23.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  23.00-24.00  sec  34.5 MBytes   289 Mbits/sec
[  4]  24.00-25.00  sec  35.4 MBytes   297 Mbits/sec
[  4]  25.00-26.00  sec  34.6 MBytes   290 Mbits/sec
[  4]  26.00-27.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  27.00-28.00  sec  35.2 MBytes   296 Mbits/sec
[  4]  28.00-29.00  sec  35.0 MBytes   294 Mbits/sec
[  4]  29.00-30.00  sec  34.9 MBytes   292 Mbits/sec
[  4]  30.00-31.00  sec  34.6 MBytes   291 Mbits/sec
[  4]  31.00-32.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  32.00-33.00  sec  34.8 MBytes   291 Mbits/sec
[  4]  33.00-34.00  sec  34.6 MBytes   290 Mbits/sec
[  4]  34.00-35.00  sec  34.2 MBytes   288 Mbits/sec
[  4]  35.00-36.00  sec  35.0 MBytes   293 Mbits/sec
[  4]  36.00-37.00  sec  35.2 MBytes   296 Mbits/sec
[  4]  37.00-38.00  sec  34.8 MBytes   292 Mbits/sec
[  4]  38.00-39.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  39.00-40.00  sec  34.9 MBytes   292 Mbits/sec
[  4]  40.00-41.00  sec  34.1 MBytes   287 Mbits/sec
[  4]  41.00-42.00  sec  34.8 MBytes   291 Mbits/sec
[  4]  42.00-43.00  sec  34.2 MBytes   287 Mbits/sec
[  4]  43.00-44.00  sec  34.6 MBytes   291 Mbits/sec
[  4]  44.00-45.00  sec  34.6 MBytes   291 Mbits/sec
[  4]  45.00-46.00  sec  34.2 MBytes   287 Mbits/sec
[  4]  46.00-47.00  sec  34.6 MBytes   291 Mbits/sec
[  4]  47.00-48.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  48.00-49.00  sec  34.9 MBytes   292 Mbits/sec
[  4]  49.00-50.00  sec  34.6 MBytes   291 Mbits/sec
[  4]  50.00-51.00  sec  35.4 MBytes   296 Mbits/sec
[  4]  51.00-52.00  sec  35.0 MBytes   294 Mbits/sec
[  4]  52.00-53.00  sec  34.6 MBytes   290 Mbits/sec
[  4]  53.00-54.00  sec  35.6 MBytes   299 Mbits/sec
[  4]  54.00-55.00  sec  35.1 MBytes   295 Mbits/sec
[  4]  55.00-56.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  56.00-57.00  sec  35.0 MBytes   294 Mbits/sec
[  4]  57.00-58.00  sec  34.8 MBytes   291 Mbits/sec
[  4]  58.00-59.00  sec  34.9 MBytes   293 Mbits/sec
[  4]  59.00-60.00  sec  34.6 MBytes   290 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 133 lines cut ...]
[  4]  11.00-12.00  sec  8.88 MBytes  74.3 Mbits/sec
[  4]  12.00-13.00  sec  9.38 MBytes  78.7 Mbits/sec
[  4]  13.00-14.00  sec  9.75 MBytes  81.8 Mbits/sec
[  4]  14.00-15.00  sec  10.6 MBytes  89.2 Mbits/sec
[  4]  15.00-16.00  sec  12.2 MBytes   103 Mbits/sec
[  4]  16.00-17.00  sec  10.9 MBytes  91.2 Mbits/sec
[  4]  17.00-18.00  sec  11.6 MBytes  97.5 Mbits/sec
[  4]  18.00-19.00  sec  10.0 MBytes  84.0 Mbits/sec
[  4]  19.00-20.00  sec  11.6 MBytes  97.6 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec   176 MBytes  73.6 Mbits/sec                  sender
[  4]   0.00-20.00  sec   175 MBytes  73.6 Mbits/sec                  receiver

iperf Done.

```

No more speed drops to 40Mbps and stable performance :/

So imho the bad performance with my Intel 8260 Adapter has to be something with the Marvel Wifi driver within OpenWRT.
The good thing is that i now know the hardware of my WRT3200acm isn't damaged.
I bought the device only for OpenWRT and the stock firmware doesn't even give me 10% of the functions i need for my network and internet setup.

@yuhhaurlin, can you please fix this problem ? I beg you...

yuhhaurlin, 2018-07-05:
Thanks for your information. I will check it later.

Kherby, 2018-07-05:
@yuhhaurlin , i have an additional information that might help you...
I had some setup problems with the lastest build of davidc502 so i reverted back to the latest build before r7360...

Right now i'm using build r7210 and my initial testings @5GHz with **iperf reverse mode (-R)** are looking quite good... But without -R the Speed is still dropping down to 40Mbps.
So when my Intel adapter is receiving the data instead of sending it everything seems to be alright...

```
root@WRT3200ACM:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180614
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384
```

5GHz @ auto channel (52) @ 80MHz, ~10-15m away from the Router **(-R)**:
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.199 -t 60s -R
Connecting to host 192.168.1.199, port 5201
Reverse mode, remote host 192.168.1.199 is sending
[  4] local 192.168.1.230 port 53551 connected to 192.168.1.199 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  16.3 MBytes   137 Mbits/sec
[  4]   1.00-2.00   sec  16.0 MBytes   134 Mbits/sec
[  4]   2.00-3.00   sec  16.0 MBytes   134 Mbits/sec
[  4]   3.00-4.00   sec  15.4 MBytes   129 Mbits/sec
[  4]   4.00-5.00   sec  16.2 MBytes   136 Mbits/sec
[  4]   5.00-6.00   sec  15.9 MBytes   134 Mbits/sec
[  4]   6.00-7.00   sec  14.6 MBytes   122 Mbits/sec
[  4]   7.00-8.00   sec  15.2 MBytes   128 Mbits/sec
[  4]   8.00-9.00   sec  15.3 MBytes   128 Mbits/sec
[  4]   9.00-10.00  sec  15.9 MBytes   133 Mbits/sec
[  4]  10.00-11.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  11.00-12.00  sec  16.0 MBytes   134 Mbits/sec
[  4]  12.00-13.00  sec  16.1 MBytes   135 Mbits/sec
[  4]  13.00-14.00  sec  15.6 MBytes   131 Mbits/sec
[  4]  14.00-15.00  sec  15.3 MBytes   129 Mbits/sec
[  4]  15.00-16.00  sec  14.8 MBytes   124 Mbits/sec
[  4]  16.00-17.00  sec  14.3 MBytes   120 Mbits/sec
[  4]  17.00-18.00  sec  16.3 MBytes   137 Mbits/sec
[  4]  18.00-19.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  19.00-20.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  20.00-21.00  sec  15.6 MBytes   131 Mbits/sec
[  4]  21.00-22.00  sec  16.1 MBytes   135 Mbits/sec
[  4]  22.00-23.00  sec  15.1 MBytes   127 Mbits/sec
[  4]  23.00-24.00  sec  14.9 MBytes   125 Mbits/sec
[  4]  24.00-25.00  sec  15.1 MBytes   126 Mbits/sec
[  4]  25.00-26.00  sec  15.4 MBytes   129 Mbits/sec
[  4]  26.00-27.00  sec  16.0 MBytes   134 Mbits/sec
[  4]  27.00-28.00  sec  15.8 MBytes   132 Mbits/sec
[  4]  28.00-29.00  sec  15.7 MBytes   132 Mbits/sec
[  4]  29.00-30.00  sec  16.2 MBytes   136 Mbits/sec
[  4]  30.00-31.00  sec  15.7 MBytes   132 Mbits/sec
[  4]  31.00-32.00  sec  14.5 MBytes   121 Mbits/sec
[... 85 lines cut ...]
[  4]  42.00-43.00  sec  19.6 MBytes   165 Mbits/sec
[  4]  43.00-44.00  sec  15.6 MBytes   131 Mbits/sec
[  4]  44.00-45.00  sec  16.9 MBytes   142 Mbits/sec
[  4]  45.00-46.00  sec  20.1 MBytes   169 Mbits/sec
[  4]  46.00-47.00  sec  19.1 MBytes   160 Mbits/sec
[  4]  47.00-48.00  sec  3.38 MBytes  28.3 Mbits/sec
[  4]  48.00-49.00  sec  3.50 MBytes  29.4 Mbits/sec
[  4]  49.00-50.00  sec  5.38 MBytes  45.0 Mbits/sec
[  4]  50.00-51.00  sec  21.4 MBytes   180 Mbits/sec
[  4]  51.00-52.00  sec  22.0 MBytes   184 Mbits/sec
[  4]  52.00-53.00  sec  21.9 MBytes   183 Mbits/sec
[  4]  53.00-54.00  sec  20.6 MBytes   173 Mbits/sec
[  4]  54.00-55.00  sec  20.8 MBytes   174 Mbits/sec
[  4]  55.00-56.00  sec  7.50 MBytes  62.8 Mbits/sec
[  4]  56.00-57.00  sec  4.50 MBytes  37.8 Mbits/sec
[  4]  57.00-58.00  sec  4.50 MBytes  37.8 Mbits/sec
[  4]  58.00-59.00  sec  15.9 MBytes   133 Mbits/sec
[  4]  59.00-60.00  sec  20.8 MBytes   174 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec   842 MBytes   118 Mbits/sec                  sender
[  4]   0.00-60.00  sec   842 MBytes   118 Mbits/sec                  receiver

iperf Done.
```

yuhhaurlin, 2018-07-05:
Yes. Look like Tx has problem with intel 8260.

PalebloodSky, 2018-07-06:
It's absurd there are totally separate developers and projects for the closed source and open source Wi-Fi drivers for these WRT routers. The reputation is already poor for Belkin/Linksys' lack of support and this kind of stuff keeps happening. It should be one source code for all and things would get much better.

yuhhaurlin, 2018-07-07:
How? Open source is not the same as proprietary driver.

bbogey, 2018-07-07:
I think @palebloosky means the firmware blob what is not opensource? But similar it is for atheros as well?

b-h-l-c, 2018-07-07:
Same issue here with this client! Thought I was crazy

Briton H.L. Claridge

Kherby, 2018-07-07:
@yuhhaurlin 
Any news ? Can we expect a fix for the Intel AC8260 adapters ? :)

@b-h-l-c 
It was driving me crazy as well...
At the beginning i thought that my adapter is faulty or the antennas of my Notebook are damaged in some way, but I quickly realized that the poor performance only occurs with my WRT3200ACM@OpenWRT.
With my other AP's the Intel 8260 card is working just fine.
I realy hope that yuhhaurlin will be able to fix this problem as the Intel 8260 is a pretty common adapter used in many notebooks...

PalebloodSky, 2018-07-07:
Linksys firmware update had this: "A-MSDU max frame size capped at 7.9K to improve interoperability with Intel 9260-based clients", I wonder if something related could fix the 8260s too.

yuhhaurlin, 2018-07-08:
Thanks for your information.

BrainSlayer, 2018-07-11:
okay i put some informations here about iperf. running iperf on the WRT itself is a bad idea. iperf will decrease the network performance by its own cpu load and the timer accuracy in these devices is not as good as in a standard pc. so it will fuck up every measurement. run iperf on a second pc connected to the router and then do your measurements. everything else will just give you bogus results

BrainSlayer, 2018-07-11:
next hint. use always multiple tcp streams. single tcp streams are meaningless and will never show the full performance. and best for measuring the real throughput is using udp for sure. all these tests here are tcp based which isnt really the best way to get the maximum throughput

yuhhaurlin, 2018-07-11:
1. Test from router is simply to check if throughput has problem.
2. For our QA, the throughput is measured with multiple streams in shielding room for UDP and TCP.
3. In open air, it is better to have a reference AP to compare throughput.
4. Different clients with difference version of drivers sometimes will get different test result.
5. You need to know capability of your client to know the throughput it can reach.

Kherby, 2018-07-11:
@BrainSlayer 
I've always tested the Wifi throughput with a decent PC (i7, 16GB RAM, SSD) as server, which is connected via Gigabit LAN.
Could you please tell me the iperf commands (server+client) for proper testing with UDP?

LAN >> LAN
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  6.56 GBytes   939 Mbits/sec                  sender
[  4]   0.00-60.00  sec  6.56 GBytes   939 Mbits/sec                  receiver

iperf Done.
```

alexhartl, 2018-07-13:
Same issue here with an Intel 6300agn as client (and WRT3200ACM as router). When sending using the stock image, the bitrate stays at about 200 MBit/s, using OpenWrt it periodically fluctuates between ~200 MBit/s and ~40 MBit/s, resulting in an effective rate of ~120 MBit/s. In both cases Windows reports a constant nominal bitrate of 450 MBit/s.
I would really be interested in a fix. Let me know if you need some information from me.

Kherby, 2018-07-17:
I'm going to test Intel 8265 in a few days (HP Probook) with the WRT3200 @OpenWRT and let you know if the same problem occurs.

And i would appreciate if someone could tell me the right iperf commands for multiple TCP streams and UDP testings...
> use always multiple tcp streams. single tcp streams are meaningless and will never show the full performance. and best for measuring the real throughput is using udp for sure. all these tests here are tcp based which isnt really the best way to get the maximum throughput

makedir, 2018-07-18:
Was hinted by someone else to this thread.

I have the following issue on my WRT1200 with the latest firmware by David502. I got a new laptop which is my first unit, which is able to do at least 866mbit over 5Ghz, my older laptops just all could do 300mbit. It has an Intel 9260 card. The issue now is, that the moment I send files from it to the network, the file copy process breaks down after 1-2 seconds and goes to zero kb/s and idle out after a long time and give a network error. Doing a ping while this happens, the ping to the router suddenly all goes up to about 133ms. Downspeed from the LAN to the laptops works normally of about 33MB/s (which is also though not close to half of 866mbit, it should be in theory for half duplex be around at least 420mbit). Windows show 866mbit and 4 green bars to the WRT1200. Config for the 5Ghz is option htmode 'VHT80'. Setting it to HT40, and the upload works stable for the 300mbit. Anyone else have similar issues or know what could cause it?

I tested a second router (non Openwrt, factory firmware) I had to hand which also can do 866mbit on 5Ghz, and the upload works stable from the laptop to the LAN, though a bit slow, about 27mb/s.

Kherby, 2018-07-22:
@yuhhaurlin Any news ?

alexhartl, 2018-08-04:
@Kherby , could you test your performance when running Linux on your notebook? On my machine (Intel 6300agn) wifi performance is quite bad running Linux with 2%-10% packet loss for both the stock image and OpenWRT. I wonder if this is a problem for all Intel cards or just for my card specifically.
Not sure if this is related to this bug, though.

makedir, 2018-08-04:
I actually noticed, that I also have bad performance with an Intel 6235 card with my old laptop. That was actually the reason why I bought a laptop... which now has a 9260 card. The 6235 has random high ping jumps (60-100ms) and also it seems to have a bit of packet loss too, because I always have random lags when I use InputDirector to remote control over to a PC, and the connection uses UDP.

Kherby, 2018-08-04:
@alexhartl, i will test the performance under linux but i can only speak for the Intel 8260 chipset...
In the meanwhile i've also updated to the lastest Intel driver (20.70.0.5) and the lastest mwlwifi driver version but the bad performance (Tx) haven't changed. I'm still hoping that @yuhhaurlin will sort out these issues...
```
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180716
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address:
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e1100000
iobase1: e1380000
tx limit: 1024
rx limit: 16384
```
```
866.7 Mbit/s, 80MHz, VHT-MCS 9, VHT-NSS 2, Short GI
866.7 Mbit/s, 80MHz, VHT-MCS 9, VHT-NSS 2, Short GI
```
```
C:\Users\Admin\iperf-3.1.3-win64>iperf3.exe -c 192.168.1.249 -t 30s
Connecting to host 192.168.1.249, port 5201
[  4] local 192.168.1.230 port 58686 connected to 192.168.1.249 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  28.4 MBytes   238 Mbits/sec
[  4]   1.00-2.00   sec  24.2 MBytes   203 Mbits/sec
[  4]   2.00-3.00   sec  25.8 MBytes   216 Mbits/sec
[  4]   3.00-4.00   sec  24.5 MBytes   205 Mbits/sec
[  4]   4.00-5.00   sec  25.2 MBytes   212 Mbits/sec
[  4]   5.00-6.00   sec  20.9 MBytes   175 Mbits/sec
[  4]   6.00-7.01   sec  5.00 MBytes  41.6 Mbits/sec
[  4]   7.01-8.01   sec  4.75 MBytes  40.0 Mbits/sec
[  4]   8.01-9.00   sec  7.75 MBytes  65.2 Mbits/sec
[  4]   9.00-10.00  sec  23.6 MBytes   199 Mbits/sec
[  4]  10.00-11.00  sec  28.4 MBytes   237 Mbits/sec
[  4]  11.00-12.00  sec  30.1 MBytes   253 Mbits/sec
[  4]  12.00-13.00  sec  27.8 MBytes   233 Mbits/sec
[  4]  13.00-14.00  sec  26.5 MBytes   222 Mbits/sec
[  4]  14.00-15.01  sec  6.50 MBytes  54.2 Mbits/sec
[  4]  15.01-16.01  sec  5.00 MBytes  42.1 Mbits/sec
[  4]  16.01-17.01  sec  4.88 MBytes  40.9 Mbits/sec
[  4]  17.01-18.00  sec  24.0 MBytes   202 Mbits/sec
[  4]  18.00-19.01  sec  32.8 MBytes   273 Mbits/sec
[  4]  19.01-20.00  sec  32.8 MBytes   277 Mbits/sec
[  4]  20.00-21.00  sec  20.4 MBytes   170 Mbits/sec
[  4]  21.00-22.00  sec  32.2 MBytes   271 Mbits/sec
[  4]  22.00-23.00  sec  15.9 MBytes   133 Mbits/sec
[  4]  23.00-24.01  sec  4.88 MBytes  40.8 Mbits/sec
[  4]  24.01-25.01  sec  5.00 MBytes  41.6 Mbits/sec
[  4]  25.01-26.00  sec  18.8 MBytes   159 Mbits/sec
[  4]  26.00-27.00  sec  33.4 MBytes   280 Mbits/sec
[  4]  27.00-28.00  sec  33.5 MBytes   281 Mbits/sec
[  4]  28.00-29.00  sec  31.6 MBytes   265 Mbits/sec
[  4]  29.00-30.00  sec  29.6 MBytes   249 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-30.00  sec   634 MBytes   177 Mbits/sec                  sender
[  4]   0.00-30.00  sec   634 MBytes   177 Mbits/sec                  receiver

iperf Done.
```

makedir, 2018-08-05:
Don't you have another router to test, if the Intel cards behave that bad on them too? Because I tested it, and they do, at least, someway. Maybe the better place to post the problems is on the Intel forums, or next to this.

alexhartl, 2018-08-05:
I tried with a TP-Link router and a Ubee router and didn't have any packet loss and good rates with both of them. However, they don't have the same capabilities as the WRT3200 (only 2x2 mimo), so yes, this could also be an issue on the Intel side. Yet I think if Intel cards still weren't able to handle 11n properly on Linux there would be tons of discussions about this around the Internet...
I'm not sure if we're talking about the same issue here, because my packet loss problem only occurs with Linux and concluding from your use of InputDirector, you were running Windows on your old laptop, right?

makedir, 2018-08-05:
Laptop Wifi cards just can do 2x2. My 6235 card was never stable for years and had always problems on the 5Ghz, it was a bit better on 2.4Ghz. The moment you used n mode it became unstable. But the 9260 is terribly worth, it seems to get even worse the more channels it uses simultaneously. 1 channel (50mbit) ok, 2 not so ok, 3-4 bad bad bad. Yes Windows, and when I use InputDirecotr and go with the mouse over to other PCs, I have random lags with it. Because it uses UDP, it is really "picky" for tiny little packet losses. I always thought it was because of bad OpenWRT wifi drivers, but I have the same issue on another router with the Intel card.

[image]

Kherby, 2018-08-05:
> Don't you have another router to test, if the Intel cards behave that bad on them too?

My Intel 8260 card does work completly fine with my DIR-860L B1@OpenWRT (MediaTek MT7612E, MT7602E) and as i said it's also working fine with the lastest stock firmware of the WRT3200ACM. Only mwlwifi@OpenWRT seems to have problems with this card...

marc-o94, 2018-08-15:
I have a similar issue with my Google Pixel smartphone using an WRT1900ACS v2. Here are some iperf3 statistics with really low transfer rates in the first 11 seconds:

```
root@OpenWrt:~# iperf3 -c 192.168.42.148 -i 1 -t 60
Connecting to host 192.168.42.148, port 5201
[  5] local 192.168.42.1 port 44878 connected to 192.168.42.148 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   328 KBytes  2.68 Mbits/sec    1   63.6 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec   13   5.66 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    6   5.66 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec   11   2.83 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   6.00-7.00   sec   509 KBytes  4.17 Mbits/sec   27   35.4 KBytes       
[  5]   7.00-8.00   sec   382 KBytes  3.13 Mbits/sec    2   17.0 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec   18   1.41 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    3   1.41 KBytes       
[  5]  10.00-11.00  sec   764 KBytes  6.26 Mbits/sec   31   45.2 KBytes       
[  5]  11.00-12.00  sec  30.4 MBytes   255 Mbits/sec    0   1.10 MBytes       
[  5]  12.00-13.00  sec  28.4 MBytes   238 Mbits/sec    0   1.46 MBytes       
[  5]  13.00-14.00  sec  35.0 MBytes   294 Mbits/sec    0   2.03 MBytes       
[  5]  14.00-15.00  sec  37.5 MBytes   314 Mbits/sec    0   2.03 MBytes       
[  5]  15.00-16.00  sec  36.2 MBytes   304 Mbits/sec    0   2.03 MBytes       
[  5]  16.00-17.00  sec  37.5 MBytes   314 Mbits/sec    0   2.03 MBytes       
[  5]  17.00-18.00  sec  33.8 MBytes   283 Mbits/sec    0   2.03 MBytes       
[  5]  18.00-19.00  sec  32.5 MBytes   273 Mbits/sec    0   2.03 MBytes       
[  5]  19.00-20.00  sec  32.5 MBytes   273 Mbits/sec    0   2.03 MBytes       
[  5]  20.00-21.00  sec  27.5 MBytes   231 Mbits/sec    0   2.03 MBytes       
[  5]  21.00-22.00  sec  26.2 MBytes   220 Mbits/sec    0   2.03 MBytes       
[  5]  22.00-23.00  sec  26.2 MBytes   220 Mbits/sec    0   2.03 MBytes       
[  5]  23.00-24.00  sec  32.5 MBytes   273 Mbits/sec    0   2.03 MBytes       
[  5]  24.00-25.00  sec  31.2 MBytes   262 Mbits/sec    0   2.03 MBytes       
[  5]  25.00-26.00  sec  36.2 MBytes   304 Mbits/sec  125    663 KBytes       
[  5]  26.00-27.00  sec  36.2 MBytes   304 Mbits/sec    0    708 KBytes       
[  5]  27.00-28.00  sec  31.2 MBytes   262 Mbits/sec    0    738 KBytes       
[  5]  28.00-29.00  sec  30.0 MBytes   252 Mbits/sec    0    757 KBytes       
[  5]  29.00-30.00  sec  33.8 MBytes   283 Mbits/sec    0    789 KBytes       
[  5]  30.00-31.00  sec  30.0 MBytes   252 Mbits/sec    0    816 KBytes       
[  5]  31.00-32.00  sec  32.5 MBytes   273 Mbits/sec    0    841 KBytes       
[  5]  32.00-33.00  sec  37.5 MBytes   315 Mbits/sec    0    875 KBytes       
[  5]  33.00-34.00  sec  36.2 MBytes   304 Mbits/sec    0    906 KBytes       
[  5]  34.00-35.00  sec  38.8 MBytes   325 Mbits/sec    0    938 KBytes       
[  5]  35.00-36.00  sec  30.0 MBytes   252 Mbits/sec    0    953 KBytes       
[  5]  36.00-37.00  sec  35.0 MBytes   294 Mbits/sec    0    976 KBytes       
[  5]  37.00-38.00  sec  38.8 MBytes   325 Mbits/sec    0   1005 KBytes       
[  5]  38.00-39.00  sec  38.8 MBytes   325 Mbits/sec    0   1.01 MBytes       
[  5]  39.00-40.00  sec  38.8 MBytes   325 Mbits/sec    0   1.06 MBytes       
[  5]  40.00-41.00  sec  36.2 MBytes   304 Mbits/sec    0   1.18 MBytes       
[  5]  41.00-42.00  sec  38.8 MBytes   325 Mbits/sec    0   1.35 MBytes       
[  5]  42.00-43.00  sec  37.5 MBytes   315 Mbits/sec    0   1.54 MBytes       
[  5]  43.00-44.00  sec  38.8 MBytes   325 Mbits/sec    0   1.76 MBytes       
[  5]  44.00-45.00  sec  40.0 MBytes   335 Mbits/sec    0   2.00 MBytes       
[  5]  45.00-46.00  sec  31.2 MBytes   262 Mbits/sec    4    195 KBytes       
[  5]  46.00-47.00  sec  31.2 MBytes   262 Mbits/sec  137   1.87 MBytes       
[  5]  47.00-48.00  sec  38.8 MBytes   325 Mbits/sec    0   2.11 MBytes       
[  5]  48.00-49.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
[  5]  49.00-50.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
[  5]  50.00-51.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
[  5]  51.00-52.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
[  5]  52.00-53.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
[  5]  53.00-54.00  sec  32.5 MBytes   273 Mbits/sec    0   2.11 MBytes       
[  5]  54.00-55.00  sec  33.8 MBytes   283 Mbits/sec    0   2.11 MBytes       
[  5]  55.00-56.00  sec  32.5 MBytes   273 Mbits/sec    0   2.11 MBytes       
[  5]  56.00-57.00  sec  31.2 MBytes   262 Mbits/sec    0   2.11 MBytes       
[  5]  57.00-58.00  sec  31.2 MBytes   262 Mbits/sec    0   2.11 MBytes       
[  5]  58.00-59.00  sec  32.5 MBytes   273 Mbits/sec    0   2.11 MBytes       
[  5]  59.00-60.00  sec  37.5 MBytes   315 Mbits/sec    0   2.11 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec  1.65 GBytes   236 Mbits/sec  379             sender
[  5]   0.00-60.00  sec  1.64 GBytes   235 Mbits/sec                  receiver

iperf Done.

```

See also my post in [OpenWRT Forum](https://forum.openwrt.org/t/linksys-wrt1900acs-v2-terrible-wifi-performance-with-android-google-pixel-on-18-06-0-release/19070)

makedir, 2018-08-19:
still no word from the devs...

chris-dk, 2018-09-08:
Unfortunately I am facing the exactly the same problem ("speed drops reducing the average speed") with my setup:
WRT1200AC v2 running OpenWrt 18.06.1 and Lenovo T480s (Intel 8265) running Fedora.
It is a pitty to not have the full speed available. 

Are there any news for future development or any plans to fix this problem?

lucize, 2018-09-26:
I tested an asus laptop with this wifi client on wrt3200acm, I got some disconnects from the router and in dmesg I have this

```
[70753.348886] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4355 ieee80211_rx_napi+0x160/0x9cc [mac80211]
[70753.359994] Rate marked as a VHT rate but data is invalid: MCS: 10, NSS: 1
[70753.366910] Modules linked in: pppoe ppp_async l2tp_ppp cdc_mbim qmi_wwan pppox ppp_mppe ppp_generic nf_conntrack_netlink nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm dm9601 cdc_subset cdc_ncm xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard usbnet slhc rfcomm r8152 nfnetlink nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack
[70753.438279]  iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hidp hci_uart crc_ccitt cdc_wdm btusb btmrvl_sdio btmrvl btintel br_netfilter bnep bluetooth hid evdev input_core mwlwifi mac80211 cfg80211 compat cryptodev ip6t_rt ip6t_frag ip6t_hbh ip6t_eui64 ip6t_mh ip6t_ah ip6t_ipv6header ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_gre ip_gre gre l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 ecdh_generic sha256_generic jitterentropy_rng
[70753.510107]  drbg md5 kpp hmac echainiv ecb cmac cbc authenc gpio_button_hotplug mii
[70753.517912] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.71 #0
[70753.523944] Hardware name: Marvell Armada 380/385 (Device Tree)
[70753.529902] [<c010f5bc>] (unwind_backtrace) from [<c010b2bc>] (show_stack+0x10/0x14)
[70753.537683] [<c010b2bc>] (show_stack) from [<c0606b48>] (dump_stack+0x88/0x9c)
[70753.544939] [<c0606b48>] (dump_stack) from [<c0121f00>] (__warn+0xe4/0x100)
[70753.551932] [<c0121f00>] (__warn) from [<c0121f54>] (warn_slowpath_fmt+0x38/0x48)
[70753.559484] [<c0121f54>] (warn_slowpath_fmt) from [<bf2b7ed0>] (ieee80211_rx_napi+0x160/0x9cc [mac80211])
[70753.569149] [<bf2b7ed0>] (ieee80211_rx_napi [mac80211]) from [<bf3264d8>] (pcie_rx_recv_ndp+0x7e4/0xd54 [mwlwifi])
[70753.579561] [<bf3264d8>] (pcie_rx_recv_ndp [mwlwifi]) from [<c0126368>] (tasklet_action+0x94/0x10c)
[70753.588648] [<c0126368>] (tasklet_action) from [<c0101628>] (__do_softirq+0xe0/0x240)
[70753.596513] [<c0101628>] (__do_softirq) from [<c01266e4>] (irq_exit+0xd4/0xe4)
[70753.603772] [<c01266e4>] (irq_exit) from [<c015d79c>] (__handle_domain_irq+0x9c/0xac)
[70753.611638] [<c015d79c>] (__handle_domain_irq) from [<c0101464>] (gic_handle_irq+0x5c/0x90)
[70753.620025] [<c0101464>] (gic_handle_irq) from [<c010bf4c>] (__irq_svc+0x6c/0x90)
[70753.627539] Exception stack(0xc0a01f50 to 0xc0a01f98)
[70753.632613] 1f40:                                     00000001 00000000 00000000 c0114c20
[70753.640826] 1f60: ffffe000 c0a03c74 c0a03c28 00000000 00000000 414fc091 00000000 00000000
[70753.649040] 1f80: c0a01f98 c0a01fa0 c010847c c0108480 60000013 ffffffff
[70753.655684] [<c010bf4c>] (__irq_svc) from [<c0108480>] (arch_cpu_idle+0x34/0x38)
[70753.663115] [<c0108480>] (arch_cpu_idle) from [<c0154600>] (do_idle+0xdc/0x19c)
[70753.670456] [<c0154600>] (do_idle) from [<c015491c>] (cpu_startup_entry+0x18/0x1c)
[70753.678061] [<c015491c>] (cpu_startup_entry) from [<c0900d48>] (start_kernel+0x46c/0x478)
[70753.686301] ---[ end trace c765acf18d74c432 ]---

```

 OpenWrt SNAPSHOT, r8119-87d7a596ec
wifi drivers from `Fixed hidden SSID problem of 88W8997. `

diizzyy, 2018-10-19:
Have in mind that there are a multitude of environmental factors that can and will affect your wireless performance which makes debugging hard apart from obtaining the exact same hardware if it's even possible.

For what it's worth, I'm having zero issues with various phones, laptops and consoles and they're getting at least ~200mbit depending whether they're 11n or 11ac (5Ghz) if there's good signal strength.

A few devices that works fine (I don't know specs of all devices used) - WRT3200ACMs running  OpenWrt master
miniPCIe AR9382 (Windows 7+Linux, 11n), Intel 9260 (Windows 10, 11ac)
Qualcomm Snapdragon 615/617/625/810 (11n and 11ac)
Playstation 4 (low signal strength, doesn't seem to perform any slower than with my old TL-WDR3600)

In my personal experience older Intel cards (6xxx) and (8xxx) have poor reception and can be quite unstable irregardless of radio used in AP (both on Windows and Linux) while they may occasionally work fine but it's rare. In general I'd say that older hardware irregardless of brand works quite poor compared to current gen or previous.

yuhhaurlin, 2018-10-31:
Air to air the throughput should be divided by 2.

makedir, 2018-10-31:
NO WAAAAA : O NOO WAAY :O I get about 12-22mb/s throughout, it just randomly drops by 50%, even goes to zero sometimes. And 22mb/s isnt (866/2=433mbit), it is something like 180mbps.

diizzyy, 2018-10-31:
@makedir 
Snarky comments doesn't help at all, in fact it will make people care even less about your issues.

makedir, 2018-11-01:
What issue? It is not my issue anymore. I bought a proper AP months ago and disabled the wifi of the WRT. Marvell dev team is incompetent. but I guess they just develp the wifi driver in ther free time it seems so it is not really a complain. Wifi never worked on these routers, and if they changed a driver, they brought in 10 new bugs.

s-pimenta, 2018-11-01:
As I said before, they are not taking the issues seriously, even seen people offering MONEY to get this DONE quickly, but seems no one cares at Marvell... 

It's a shame having a good hardware with so poor software support...

swg0101, 2018-12-16:
On my WRT1900ACS with the Intel 8260, the November (and December) drivers resulted in an unusable 5Ghz connection. Maximum throughput was about 2Mbps (with peaks to 5-10Mbps), but as more apps get launched the connection will almost be completely unusable with not even pings going out. Opening web pages would then become an impossible task since most will timeout with either DNS resolution errors or just the request timing out completely.

Have reverted back to the driver dating 2018-10-29 and I am back to getting 300+Mbps on my 5Ghz... On dmesg I am getting quite a lot of messages regarding Start BA with my MAC address on it, so I am assuming this problem has something to do with block ACKs not completing properly or something the power saves not being implemented properly (since there are changes to U-APSD in the recent code changes)...

This is the commit I am on for the Makefile that does not have this issue:

PKG_SOURCE_URL:=https://github.com/kaloz/mwlwifi
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2018-10-29
PKG_SOURCE_VERSION:=382700ce5744fe80271c57a89c6589e767d91620
PKG_MIRROR_HASH:=7378b7d391aeec7a86b6548723911d21bc780ea84ceea858e910cc65a1d925c6

:~# opkg list | grep mwlwifi
kmod-mwlwifi - 4.14.88+2018-10-29-382700ce-1
mwlwifi-firmware-88w8864 - 2018-10-29-382700ce-1

tiagogaspar8, 2019-01-13:
After reading every comment here I just want to say some things.
First I have to say I face the same Issue, but I see it in file transfer, when sending files from my PC to my server I see speeds of 500+mbps suddently dropping to the same speed of around 40mbps, that should be a hint.
Second thing I have to say is that this issue is valid for only my Intel card (8265), as my phone, a OnePlus 6 can easily achive 500+ speeds without sudden drops.
Third thing, I have contacted Intel for them to try to solve this issue (?) and if @yuhhaurlin has anything for me to forward to Intel to help them solve this issue I'd be glad to do it. 😄

makedir, 2019-01-13:
@tiagogaspar8 There is only one solution: Buy a standalone access point and plug it into your router. This issue wont ever be resolved. I also already had contacted Intel about it 6 months ago. They ignored it and were no help too.

bbogey, 2019-01-13:
Could it be similar issue I had with OnePlus devices that it didn't follow up the advertised capabilities?

Especially not exceeding the packet size
Max MPDU length 3895

One way to check this is to setup another AP as a sniffer (monitor interface) what can handle larger packets, e.g. ath10k.

tiagogaspar8, 2019-01-13:
@makedir I have already spent too much money on this router and every other device achieves very high speeds in wireless so I'd rather replace the wireless card on my PC with another one.
Intel has been nagging me with this issue as I don't answer fast enough and they want to solve this issue as fast as possible. Now, I need to know if this problem we're all having is with the intel driver or with the mwlwifi firmware. Maybe the Intel driver has a bug that other AP's "ignore" and mwlwifi just can't deal with it and forces those sudden drops (?)
I'm willing to test anything @yuhhaurlin wants to help fix everything.

ryan-dev, 2019-01-13:
I gave up on the router and sold it on Amazon and bought a Linksys Velop
AC6600 three node system to replace the broken WRT3200ACM wifi.

There is more wrong with the mwlwifi than ESP and Intel card compatibility.

Once I switched, performance has been tenfold better all the time, plus
each node can be directly wired -:)

diizzyy, 2019-01-13:
fwiw, I have Intel 9260 clients running just fine off 5Ghz (Windows 10)

tiagogaspar8, 2019-01-13:
@diizzyy really? I'll be thinking about buying that card. Thanks!

makedir, 2019-01-13:
So hard to read? I said the opposite: https://github.com/kaloz/mwlwifi/issues/307#issuecomment-406100552

diizzyy, 2019-01-13:
@makedir 
Different radios, WRT1200AC vs WRT3200ACM

Kherby, 2019-01-29:
So I guess this problem will never be fixed?

makedir, 2019-01-29:
> So I guess this problem will never be fixed?

Correct. I advice you to sell the WRT3200, then buy a WRT1200 (you get them really cheap, it has the same CPU power than the 3200), and then buy a stand alone access point.

BrainSlayer, 2019-01-30:
the wrt3200 has a higher cpu clock speed

infabo, 2019-04-08:
> @makedir
> Different radios, WRT1200AC vs WRT3200ACM

I have both the WRT1200 and WRT3200. Although they have different radios - both WIFI suck really hard. I always had issues with the 1200 and thought with an upgrade to WRT3200 all problems will disappear. Haha ROFL. The wifi of WRT3200 is even **worse**! I could not believe it. This damn router has 2 additional antennas and performs so badly. It was a true downgrade. Haha, a downgrade from a troublesome WRT1200. One could call it disaster.

tiagogaspar8, 2019-04-26:
@yuhhaurlin any updates about this issue?

makedir, 2019-04-26:
> @yuhhaurlin any updates about this issue?

lol. How many times? It wont get fixed. Do yourself a favor and buy a standlone AP, something like a Ubiquiti.

tiagogaspar8, 2019-04-27:
I see what you mean, but I already bought this router, It works so well for me, there are no routers in the market that perform as well as this one besides this Intel issue, at least for the same price, so I'm here to hep whoever is building this driver to fix it. I would do it myself but I ave no idea how to....

PS: @yuhhaurlin I don't know if I should be saying this to you or to @kaloz but there seems to be a issue with the WMM queues in that some of them are swapped, and that makes that priorities aren't correctly assigned (?)
As I've said before I'm willing to make the needed tests to solve these issues.

makedir, 2019-04-27:
> It works so well for me,

lel. Dont you see your mistake here? You even gave yourself the correct answer. Never buy a router with build in wifi.

[6 more replies]