kaloz/mwlwifi #309: WRT3200ACM <> WRT1900AC in WDS mode

gufus, 2018-07-13:
Test units

WRT3200ACM
WRT1900AC

Wifi driver (both units)
10.3.8.0-20180615

Encryption (both units)
WPA2 PSK (CCMP)

WRT1900AC in WDS AP
LED  Kernel Version	4.4.92

Mode: Master | SSID: gypsynet_5g
BSSID: B4:75:0E:FE:A1:84 | Encryption: WPA2 PSK (CCMP)
Channel: 161 (5.805 GHz) | Tx-Power: 30 dBm
Signal: -51 dBm | Noise: -83 dBm
Bitrate: 1300.0 Mbit/s | Country: CA

WRT3200ACM in WDS Station
DD-WRT Kernel Version	4.9.110

Issue in both 5g and 2g

WRT3200ACM in WDS Station keeps on going DOWN then quickly comes back UP

BrainSlayer, 2018-07-14:
QCOM / Atheros WDS AP works with WRT3200ACM WDS STA on dd-wrt. but never tried wrt1900ac

eduperez, 2018-07-14:
See issue #242, you probably need to dissabled group key handshake. I'm currently using WDS-AP on a WRT3200, and it's perfectly stable. I can provide config files and logs, if needed.

gufus, 2018-07-14:
@2yuhhaurlin

BTY

Using 2 1900ac(s) and Wifi driver (both units) 10.3.8.0-20180615, WDS STA  stays up.

see here...
http://www.gypsy-designs.com/671.jpg

yuhhaurlin, 2018-07-14:
I close this one, there is another issue to track WDS group key problem for WRT3200ACM and there is a way to work around it.

closed 2018-07-14 by yuhhaurlin