kaloz/mwlwifi #313: Hi pings on 5 ghz

tapper82, 2018-07-31:
Hi I have a wrt3200acm as my mane router and am using a wrt1900ac v2 as a WiFi repeater to get internet to a pc i have up stairs. If i set it to use the 2.4 bandmy pings are low like 20 30 but if i use the 5 ghz radios to form the link the pings are 100 +. The wrt3200acm is running openwrt OpenWrt SNAPSHOT r7540 and the wrt1900ac v2 is running gargoyle.

yuhhaurlin, 2018-08-01:
If application is running all right, it should be all right.

closed 2018-08-01 by yuhhaurlin

BrainSlayer, 2018-08-01:
@tapper82. even 20 and 30ms is very high. what i miss here are signal levels you get. consider that 5 ghz has much higher signal loss / absorbtion effects than 2.4 ghz depending on the material they have to pass. concrete is a total no go. standard brick walls might work (except if they are wet)
if you use the antennas in standard direction you may also have the problem that the signal isnt transmitted in vertical direction. they form more like a horizontal disc sourrounding your router  which isnt a good idea if you transmit your signal upstairs. i hope this is a hint for you

yuhhaurlin, 2018-08-01:
Maybe you can disable tx amsdu for WRT1900AC, this is the only high ping issue I known.

echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
or 
echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu

This high ping won't affect normal applications.

tapper82, 2018-08-01:
@yuhhaurlin  Were do i paste those lines? At the command line? Pings like that are not good.

yuhhaurlin, 2018-08-01:
Issue via command line. You only need to do this for WRT1900AC. WRT3200ACM does not have this kind of problem.

eduperez, 2018-08-01:
My two cents:
* Main router: WRT3200ACM running LEDE 17.01.4 + mwlwifi 10.3.8.0-20180716
* Wifi extender: WDR4300 running stock LEDE 17.01.5
* Both connected using WDS, on the 5Ghz channel, no AMSDU tinkering
* Wired and wireless devices connected to main router and extender
* PING times, from a wired client on the main router, to a wired or wireless client on the extender, are around 3-4 ms.

I can perform more tests or provide config files, if required.

yuhhaurlin, 2018-08-01:
@eduperez Thanks for your information. As I mentioned, WRT3200ACM is all right. I think if @tapper82 disable tx amsdu of WRT1900AC, maybe high ping is gone.

BrainSlayer, 2018-08-01:
@yuhhaurlin : that wont explain why amsdu aggregation is a problem. i mean disabling it will decrease the performance. is there a vendor specific compatiblity problem? i mean he connected to marvell chipsets even if different types

yuhhaurlin, 2018-08-01:
Maybe I will check it later.

@tapper82 Can you try to ping router to check if the ping is all right?

tapper82, 2018-08-08:
Still geting Hi pings. AMSDU  is off on the wrt1900ac-v2. I am not using WDS the wrt1900ac is running gargoyle and the wrt3200acm is running OpenWRT 18.06. The Gargoyle build that the wrt1900ac-v2 is running is baste on OpenWRT 18.06 so should be the same driver.

yuhhaurlin, 2018-08-08:
Please ping to router.