kaloz/mwlwifi #314: [wrt1200ac] 2.4ghz connection very unstable

kbios, 2018-08-01:
This happens with openwrt 18.06, but also with 17.01.4.
I'm using the 2.4g radio in client mode to connect to another ap; my pc is connected to the 1200ac with a wire. The connection is very erratic, it can work ok for an hour or two but eventually it starts pausing and choking: every few seconds downloads go to 0b/s and stay there for about 10 seconds before continuing, ping shows more than 50% packet loss. Sometimes the connection hangs completely and I have to restart the router.
If I connect directly to the upstream ap everything works well.
I've also tried the latest trunk builds of mwlwifi but they are worse: it disconnects every few seconds even right after a reboot.
If you need more information for debugging just tell me

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info 

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20180330
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:fe:e2
2g: enable
5g: disable
antenna: 2 2
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
radio: enable
iobase0: e1100000
iobase1: e1280000
tx limit: 768
rx limit: 64
qe trigger number: 3896426

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/sta

mac address: 00:19:70:44:2a:4c
aid: 1
ampdu: true
amsdu: false
wds: false
ba_hist: disable
ht_cap: 0x100c, ampdu: 03, 06
rx_mask: 0xff, ff, 00, 00
rx_bw: 0, rx_nss: 2
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000000001

```

yuhhaurlin, 2018-08-02:
Did you use relays?

kbios, 2018-08-02:
No, it's a normal NAT setup (upstream router uses 192.168.1.x, wrt1200ac uses 192.168.3.x)

yuhhaurlin, 2018-08-02:
It looks like 88W8864 has issue for latest mwlwifi driver. I will check these issues related to 88W8864 later. Thanks.

dbischof90, 2018-08-07:
I might have the same problem, my 2.4GHz is currently very close to unusable. Is there any progress?

howl, 2018-08-23:
I think this one should be closed, seems to happen with a old version and it's now a more recent one included in OpenWrt 18.06.1.

yuhhaurlin, 2018-08-24:
@howl Thanks. I close this issue.

kbios, 2018-08-29:
This still happens for me with openwrt 18.06.1. Should I open a new issue or is it better to reopen this one?

howl, 2018-08-29:
I think you could reopen it, I suggested the close because another similar bug where solved by using the latest stable.

kbios, 2018-08-30:
Ok thanks, but unfortunately I don't have the rights to reopen it :(

kbios, 2018-09-08:
If it helps with debugging, I'm noticing that the main problem is that the driver reacts poorly to collisions/congested areas: if there's only my wifi connection it can work ok for days, but if I put it in a place with other busy connections it locks up every few minutes or even seconds in the worst case.

kbios, 2018-09-10:
Sorry, but even under noise it is not acceptable that the driver locks up and I have to reboot the router to get it working again. Don't you agree?

yuhhaurlin, 2018-09-10:
Can you show me the log to make sure it is the same as other reported issues related to 88W8864?

yuhhaurlin, 2018-09-10:
I hope I can find command timeout in the log?

kbios, 2018-09-11:
[openwrtlog.txt](https://github.com/kaloz/mwlwifi/files/2372480/openwrtlog.txt)
Here's the log after it locked today, every time there are those "change" messages and authentication with the other AP times out (the 2.4G radio is wlan1/phy1)

jbsky, 2018-09-19:
Hey,

For the WRT1900ac v1, 5Ghz band completely unstable performance level (10MBit + some high ping).

JB

lyssdod, 2019-01-20:
Hi, I'm also experiencing issues with 2.4ghz in AP mode – network randomly disappears and never comes back until reboot; 5ghz "works" fine (though also occasionally disappears once per few days, but then comes back). I was running stock openwrt 18.06.1, then upgraded to recent [prebuilt version](https://github.com/eduperez/mwlwifi_LEDE/releases/tag/e5e0700), which made it even worse.

```
# uname -a
Linux wrt 4.14.63 #0 SMP Wed Aug 15 20:42:39 2018 armv7l GNU/Linux
```

```
# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info 

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181112
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: xx:xx:xx:xx:xx:xx
2g: enable
5g: disable
antenna: 2 2
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1a00000
iobase1: e1b80000
tx limit: 768
rx limit: 64
qe trigger number: 44395
```

kbios, 2019-07-04:
So I've finally managed to test a few versions of the driver and apparently the most stable for me is the one that comes with openwrt 18.06.1. It hangs every few hours but sort of works.
Later versions (18.06.2, 18.06.4) hang every few seconds and are unusuable...