kaloz/mwlwifi #318: WiFi not coming up if one network in Client mode, while other in Master mode (to extend range)

kristian, 2018-08-07:
Tested on a Linksys WRT1200AC (Marvell 88W8864 chip) using mwlwifi driver version 10.3.8.0-20180716 commit d9c8c59 for OpenWrt 18.06.0.

Configured one WLAN in Client mode, connecting to an existing WiFi network on radio0 (5GHz). Configured a [relay network](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration). Working fine for clients connected to the AP, or for the clients connected to the 2,4GHz WLAN on radio1, see:
[image]

When creating / enabeling a second network as Master, using the same or a separate SSID on radio0, to extend the WLAN range, radio0 / both networks fail to come up ("Wireless network not associated"), see:
[image]

I am thankful for any suggestion! :)

BrainSlayer, 2018-08-08:
thats what he did. the first vap is client, the second vap on radio 0 is ap

yuhhaurlin, 2018-08-08:
AP mode won't show "Wireless is not associated"!

BrainSlayer, 2018-08-08:
thats true in theory, but mode : master is "ap mode" in fact. might be a gui glitch in openwrt. curious

kristian, 2018-08-08:
Thanks a ton for looking at my issue @yuhhaurlin and @BrainSlayer!

> You can only have one client. For other virtual interfaces, it should be AP mode.
As @BrainSlayer said, that is what I tried to show in my screenshots. "BKA Ueberwachung 12" is the one virtual access point in client mode, whilst "BKA Ueberwachung 11" is in ap mode, as you suggested. The idea is that the first is connecting to my internet gateway, while the other one should extend the range of my existing 5GHz WLAN. However:

> AP mode won't show "Wireless is not associated"!

Both virtual interfaces do and also, in case I have them both enabled (what I showed in my second screenshot), both interfaces on radio0 won't come up showing the same "error message": Wireless is not associated.

I am also very open to open this to OpenWrt, but it would be great if you could point me into a direction of any log file, where to find out if it is a driver or OpenWrt issue. As far as I can tell my configuration should work (in theory) nevertheless, it doesn't, therefore I opened it here, instead of in OpenWrt.

Thank you so much for any help!

kristian, 2018-08-09:
Mhm, I still don‘t know why „Wireless is not associated“ is shown for both VAPs.

Other than that, I think I found the root cause to my problem though. To make it work I had to manually assign the same channel to my client mode VAP, than my main network „BKA Ueberwachung 12“ is running in. For me that makes kind of sense in the whole scheme of things, as the client mode VAP must communicate in exactly that channel to get a connection to my main network. What I don‘t understand though, is why setting the client VAP to „auto“ channel selection did not select the right channel in the first place, but just stalled to connect the APs.

Therefore my follow up question would be: Is it supposed to work like that ...
a) for this driver and / or
b) for other drivers / devices running OpenWRT?

For a): I would imagine this to be unintended behaviour by this driver, as if my main network channel changes, I would have to manually adapt my second APs channel. I would assume „auto“ channel selection to work just fine. If that is the case, any chance to reopen this issue @yuhhaurlin?

For b): I would imagine in case a client VAP *never* connects when choosing auto channel selection, also for other drivers / routers (so in case this is generally the expected behaviour), it should not be allowed to set it to „auto“ in LuCI in the first place, isn‘t it? @BrainSlayer, as I saw you are also contributing to OpenWRT, what do you think, does it make sense to address this in an OpenWRT issue as well? Thanks!

eduperez, 2018-08-09:
In OpenWrt, "auto" as a channel means "first available channel", and nothing else.

kristian, 2018-08-09:
Thanks, that clarifies it. So it‘ll never work in client mode I guess, as it must pick a channel which is already in use and therefore is contradictory to your statement.

eduperez, 2018-08-09:
But "available" does not mean "not in use" but "legal in your country". In the 2.4 GHz band, "auto" equals to "1"; in the 5GHz band, it equals to "36".

kristian, 2018-08-09:
Then, again, I don‘t understand why the VAP did not connect when it‘s set to „auto“ and I would consider it a bug. My main AP is running on channel 36. When I set the client AP to „auto“ it did not connect („Wireless is not associated“), now I set it to channel 36 fix and it connects flawlessly.

BrainSlayer, 2018-08-09:
@kristian i dont know if its a openwrt issue. i contribute code i wrote initially for dd-wrt. i'm not a official openwrt developer. i'm doing dd-wrt. for the repeater mode. what @eduperez is not true. in fact the client associates and signals the channel to the second vap for configuration. this is part of a modification in hostapd which is used by openwrt. but the point is maybe that openwrt does not take care about vht 80 and vht160 here. this wont work in openwrt in these modes. this is a issue i fixed in dd-wrt but it got never into openwrt. maybe this is the answer. so try to use 20 mhz or 40 mhz only

kristian, 2018-08-09:
Thanks @BrainSlayer. But unfortunately not really. I am exclusively using 40 MHz... So that can't be the root cause of the issue... Mhm strange! Anyways as said, fixing the channel settings seems to be a working for me.

BrainSlayer, 2018-08-10:
same answer. try 20 mhz first. oh and yes this can be very the root cause of the problem. if the secondary vap interface receives wrong channel informations (40 mhz has 2 channels. center and side channel) it simply wont work. so please do it an look if it works with 20 mhz. if it works then but not with 40 mhz, openwrt needs to be fixed. i remember that i fixed alot of hostapd regarding this center / side channel problem in repeater situations

Massimo-B, 2020-04-29:
On the Linksys WRT1200AC, using the Marvell 88W8864 802.11nac, I can't get an ap network enabled while another one is connected in client mode. I guess this is the same issue as mentioned here. Disabling the client connection, the ap network comes up.
Channel for the ap network is auto. I always see these lines in returning in the log:
```
Wed Apr 29 15:30:11 2020 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Wed Apr 29 15:30:11 2020 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Wed Apr 29 15:30:11 2020 daemon.err hostapd: Failed to set beacon parameters
```

eduperez, 2020-04-30:
@Massimo-B Choose a channel outside the DFS range in your region, and configure both devices to that channel.

Massimo-B, 2020-06-09:
On a mobile access point I have many client networks on different channels. So I would guess that the approach is to set both the client and mastered network to auto channel, no? Especially when using the travelmate package.

eduperez, 2020-06-09:
@Massimo-B What you need is a package called "travelmate".

gillg, 2020-10-09:
For information, about initial issue a solution seems available here ! https://github.com/jordi-pujol/wwanHotspot/
AP mode seems not work if client is not connected. So, in case of volatile mobile hotspot your AP wifi not works.

This daemon switch between AP only or AP+STA modes if STA could be connected or not.

eduperez, 2020-10-10:
> AP mode seems not work if client is not connected. So, in case of volatile mobile hotspot your AP wifi not works.

AP cannot work until STA is connected, because both must be on the same channel, and the STA channel is determined by the upstream AP.

gillg, 2020-10-11:
Hum... yes, you're true STA chanel is determined by AP and vice-versa (on OpenWRT configure the channel on any interface implies the change on all interfaces), but it's not the problem. They can works together at the same time, but if you configure an AP + a STA, it implies that STA **must be** connected to activate the AP. It seems more a feature than an hardware limitation, so if only AP interface is enabled the AP works.

See example, I have disabled completly my STA interface, and AP is working.
[image]

Now I enable STA interface, but my "mobile hotspot" Guifwi (Android smartphone) is not enabled.
So AP didn't works anymore... (Case of this issue)
[image]

Now I enable my mobile hotspot and AP + STA are automaticaly working !
[image]

So wwanHotspot projects seems have the goal to make AP active even if STA is not connected.