kaloz/mwlwifi #321: Some 5 GHz channels missing

howl, 2018-08-25:
At least in Europe there are some channels missing. Channels 144, 149, 151, 153, 155, 157, 159, 161, 165, 169 and 173 can be used without DFS radar scan with maximum power of 25 mW about 14 dBm.

Also channel 144 can be used at the same time with DFS making it possible to have one more 80 Mhz band in union with the actual 132, 136 and 140. With DFS 144 can operate at 500 mW about 27 dBm.

Source https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax)

I wanted to report the before by forget it till this OpenWrt commit that enables some, but not all these channels for mwl8k https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=672c430d6e2a1dbc5f99fd5d8d9d47e6af02ad5e also ath10k nowadays let use the non DFS channel despite it fails to let 144 work at all, but that is a report for another place.

lantis1008, 2018-08-25:
The patch linked for mwl8k only enables the upper non-DFS channels. These channels already work perfectly fine on mwlwifi. I was running an 80mhz channel at 149+ until switching to 160mhz at 36+. 

Also, adding 144 as a non-DFS channel shouldn’t add any benefits. It isn’t contiguous with the upper range so can’t be used as a channel extension. 

If you want to see these channels supported, you should start with the kernel regulatory database. Even though it is not referenced by these routers specifically, it will open up many others. 
See GB here which has no reference to the upper band. In theory, you could do a listing at the lower power, but you would need to plead your case to the maintainer and prove the legality. 
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n483

For fixing it on these routers, you’ll need to wait until the outcome of #280 to see what you can and can’t do.

yuhhaurlin, 2018-08-26:
Please report problems related to mwlwifi here.

closed 2018-08-26 by yuhhaurlin

howl, 2018-08-26:
@lantis1008 EU router here, none of these channels can be selected. 144 as non DFS is per regulation in the regulatory rules not my own invention.

@yuhhaurlin with other drivers I can select some of these channels but no with mwlwifi. Could you point me where the problem is if it's not in mwlwifi? Thanks in advance.

yuhhaurlin, 2018-08-26:
Allowable channels are not controlled by mwlwifi. Regulatory database will decide allowable channels.

yuhhaurlin, 2018-08-26:
BTW, mwl8k is not mwlwifi.

howl, 2018-08-27:
I know, mwl8k is not mwlwifi, just was an example of recent change to support these channels, it's a fact that me and your can understand each other.

So if I understand correctly, updates in the FCC or ETSI, or as you wish regulatory organization, won't never be applied in devices that have regulatory rules hardcoded in EPROM. Am I wrong?

yuhhaurlin, 2018-08-27:
It is correct for WRT3200ACM. After issue #280 is done, this limitation will be removed.

howl, 2018-08-28:
@yuhhaurlin Thanks, that bug is the one blocking for me.

@lantis1008 @yuhhaurlin As the GB comment, I thought FR and ES had same regulatory rules but there are some differences, same with other European countries, there are some little changes per country so in practice EEPROM regulatory rules are not a good idea, no only for updates in the rules but also because couldn't be EU router version, there should be as many as countries to really respect rules.

lantis1008, 2018-08-28:
I can't speak to what is stored in the EEPROM. I don't know.
And you don't need to convince me that its a bad idea. I'd prefer it was done the old way. But the FCC mandated this and it is the way it is if we want open source routers to still be produced. Ultimately yuhhaurlin can't control that either. But he has agreed to try and make the system we have slightly better through #280

howl, 2018-08-28:
@lantis1008 that's the point I never understood, FCC rules are for certain territories, why their decisions should affect others like ETSI, TELEC or KCC (Kentucky Cooked Chicken)?

yuhhaurlin, 2018-08-28:
This is what fix for issue #280.

aaron1959, 2018-08-28:
It has to do with people here making up their own idea of the F.C.C. rules, rather then complying with then ONLY as needed, since these folks can't figure out how to code it properly they blame the rules lol.

howl, 2018-08-28:
@aaron1959 first of all I don't have nothing to done with FCC cause I'm not living in a place they rule, but who knows if it could be possible in the future, or, if a could see a router in sale in a place where FCC rules affect and you can't use it where you live. The real problem is the so restrictive the US laws are and that they don't serve to their propose as if someone have the wish to emit illegally is very probable that he/she have the knowledge to use a homebrew device.

Indeed there are other ways to do this without restricting devices, Motorola has or at least had their mobile phones to use the gps location to set the regulatory rules automatically (it's annoying when you can see networks in channels 12 or 13 in Europe but it works) or the one I like the most, the device "ear" the surrounding networks to set the regulatory rules (the problem here is that if someone set up a wrong country in his network it can mess nearly devices).

About coding, it's easy to circumvent the blocked country code by mwlwifi, even if you don't know to code, because the patch to do it is there and it's not to difficult to apply it in the project you want.

PD: I do not blame the regulatory rules, I blame the laws that enforce regulatory rules making the device developers to do things that have no sense. For example, one main difference between ETSI and other regulations is that ETSI is a recommendation, not a regulation perse like FCC, thats why in Europe the countries differ in they regulations because they tend to follow ETSI but adjust it to their specific needs. And of course, as needed, if I dind't have a use for the sdio wifi in the WRT3200ACM or I didn't want to set the max power that my country allows I don't think I had spotted this issues. I only wish that Europe doesn't start to make US style laws, but unfortunately the laws tend to protect or benefit big companies and not to reflect the people needs, but that is a topic that start to get too far from the original one.

BrainSlayer, 2018-08-29:
@howl in europe the regulations wont differ. they are harmonized by the etsi standards. all europe has the same harmonized standard. no difference in power or frequency set. only the BFWA (broadband fixed wireless access) channels may differ in some countries, but they arent supported by mwlwifi in any way but in fact europe has now also laws which have the same restrictive policy like the usa, but its not enforced right now. so router locking etc. is still not a big deal here. the bad thing is that open projects like openwrt and dd-wrt are indirect responsible for these laws. remember the fcc paper where vendors where questioned how they prevent installing of third party firmwares like "dd-wrt" etc. this was making me really upset.

BrainSlayer, 2018-08-29:
and this paper is part of the certification.

howl, 2018-08-29:
@BrainSlayer I thought the same you say about ETSI, but it's not. Practical example, FR allows 5170 MHz to 5250 MHz at maximum power of 20 dB (100 mW) where DE and ES allow 5150 MHz to 5250 MHz at maximum power of 40 dB (200 mW), it's the twice of power. Also there is a difference in the wide, as FR sets in a 80 MHz band and DE and ES are setting a 100MHz, despite they also allow only a maximum of 80 MHz bandwidth in that frequencies.

Also is possible to find per country regulations that are the rules countries follow instead of ETSI, ETSI is just a recommendation here.

So I guest we coincide at 100% in the law part, that's why I said in US the laws tend yo be unfair protective methods for big companies and not the people interest.

BrainSlayer, 2018-08-29:
nope. 5150 - 5250 is correct. but power is 100mw. 5470-5725 allows 1000mw with tpc and dfs control. without tpc its 500mw. but thats no big deal. no router has that power. the antenna makes the point here and a antenna can be manually mounted and changed without buying different hardware. both countries. france and germany allow vht160 in 5490 - 5710. but in lower band its vht 80 for both countries. but what i want to say is its easy to have a subset for all countries. you have small differences but the router output power isnt a point here since the radio cannot do such high output power without having a directional antenna

howl, 2018-08-29:
@BrainSlayer yes DE only 100 mW, thought both DE and ES where similar, ES allows the double https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n442 Indeed that is the only difference nowadays from your and my country regulation.

It's a lot of power to achieve the maximum power but for example WRT3200ACM allows not that high power but more than it can be set nowadays.

Taking advantage from this offtopic closed issue, I wanted to congratulate all the efforts you have done with dd, routers and networking in general.

BrainSlayer, 2018-08-29:
@howl there arent many devices which go over 100 mw. except for ubiquiti and ubiquiti is violating all possible regulations all over the world. this company is also the dark side of the technology in many other ways

tedascii, 2020-10-31:
> @howl there arent many devices which go over 100 mw. except for ubiquiti and ubiquiti is violating all possible regulations all over the world. this company is also the dark side of the technology in many other ways

disagree.