kaloz/mwlwifi #322: No device able to connect to 5GHz Wifi if choosing channel above 48 (> 5,250 GHz)

kristian, 2018-09-05:
Hello,

I have a very weird issue with my Linksys WRT3200ACM router, which I even take a hardware failure into consideration. When choosing any channel above channel 48 (> 5,250 GHz), also to increase TXPOWER, no devices will any longer find, nor can connect to the WiFi. I tried it multiple times, different channels (52, 64, 100, 124), different client devices (iPhones, ThinkPad), restarted the router multiple times, same result: If choosing a WLAN channel above 48, no device is able to connec to the WiFi. Isn't that strange?

As soon as I switch back to a channel <= 48, devices immediately connect to the AC network. The radio0 shows as "up" in LEDE, so I have absolutely no clue what is happening...

Any help is higly appreciated! Thanks so much & best regards,
Kristian

derosier, 2018-09-05:
You didn't specify your regulatory environment. As your profile states Germany, I presume EU rules. All channels between 48 and 144 (inclusive) are DFS channels, and 149-173 are defined for 25 mW operation only.  Generically speaking of course - I don't know what the regulatory tables on a EU WRT3200ACM might be.  My guess is that your router has the DFS channels disabled for some reason. Out-of-the-box (ie before you flashed it with LEDE), did any of these channels work in your area?

If you want to learn more, this is a start: https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax)

derosier, 2018-09-05:
You may know this, but in case you don't: Channels above 48 are DFS channels in Germany. They're special. Basically, the rule is if the AP hears radiation on the channel, they're not supposed to use it for a period of time. In some cases, to avoid bothering, the manufacturers will just block out those channels.  Depends on the device in question and the environment.  Again, speaking very generally.

You might have DFS channels disabled (this is often a toggle setting on APs). Or perhaps the WRT3200ACM won't do DFS in the EU.  Or you have weather radars or other signals on the channels that are keeping it from working.

I do know there's been several issues with mwlwifi opened related to DFS and regulatory issues. I suggest you do an issue search and see if you can find some help there.  For example: https://github.com/kaloz/mwlwifi/issues/173

I also recall an issue with having radio2 enabled screwing up the regulatory behavior of the device.

I don't really know your environment or LEDE or mwlwifi well enough to give you specific suggestions. Perhaps someone else here with specific knowledge of using an WRT3200ACM in the EU will help.

kristian, 2018-09-05:
Thanks so much Steve!

Seems really resetting to the default radio0/radio1/radio2 countries seems to have fixed the issue!

```
uci delete wireless.radio0.country
uci delete wireless.radio1.country
uci delete wireless.radio2.country
uci commit
reboot
```

What a strange issue!! Thanks so much Steve! Saved my day. =)

closed 2018-09-05 by kristian

yuhhaurlin, 2018-09-05:
For DFS channel, you need to wait CAC (60 seconds) to let the AP work if no radar signal is detected.

kristian, 2018-09-06:
Thanks @yuhhaurlin! Appriciate your support. 👍 Also that is very valuable information. But when there was a country set (I changed the defaults), even after removing phy2 as suggested, the radio0 showed "up", but the signal wansn't visible by any device. As said only after removing the country setting and rebooting the device, the WiFi turned up on the devices... This just I think underlines your suggestion on the README.md to NOT mess about with the country settings, but let mwlwifi choose it for you in order to avoid any problems on the WRT3200ACM. As said, even removing phy2 didn't help in my case.

One issue I see: Not everyone who is going to flash LEDE will reach down to the WiFi drivers if doing settings like changing the country code. So the issue really drove me crazy for about an hour or two, as it was really strange to see the adapter as up, but still no device able to connect. I really thought iPhone or the ThinkPad is not supporting the 2nd subband or something, so this hit me like a truck.

Thanks again & best regards!