kaloz/mwlwifi #323: lede17.01源码编译mwlwifi-10.3.8.0-2018-09-06出错

xiasonming, 2018-09-09:
很抱歉我不太懂英文，只能用中文向各位请求帮助。
我用的系统是ubuntu1604，我在lede17.01源码编译mwlwifi-10.3.8.0-2018-09-06时出错，具体如下：

make[4]: Entering directory '/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.153'
  CC [M]  /home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.o
/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.c: In function 'mwl_vendor_cmd_set_bf_type':
/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.c:44:7: error: too many arguments to function 'nla_parse'
  rc = nla_parse(tb, MWL_VENDOR_ATTR_MAX, data, data_len,
       ^~~~~~~~~
In file included from /home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211-backport/net/netlink.h:3:0,
                 from include/net/rtnetlink.h:5,
                 from include/net/sch_generic.h:12,
                 from /home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211-backport/net/sch_generic.h:3,
                 from include/net/pkt_sched.h:7,
                 from /home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211/net/codel.h:47,
                 from /home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211/net/mac80211.h:24,
                 from /home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.c:18:
include/net/netlink.h:238:5: note: declared here
 int nla_parse(struct nlattr **tb, int maxtype, const struct nlattr *head,
     ^~~~~~~~~
scripts/Makefile.build:269: recipe for target '/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.o' failed
make[5]: *** [/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/vendor_cmd.o] Error 1
Makefile:1421: recipe for target '_module_/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06' failed
make[4]: *** [_module_/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06] Error 2
make[4]: Leaving directory '/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.153'
Makefile:61: recipe for target '/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/.built' failed
make[3]: *** [/home/xsm/lede/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.8.0.git-2018-09-06/.built] Error 2
make[3]: Leaving directory '/home/xsm/lede/openwrt/package/kernel/mwlwifi'
package/Makefile:105: recipe for target 'package/kernel/mwlwifi/compile' failed
make[2]: *** [package/kernel/mwlwifi/compile] Error 2
make[2]: Leaving directory '/home/xsm/lede/openwrt'
package/Makefile:101: recipe for target '/home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/xsm/lede/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xsm/lede/openwrt'
/home/xsm/lede/openwrt/include/toplevel.mk:205: recipe for target 'world' failed
make: *** [world] Error 2

eduperez, 2018-09-09:
See my comments at commit d9daa1dfd418a0a790188b805c95c99dea23f1bb.

yuhhaurlin, 2018-09-09:
Your kernel is older, you need to remove the last parameter from this function (nla_parse()).

closed 2018-09-09 by yuhhaurlin