kaloz/mwlwifi #325: Problem for getting 160mhz work in Canada.

aqsandy, 2018-10-10:
Hello, I am currenting using 20180920 making for my wrt3200acm. However, when I was checking the regulation for my device, it shows:

` global
country 98: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5600 @ 80), (N/A, 23), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#1
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#0
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

`

I searched up the regulation for Canada, it should be as same as US. 
When I apply 160 MHz on band 149 with power 30 dbm. Phy0 doesn't boost even I reboot the device. After reboot I ran cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info, it gave me:

` driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180920
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x20
mac address: 60:38:e0:xx:xx:xx
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: disable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384
`

(I already enable the radio in luci but actually it didn't go up.)
Is there any solution for this issue or I just simply set up in a wrong way?

Looking forward to reply.

yuhhaurlin, 2018-10-10:
Please check README.

closed 2018-10-10 by yuhhaurlin

aqsandy, 2018-10-10:
>           Please check README.

I actually removed those two packages before I post the issue, but still no help.

aqsandy, 2018-10-10:
I removed the country code option in the config.

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option legacy_rates '1'
        option channel '149'
        option txpower '30'
        option htmode 'VHT160'

After the editing, I reboot the device, but I still can't make 5ghz working under 160mhz on band 149.
I ran  cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info, it gave

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20180920
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x20
mac address: 60:38:e0:b9:5f:42
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: disable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

Is there any setting error for my configuration? 

Also, the global country value on regulation table is normal now as CA

global
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#1
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

phy#0
country CA: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5600 @ 80), (N/A, 24), (0 ms), DFS
        (5650 - 5730 @ 80), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

but still no 160 MHz info on it.

yuhhaurlin, 2018-10-10:
You can set channel 36 160 MHz.

BrainSlayer, 2018-10-10:
have you checked your local crda db.txt file? maybe your version is very 
old and contains outdated rulesets