kaloz/mwlwifi #326: MOD : Fix in iw list

jbsky, 2018-10-14:
Hello;

I would like to make a small comparison about a simple function like iw list.

[image]

When I type this command to display the map information, I am surprised to see 0. I circled in red to help you understand.

Below is a capture from one of your commercial competitors, and we can see that it is not at 0.
[image]

Can you make sure that you display a value that reflects the real capacity of the wifi card?

With kind regards.

Jbsky

ryan-dev, 2018-10-15:
> I can reproduce the issue, using OpenWrt 18.06.1, and latest mwlwifi
> available (3520811
> ).

I can confirm as well, this has been an issue for awhile but just ignored
it.

Would be nice if it actually did report back max speeds for the wifi driver.

jbsky, 2023-11-20:
Already fixed

closed 2023-11-20 by jbsky