kaloz/mwlwifi #332: WRT12000 V1 txpower not adjustable

schm0, 2018-11-20:
Hi!

I have the feeling my wrt1200 v1 always uses the maximum tx power.
When i set the tx power to something really low, like 1mw, i still get full signal on my devices.
Distance is like ~3m and through 1 wall.
iw shows the correct tx power.
But this seems wrong to me.
The v1 version should not have a power table that is locked by eeprom?

kubrickfr, 2018-11-20:
This has been discussed ad-nauseum in the various issues here. For example: https://github.com/kaloz/mwlwifi/issues/291

This is not a technical problem, this is a problem with the interpretation of FCC rules

schm0, 2018-11-21:
No.
The wrt1200 v1 doesn't use a power table that is locked by eeprom.
So power levels should be adjustable.

Even if the wrt1200ac doesn't have eeprom locked power table.

yuhhaurlin:
Sorry. If device is programmed with EEPROM power table, power can't be changed.

What is the purpose of having a power table in eeprom, when it is not possible to adjust the power (down)?
So all v2 devices with eeprom locked power table are sending at max power all the time?
Seems like a clear bug to me.

yuhhaurlin, 2018-12-14:
Mwlwifi does not allow adjust power setting now.

closed 2018-12-14 by yuhhaurlin

schm0, 2018-12-14:
Isn't this a violation of FCC regulations?

yuhhaurlin, 2018-12-14:
There is request to set lower power level, however mwlwifi doesn't support it now.

kubrickfr, 2018-12-14:
> There is request to set lower power level, however mwlwifi doesn't support it now.

Is there an issue to track the progress of this request?

yuhhaurlin, 2018-12-14:
I think I have no time to implement it.

aaron1959, 2018-12-14:
It is a violation of F.C.C. rules to use minimum power necessary to make reliable connection. The language seems to be the issue they can't understand the F.C.C. doesn't demand a locked power table.

BrainSlayer, 2018-12-15:
a router without power reduction possiblity is really a mess. its part of the problem of the 2.4 ghz signal hell in our world.

kubrickfr, 2018-12-16:
one again @yuhhaurlin , I know it may not be in your power, but you'd be amazed with what people here could do if you get Marvell to release the firmware documentation. If you don't have the time to work on this project, that would really help.