kaloz/mwlwifi #333: Error when setting almost everything

tiagogaspar8, 2018-11-30:
This is the error I get:

```
Fri Nov 30 02:23:10 2018 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Nov 30 02:23:10 2018 kern.info kernel: [  388.962759] device wlan0 entered promiscuous mode
Fri Nov 30 02:23:15 2018 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Fri Nov 30 02:23:15 2018 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Fri Nov 30 02:23:15 2018 daemon.notice hostapd: wlan0: ACS-STARTED
Fri Nov 30 02:23:15 2018 kern.debug kernel: [  393.972328] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:15 2018 kern.debug kernel: [  393.981364] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:15 2018 kern.debug kernel: [  394.192177] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:16 2018 kern.debug kernel: [  394.412207] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:16 2018 kern.debug kernel: [  394.632287] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:16 2018 kern.debug kernel: [  394.852378] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:16 2018 kern.debug kernel: [  395.062464] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:17 2018 kern.debug kernel: [  395.282553] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:17 2018 kern.debug kernel: [  395.502642] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:17 2018 kern.debug kernel: [  395.722729] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:17 2018 kern.debug kernel: [  395.942825] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:17 2018 kern.debug kernel: [  396.162910] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.372995] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.468038] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.477064] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.486161] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.693127] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  396.903205] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:18 2018 kern.debug kernel: [  397.123293] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:19 2018 kern.debug kernel: [  397.343384] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:19 2018 kern.debug kernel: [  397.563466] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:19 2018 kern.debug kernel: [  397.783550] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:19 2018 kern.debug kernel: [  398.003638] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:19 2018 kern.debug kernel: [  398.223727] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.443809] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.663896] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.873991] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.970030] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.979030] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  398.988117] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:20 2018 kern.debug kernel: [  399.204105] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:21 2018 kern.debug kernel: [  399.414184] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:21 2018 kern.debug kernel: [  399.634269] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:21 2018 kern.debug kernel: [  399.844480] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:21 2018 kern.debug kernel: [  400.064450] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:22 2018 kern.debug kernel: [  400.294526] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:22 2018 kern.debug kernel: [  400.514605] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:22 2018 kern.debug kernel: [  400.734688] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:22 2018 kern.debug kernel: [  400.954774] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:22 2018 kern.debug kernel: [  401.174857] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.384936] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.479979] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.488861] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.498021] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.705086] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  401.915140] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:23 2018 kern.debug kernel: [  402.135218] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:24 2018 kern.debug kernel: [  402.355302] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:24 2018 kern.debug kernel: [  402.575377] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:24 2018 kern.debug kernel: [  402.795458] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:24 2018 kern.debug kernel: [  403.015538] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:24 2018 kern.debug kernel: [  403.225618] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.445696] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.655772] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.865852] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.961902] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.970899] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  403.979999] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:25 2018 kern.debug kernel: [  404.185999] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:26 2018 kern.debug kernel: [  404.396045] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:26 2018 kern.debug kernel: [  404.616122] ieee80211 phy0: change: 0x40
Fri Nov 30 02:23:26 2018 kern.debug kernel: [  404.826416] ieee80211 phy0: change: 0x40
[... 23 lines cut ...]
Fri Nov 30 02:23:28 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Nov 30 02:23:28 2018 kern.debug kernel: [  406.886928] ieee80211 phy0: change: 0x60
Fri Nov 30 02:23:28 2018 kern.info kernel: [  406.926974] device wlan0 left promiscuous mode
Fri Nov 30 02:23:28 2018 kern.info kernel: [  406.931487] br-lan: port 3(wlan0) entered disabled state
Fri Nov 30 02:23:28 2018 kern.debug kernel: [  406.985001] ieee80211 phy0: change: 0x100
Fri Nov 30 02:23:28 2018 daemon.notice hostapd: ELOOP: remaining socket: sock=22 eloop_data=0xb6f635b0 user_data=0 handler=0x36e54
Fri Nov 30 02:23:28 2018 daemon.notice netifd: radio0 (10408): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Fri Nov 30 02:23:28 2018 daemon.notice netifd: radio0 (10408): WARNING (wireless_add_process): executable path /usr/sbin/hostapd does not match process  path (/proc/exe)
Fri Nov 30 02:23:28 2018 daemon.notice netifd: radio0 (10408): Command failed: Invalid argument
Fri Nov 30 02:23:28 2018 kern.debug kernel: [  407.162031] ieee80211 phy0: change: 0xffffffff
Fri Nov 30 02:23:28 2018 kern.info kernel: [  407.267105] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Fri Nov 30 02:23:29 2018 kern.info kernel: [  407.283787] br-lan: port 3(wlan0) entered blocking state
Fri Nov 30 02:23:29 2018 kern.info kernel: [  407.289147] br-lan: port 3(wlan0) entered disabled state
Fri Nov 30 02:23:29 2018 kern.info kernel: [  407.294588] device wlan0 entered promiscuous mode
Fri Nov 30 02:23:39 2018 kern.info kernel: [  417.298133] device wlan0 left promiscuous mode
Fri Nov 30 02:23:39 2018 kern.info kernel: [  417.302667] br-lan: port 3(wlan0) entered disabled state
```
I get this error when:
- I set it to 160mhz (in any channel and auto mode)
- I set any channel that's a DFS channel
- I set it to auto channel

Besides that, when I connect a device (either to the 5 or to the 2.4Ghz network) and add a second one the wireless freezes and the only solution is a reboot.

I hope you can help me and I hope I'm posting this in the right place 😄

tiagogaspar8, 2018-11-30:
```
Fri Nov 30 22:47:56 2018 kern.debug kernel: [  220.574978] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:00 2018 kern.err kernel: [  224.574137] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Fri Nov 30 22:48:00 2018 kern.err kernel: [  224.580000] ieee80211 phy0: return code: 0x1111
Fri Nov 30 22:48:00 2018 kern.err kernel: [  224.584561] ieee80211 phy0: timeout: 0x1111
Fri Nov 30 22:48:00 2018 kern.debug kernel: [  224.588915] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:04 2018 kern.err kernel: [  228.576413] ieee80211 phy0: cmd 0x9100=BssStart timed out
Fri Nov 30 22:48:04 2018 kern.err kernel: [  228.581885] ieee80211 phy0: return code: 0x1100
Fri Nov 30 22:48:04 2018 kern.err kernel: [  228.586552] ieee80211 phy0: timeout: 0x1100
Fri Nov 30 22:48:04 2018 kern.debug kernel: [  228.590763] ieee80211 phy0: change: 0x40
Fri Nov 30 22:48:04 2018 kern.debug kernel: [  228.590768] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:08 2018 kern.err kernel: [  232.595785] ieee80211 phy0: cmd 0x9114=SetApMode timed out
Fri Nov 30 22:48:08 2018 kern.err kernel: [  232.601319] ieee80211 phy0: return code: 0x1114
Fri Nov 30 22:48:08 2018 kern.err kernel: [  232.605897] ieee80211 phy0: timeout: 0x1114
Fri Nov 30 22:48:08 2018 kern.debug kernel: [  232.610100] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:12 2018 kern.err kernel: [  236.597027] ieee80211 phy0: cmd 0x8129=LinkAdaptCsMode timed out
Fri Nov 30 22:48:12 2018 kern.err kernel: [  236.603068] ieee80211 phy0: return code: 0x0129
Fri Nov 30 22:48:12 2018 kern.err kernel: [  236.607625] ieee80211 phy0: timeout: 0x0129
Fri Nov 30 22:48:12 2018 kern.debug kernel: [  236.611828] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:16 2018 kern.err kernel: [  240.592368] ieee80211 phy0: cmd 0x8120=80211hDetectRadar timed out
Fri Nov 30 22:48:16 2018 kern.err kernel: [  240.598584] ieee80211 phy0: return code: 0x0120
Fri Nov 30 22:48:16 2018 kern.err kernel: [  240.603133] ieee80211 phy0: timeout: 0x0120
Fri Nov 30 22:48:16 2018 kern.debug kernel: [  240.607334] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:20 2018 kern.err kernel: [  244.587546] ieee80211 phy0: cmd 0x810a=SetRfChannel timed out
Fri Nov 30 22:48:20 2018 kern.err kernel: [  244.593326] ieee80211 phy0: return code: 0x010a
Fri Nov 30 22:48:20 2018 kern.err kernel: [  244.597876] ieee80211 phy0: timeout: 0x010a
Fri Nov 30 22:48:20 2018 kern.debug kernel: [  244.602086] ieee80211 phy0: change: 0x100
Fri Nov 30 22:48:20 2018 kern.debug kernel: [  244.602091] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:21 2018 daemon.notice netifd: radio0 (6588): Command failed: Request timed out
Fri Nov 30 22:48:24 2018 kern.err kernel: [  248.595885] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
Fri Nov 30 22:48:24 2018 kern.err kernel: [  248.602101] ieee80211 phy0: return code: 0x001c
Fri Nov 30 22:48:24 2018 kern.err kernel: [  248.606651] ieee80211 phy0: timeout: 0x001c
Fri Nov 30 22:48:24 2018 daemon.notice netifd: Network device 'wlan0' link is down
Fri Nov 30 22:48:24 2018 kern.debug kernel: [  248.612092] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:28 2018 kern.err kernel: [  252.605895] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Fri Nov 30 22:48:28 2018 kern.err kernel: [  252.612027] ieee80211 phy0: return code: 0x1122
Fri Nov 30 22:48:28 2018 kern.err kernel: [  252.616579] ieee80211 phy0: timeout: 0x1122
Fri Nov 30 22:48:28 2018 kern.err kernel: [  252.620782] wlan0: failed to remove key (1, ff:ff:ff:ff:ff:ff) from hardware (-5)
Fri Nov 30 22:48:28 2018 kern.debug kernel: [  252.628485] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:32 2018 kern.err kernel: [  256.626817] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Fri Nov 30 22:48:32 2018 kern.err kernel: [  256.632680] ieee80211 phy0: return code: 0x1111
Fri Nov 30 22:48:32 2018 kern.err kernel: [  256.637243] ieee80211 phy0: timeout: 0x1111
Fri Nov 30 22:48:32 2018 kern.debug kernel: [  256.641451] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:36 2018 kern.err kernel: [  260.642441] ieee80211 phy0: cmd 0x801c=80211RadioControl timed out
Fri Nov 30 22:48:36 2018 kern.err kernel: [  260.648657] ieee80211 phy0: return code: 0x001c
Fri Nov 30 22:48:36 2018 kern.err kernel: [  260.653206] ieee80211 phy0: timeout: 0x001c
Fri Nov 30 22:48:36 2018 kern.debug kernel: [  260.657416] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Fri Nov 30 22:48:36 2018 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Fri Nov 30 22:48:36 2018 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Fri Nov 30 22:48:36 2018 daemon.notice netifd: radio0 (6642): command failed: Not supported (-95)
Fri Nov 30 22:48:36 2018 user.notice mac80211: Failed command: iw phy phy0 set distance 0
```

More logs that may help
Whe the wifi freezed after connecting two devices I restarted the adapter and this happened.

yuhhaurlin, 2018-12-14:
cat: can't open '/var/run/wifi-phy0.pid': No such file or directory

You need to check why this file is not existed.

closed 2018-12-14 by yuhhaurlin