kaloz/mwlwifi #334: ethtool does not render survey data due to failing channel comparison

ignisf, 2018-11-30:
Hello,

After I spent some time debugging why `ethtool -S <interface>` does not return the correct survey data, I am at the conclusion that it's due to this the comparison at the end of the following loop always returning true:

https://github.com/torvalds/linux/blob/94f371cb73944b410a269d570d6946c042f2ddd0/net/mac80211/ethtool.c#L159-L166

This seems to be the case since https://github.com/kaloz/mwlwifi/commit/245005057a318f689fbe5d2c58b0eef2fb7d91e9#diff-86fa451d5f8cfd6dab17fc54b7055f7bL571 which changed the channel value in the `survey_info` struct from `&hw->conf->chandef.chan` to `survey_info->channel`.

I am wondering what would be a good way to work around this issue. I am yet to inspect what exactly is different in the two structs.

ignisf, 2018-11-30:
For reference, here's how ath9k seems to set this value in the survey_info struct: 

https://github.com/torvalds/linux/blob/d146194f31c96f9b260c5a1cf1592d2e7f82a2e2/drivers/net/wireless/ath/ath9k/main.c#L1952-L1990

yuhhaurlin, 2019-03-02:
Please use iw related wireless tools.

closed 2019-03-02 by yuhhaurlin