kaloz/mwlwifi #337: WRT3200ACM 2,4 GHz wifi monitor mode

vargalex, 2018-12-10:
I have compiled an OpenWrt 18.06.1 image, with

kmod-mwlwifi - 4.14.79+2018-06-15-8683de8e-1
mwlwifi-firmware-88w8964 - 2018-06-15-8683de8e-1

I have configured the 2,4GHz wifi device to monitor mode. I have tested with all channels, but no packages arrived (I have tested with tcpdump and with my own tool). The 5 GHz radio has not this problem.

I do something wrong?

vargalex, 2018-12-10:
I think this is same issue as 107

vargalex, 2018-12-10:
Dear 

> From driver point of view, there should be no difference between 2.4g and 5g. Have you connected station to AP?

Of course, I have many clients connected. When I run the same monitor tool, or tcpdump on my notebook, I receive the packets...

vargalex, 2018-12-12:
I have tested from Lede 17.01.5 to OpenWrt snapshot. All versions have the same issue.

vargalex, 2019-01-02:
Can anyone reproduce the problem?

BrainSlayer, 2019-01-02:
will check it this evening

vargalex, 2019-01-08:
> will check it this evening

Hi!

Have you tested?

BrainSlayer, 2019-01-08:
@vargalex yes and it works as expected. so check your configuration

i just added a monitor interface to a existing running 2.4 ghz ap configuration with 
iw phy phy1 interface add mon1 type monitor 

and did a tcpdump on mon1
tcpdump -i mon1 -n                                                                                                                                                                         tcpdump: verbose output suppressed, use -v or -vv for full protocol decode                                                                                                                             listening on mon1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes                                                                                                 23:05:28.475891 1.0 Mb/s 2452 MHz 11b -51dBm signal antenna 0 Beacon (dd-wrt-nettle) [1.0* 2.0* 5.5* 11.0* 6.0* 9.0 12.0* 18.0 Mbit] ESS CH: 9, PRIVACY                                               
 23:05:28.483987 1.0 Mb/s 2452 MHz 11b -78dBm signal antenna 0 Beacon (laerche_kabel) [1.0* 2.0* 5.5* 11.0* 9.0 18.0 36.0 54.0 Mbit] ESS CH: 9, PRIVACY                                                 
23:05:28.493133 1.0 Mb/s 2452 MHz 11b -43dBm signal antenna 0 Beacon (dd-wrt-802.11n) [1.0* 2.0* 5.5* 11.0* 6.0* 9.0 12.0* 18.0 Mbit] ESS CH: 9, PRIVACY

vargalex, 2019-01-09:
> @vargalex yes and it works as expected. so check your configuration
>
> i just added a monitor interface to a existing running 2.4 ghz ap configuration with
> iw phy phy1 interface add mon1 type monitor [...]

Hi!

This is exactly my problem. There are ony Beacon packets captured, but no data packets!

closed 2019-03-02 by yuhhaurlin

vargalex, 2019-03-03:
Why have you closed this ticket?