kaloz/mwlwifi #341: No wifi with enrcyption and freebsd 12.0 client

asteriosos, 2018-12-22:
WRT 3200ACM - Powered by LuCI openwrt-18.06 branch (git-18.340.83383-3dea6b5) 1 / OpenWrt 18.06.1 r7258-5eb055306f

I have a lot of clients working very well with it. ( IOS, Linux, TV ... ) But now i have a Computer with freebsd 12 on it. With LAN works nice, but with wifi i can not get a connection. ( only when i disable the encryption on the wrt router ) With other routers and encryption the freebsd client works well too. I tried all wifi settings i got :wink: but nothing worked. I only get a authentication sucessfull but no connection. ( it dosent matter dhcp or static )

any ideas for that, i tried also many setting on the freebsd side... with debian linux on that computer i got no problems.

asteriosos, 2018-12-27:
i got it wotk now, i disabled the wmm mode on the router

BrainSlayer, 2018-12-27:
@asteriosos but same for you. disable wmm will disable 802.11n and 802.11ac and turns back to legacy 802.11a/b/g mode only

asteriosos, 2018-12-27:
802.11n and 802.11ac still active. (wrt3200 acm).

BrainSlayer, 2018-12-27:
if you disable wmm. its not 802.11n / ac capable anymore. wmm is a requirement for 802.11n and ac operation mode. if wmm is disabled both is not available anymore

BrainSlayer, 2018-12-27:
so disable wmm is useless and will not solve any bug. it just disables a important feature, but does not solve the problem

aaron1959, 2018-12-27:
Like fixing a flat by removing the tire completely, you can roll on the rims but it won't work right.

asteriosos, 2018-12-28:
sure it dosent solve the issue, but i can use this computer now. and now i got an 2.4 ghz 54mbit/s wlan and a 5ghz ac full speed wlan.
[image]

asteriosos, 2018-12-28:
i tried many things. i tried all wifi options on the router, i gave all options in the wpa_supplicant config. nothing worked, only disable the encryption helped to get a connection, i tried with the integrated intel (iwn) wireless pro 5100 and an usb stick Edimax AC ( rtl chip ) both can not connect to the openwrt router (both fully supported in freebsd). with other routers or my mobile hot-spot all works nice. only at home i need a cable.... so at the moment this is the fix for getting a secure wifi connection with freebsd and the openwrt router.

BrainSlayer, 2018-12-28:
@asteriosos yes i know. this is a known longterm bug on the wrt3200acm / marvell drivers which remains unresolved. you're not the only one with that problem and marvell refuses to fix that incompatiblity

yuhhaurlin, 2019-03-02:
It looks like same as issue #278. Close this one.

closed 2019-03-02 by yuhhaurlin