kaloz/mwlwifi #343: mwlwifi eating ARP packets

65a, 2019-01-03:
I have two APs with identical firmware, except one is ath10k and one is mwlwifi. If I bring up an SSID on mwlwifi (wifi1) arp is broken and a mobile app cannot find a network device. If I move the same config to the ath10k device, all is well.

In the log below, wifi1 is the mwlwifi node, the mobile device is 172.16.1.150, and the other wifi device it is trying to talk to is 172.16.1.16. 172.16.1.1 is a wired router.

root@wifi1:~# tcpdump -nXi wlan1 arp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan1, link-type EN10MB (Ethernet), capture size 262144 bytes
09:19:36.677951 ARP, Request who-has 172.16.1.16 tell 172.16.1.150, length 28
	0x0000:  0001 0800 0604 0001 64bc 0c46 f5bb ac10  ........d..F....
	0x0010:  0196 0000 0000 0000 ac10 0110            ............
09:19:36.677968 ARP, Request who-has 172.16.1.16 tell 172.16.1.150, length 28
	0x0000:  0001 0800 0604 0001 64bc 0c46 f5bb ac10  ........d..F....
	0x0010:  0196 0000 0000 0000 ac10 0110            ............
09:19:37.551341 ARP, Request who-has 172.16.1.16 tell 172.16.1.150, length 28
	0x0000:  0001 0800 0604 0001 64bc 0c46 f5bb ac10  ........d..F....
	0x0010:  0196 0000 0000 0000 ac10 0110            ............
........ SNIP (ARP request repeats for a long time)   ............
09:19:49.199296 ARP, Request who-has 172.16.1.1 tell 172.16.1.16, length 28
	0x0000:  0001 0800 0604 0001 0009 b011 9db5 ac10  ................
	0x0010:  0110 0000 0000 0000 ac10 0101            ............
09:19:49.199648 ARP, Reply 172.16.1.1 is-at 00:22:4d:56:b7:a0, length 42
	0x0000:  0001 0800 0604 0002 0022 4d56 b7a0 ac10  ........."MV....
	0x0010:  0101 0009 b011 9db5 ac10 0110 0000 0000  ................
	0x0020:  0000 0000 0000 0000 0000                 ..........
09:20:10.189325 ARP, Request who-has 172.16.1.1 tell 172.16.1.16, length 28
	0x0000:  0001 0800 0604 0001 0009 b011 9db5 ac10  ................
	0x0010:  0110 0000 0000 0000 ac10 0101            ............
09:20:10.189632 ARP, Reply 172.16.1.1 is-at 00:22:4d:56:b7:a0, length 42
	0x0000:  0001 0800 0604 0002 0022 4d56 b7a0 ac10  ........."MV....
	0x0010:  0101 0009 b011 9db5 ac10 0110 0000 0000  ................
	0x0020:  0000 0000 0000 0000 0000                 ..........

In both cases, the wlan1 device is a brif on a bridge with multicast_to_unicast set to 1. Also in both cases, the other brif is a VLAN interface on the access point.

Since the traffic that is not working is all WIFI, I'm quite confused and suspect mwlwifi is doing something surprising that ath10k is not. I am less likely to suspect the other parts of the network because everything works if I take the same config to the ath10k device.

65a, 2019-01-03:
Marvell Mac80211 Wireless PCIE Network Driver version 10.3.8.0-20181210

PalebloodSky, 2019-02-05:
Interesting find, sadly I think the only one we have at this point to test it is BrainSlayer, there has been little to no progress lately on this driver.

BrainSlayer, 2019-02-05:
the problem is that i dont have this issue in my test setups. i'm just developing dd-wrt for this platform and trying to help other people here with answering questions. but my primary job is still focused on dd-wrt development. i'm reading here about a problem with a mobile client. but ath10k works. this is pretty unspecific. mobile client .mmh. mobile phone? which chipset? from the capture i also see that the mobile client is trying to reach another client. there is a well known feature named "client isolation" its often enabled by default and it is made to block connections inbetween clients. this behaviour here looks exactly like that.

BrainSlayer, 2019-02-05:
another thing to mention. if 2 clients are pinging to each other, it doesnt mean that the ap can see the packets. the mwlwifi chipset has a own firmware on the wifi chipset itself. its very likelly that the traffic is routed within the chipset firmware and not by the OS. in that case you may see something like "eating arp" effect. but its basicly not true. its just invisible. i dont like such fullmac chipsets. they are black boxed

yuhhaurlin, 2019-03-02:
It is better to use wifi sniffer to capture packets.

closed 2019-03-02 by yuhhaurlin