kaloz/mwlwifi #344: Used firmware susceptible to ThreadX vulnerabilities?

jow-, 2019-01-19:
An RCE vulnerability in various Marvell wireless chipset/firmware combinations has been recently reported, find more details at:
https://embedi.org/blog/remotely-compromise-devices-by-using-bugs-in-marvell-avastar-wi-fi-from-zero-knowledge-to-zero-click-rce/

Is the firmware bundled with the mwlwifi driver affected? The linked report specifically mentions `88W8897` chips and this repo contains a `bin/firmware/88W8897.bin`.

anazz20, 2019-01-19:
I read an earlier article, and they mentioned the 88W897A, not sure if that is a different chip or not, but it did say patches are in the works. I mean this chip is in a lot of IoT devices as well as routers.

tapper82, 2019-01-19:
I to would like more info about this pleas.

jeolivera, 2019-01-19:
I'd rather not jump to conclusions, but my opinion is that the lack of commits in this repo and https://github.com/mrvltest/mwlwifi-8997 may mean that they're devoting all their resources on actually patching it out. The existence (not the details) of the vulnerability was initially publicly disclosed on November 21-20 Zero Nights 2018 in St. Petersburg (https://2018.zeronights.ru/wp-content/uploads/materials/19-Researching-Marvell-Avastar-Wi-Fi.pdf). Regardless, it'd still be nice to actually hear from @yuhhaurlin to know what is affected and when the patch is available, if applicable to the WRT AC series.

kubrickfr, 2019-01-20:
Yes or they've given up on this open source project...

jeolivera, 2019-01-21:
Looks like the vulnerability has been assigned as CVE-2019-6496. However, details are still obfuscated and sparse.

http://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-6496
https://vuldb.com/?id.129973
https://nvd.nist.gov/vuln/detail/CVE-2019-6496

howl, 2019-01-21:
I gave up with the idea that the new linksys wrt series wifi interfaces were just going to have good open source support just the first day I tried them. Retained just one of them just in case the situation drastically change, but, still the same.

This driver have a lot of stability and other type of problems, and the main developer seems to be blind, as you can read that he states that the driver is very nice in different github issues. I think is going to be a miracle If you get a confirmation here from Marvell for the RCE vulnerability.

BrainSlayer, 2019-01-22:
@howl the best way to find it out, is trying it out. the security issue is well described. i wouldnt wonder if the issue persists in all chipsets since the codebase might be the same, even if there are differences in the chipsets

jeolivera, 2019-02-08:
Marvell's vendor statement is located [here](https://www.scribd.com/document/398350818/WiFi-CVE-2019-6496-Marvell-s-Statement). It seems confirmed that the 88W8897 is vulnerable. 

Additional information and an up to date summary here: https://kb.cert.org/vuls/id/730261/

yuhhaurlin, 2019-02-08:
Firmware of 88W8897 and 88W8997 are different code base. There should be no these issues.

closed 2019-02-08 by yuhhaurlin

boktai1000, 2019-02-08:
@yuhhaurlin I was wondering if you could elaborate a bit more on this response of that there "should be no issues". Please see the following information:

https://kb.cert.org/vuls/id/730261/ (Mentions 88W8787, 88W8797, 88W8801, **88W8897**, and **88W8997**)

88W8897 and 88W8997 both have .bin files provided by Marvell that are in this code repository:

* https://github.com/kaloz/mwlwifi/blob/master/bin/firmware/88W8897.bin
* https://github.com/kaloz/mwlwifi/blob/master/bin/firmware/88W8997.bin

Looking at the latest commits for each, it appears these are provided by Marvell themselves?

> Signed-off-by: David Lin <dlin@marvell.com>

I inquired to Will Dormann on Twitter as he is mentioned as document writer at CERT about this

* https://twitter.com/wdormann/status/1093941091043291136
* https://twitter.com/wdormann/status/1093944874154504192
* https://twitter.com/wdormann/status/1093945493623853056

So Marvell only mentions 88W8897 but we do know that a firmware binary has been provided by Marvell to this repository. In addition, CERT mentions 88W8997 though, as well as some other chipsets which are not present in this repository. But Will has also mentioned that it's entirely possible (but not necessarily confirmed) that it could potentially affect other chipsets in the Marvell Avastar Chipset series, meaning 'potentially' having on impact on the other chipsets provided here (88W8864.bin and 88W8964.bin, in addition to 88W8897 and 88W8997 which are confirmed affected).

So going back to the firmware binaries in this repository, would it not make sense to get a line of communication going with Marvell to inquire regarding a new updated firmware binary that could potentially resolve this issue? Has this line of communication already been done behind the scenes?

I don't think Marvell are going to willingly send out updated firmwares, I also found this thread that seems to indicate that it's more on the maintainers to reach out and Marvell themselves can't know the level of interested off hand or be responsible for sending out updated firmwares:

* https://marc.info/?l=linux-wireless&m=145818065101611&w=2

> > Yes we have some updates, and continue working on it. But we felt that
> community does not seem to be very interested in this driver, so we have not
> been updating the wireless org. Let us know if you are supportive and waiting
> for it.
> 
> My understanding was that we'd very much like the driver to be sent upstream.
> From my perspective, it looked like you'd lost interest, not us.
> 

Lastly, I compared both the marvell-wireless/mwlwifi and kaloz/mwlwifi repositories under /bin/firmware to see what versions are present in each

* https://github.com/marvell-wireless/mwlwifi/tree/master/bin/firmware
* https://github.com/kaloz/mwlwifi/tree/master/bin/firmware

Differences as follows:

88W8864.bin

* marvell-wireless - Upgrade 88W8864 firmware to 7.3.0.21	a year ago
* kaloz - Upgrade 88W8864 firmware to 7.2.9.26.	3 years ago

88W8997.bin

* marvell-wireless - Upgrade 88W8997 firmware to 8.4.0.46.	6 months ago
* kaloz - 	Upgrade 88W8997 firmware to 8.4.4.6.	2 months ago

it appears that each repository has a different more up to date firmware binary.

Anyways, @yuhhaurlin to summarize, have you and if not could you or the repository owner get in contact with Marvell and determine if these firmware binary file is indeed vulnerable and possibly get an updated firmware binary for each of the chipsets maintained by this project, as they may include a fix to the vulnerability in question?

I'm gathering this information from the outside in, so please let me know if I'm misunderstood anything here regarding the project. But I know for myself as well as others, we are looking for a more detailed response from the repository owner on why exactly this isn't vulnerable and "There should be no these issues" isn't a good enough answer, why are there no issues/etc.

dengqf6, 2019-02-13:
[image]
Glad I'm using 88W8864..

yuhhaurlin, 2019-02-13:
1. Binary file of 88W8897 and 88W8997 for AP mode are not used by open source.
2. The code base is not the same as client mode.

zhangyoufu, 2020-01-02:
How about CVE-2019-13581 & CVE-2019-13582?

Vulnerability details was published at https://keenlab.tencent.com/en/2020/01/02/exploiting-wifi-stack-on-tesla-model-s/.

For unknown reasons, Marvell decide to hide [the security advisory](https://www.marvell.com/documents/ioaj5dntk2ubykssa78s/) from their [website](https://www.marvell.com/support/security-advisories/index.jsp).

s-pimenta, 2020-01-03:
> How about CVE-2019-13581 & CVE-2019-13582?
>
> Vulnerability details was published at https://keenlab.tencent.com/en/2020/01/02/exploiting-wifi-stack-on-tesla-model-s/.
>
> For unknown reasons, Marvell [...]

There is any workaround with current OpenWrt version, or there is any future version to mitigate this?

I really doubt if will be an firmware update from Marvell...