kaloz/mwlwifi #345: mwlwifi driver crashing due to change to mvebu bridge config emulation

leighbb, 2019-01-29:
The following mainline Linux kernel commit causes a crash in the mwlwifi driver:

commit 1f08673eef1236f7d02d93fcf596bb8531ef0d12
Author: Thomas Petazzoni <thomas.petazzoni@bootlin.com>
Date:   Thu Oct 18 17:37:18 2018 +0200

    PCI: mvebu: Convert to PCI emulated bridge config space
    
    Convert the pci-mvebu driver to use the pci-bridge-emul logic, that
    helps emulating a root port PCI bridge configuration space.
    
    Signed-off-by: Thomas Petazzoni <thomas.petazzoni@bootlin.com>
    [lorenzo.pieralisi@arm.com: updated commit log]
    Signed-off-by: Lorenzo Pieralisi <lorenzo.pieralisi@arm.com>

The crash looks like this:

```
pgd = f47c1616
[d1480c14] *pgd=16096811, *pte=e0100243, *ppte=e0100013
Internal error: : 1008 [#1] SMP THUMB2
Modules linked in: sg(+) mwlwifi(+)
CPU: 1 PID: 426 Comm: systemd-udevd Not tainted 4.20.3+ #11
Hardware name: Marvell Armada 380/385 (Device Tree)
PC is at pcie_reset+0xb/0x40 [mwlwifi]
LR is at pcie_download_firmware+0x18/0x2fc [mwlwifi]
pc : [<af80ac2c>]    lr : [<af80ac79>]    psr: 60070133
sp : ce3a9c58  ip : c9318180  fp : b0847eec
r10: b0804c08  r9 : c9318140  r8 : 000029a4
r7 : 00000000  r6 : ce380440  r5 : ce380de0  r4 : ce3857a0
r3 : ce380de0  r2 : d1480000  r1 : ce380de0  r0 : ce380440
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment none
Control: 50c5387d  Table: 1e3ac04a  DAC: 00000051
Process systemd-udevd (pid: 426, stack limit = 0x1e0d6d10)
Stack: (0xce3a9c58 to 0xce3aa000)
9c40:                                                       ce3857a0 af80ac79
9c60: cf443870 000029a4 000029a8 b0804c08 00000000 ce380de0 af81336a 00000000
9c80: 00000000 000029a4 000029a8 b0804c08 b0847eec af803875 1e3fc000 c60c6040
9ca0: ce380440 b0804c08 cf443870 5e4e1a10 00100000 cf443800 ce3857a0 ce380440
9cc0: ce380de0 cf443800 af812290 cf443870 b0847eec af80a52f 00019100 af815104
9ce0: af815104 cf443870 cf443800 00000000 af8150b4 af81507c 00000013 b02629cf
9d00: b0262963 cf443870 00000000 00000000 b0847ee0 00000000 af8150b4 b028e12f
9d20: 00000000 cf443870 af8150b4 af8150b4 b028e429 00000000 00000024 00000028
9d40: c93186e4 b028e40b af8150b4 cf443870 00000000 cf443870 cf4438a4 af8150b4
9d60: b028e429 00000000 00000024 b028e473 cf443870 b0804c08 af8150b4 b028d1c9
9d80: ce3a9da8 cf414c58 ce9a1db4 5e4e1a10 00000000 af8150b4 c60c6e00 00000000
9da0: b08173fc b028d6a3 af8130db af8130dc 0000006d af8150b4 b0804c08 af833001
9dc0: ffffe000 b028e8fd af8150fc b0831300 b0804c08 b0102601 00000001 006000c0
9de0: ce29e000 ce93a240 cf4e9b40 cf501400 ce29e000 00003bdd 8040003f ce29e000
9e00: 80000000 5e4e1a10 ce37bf80 cffc0f60 00000001 00000001 cffc0f60 00000001
9e20: 00000001 5e4e1a10 af815640 af815640 c9318b80 c93186c0 00000001 b016020d
9e40: af815640 b072f5fc ce3a9f40 af815640 00000001 b015f877 af81564c 00007fff
9e60: af815640 b015dc91 ce248488 af815688 b015dbd1 b080f790 b0502ac8 d128a440
9e80: a6f1bbb8 000001f8 ce3a8000 af000000 00000000 b0804c08 ce3a9ec0 b01a133f
9ea0: 00000000 00000000 00000000 00000000 00000000 00000000 6e72656b 00006c65
9ec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
9ee0: 00000000 00000000 00000000 00000000 00000000 5e4e1a10 7fffffff 00000000
9f00: b0804c08 a6f1bbb8 00000006 b0101224 ce3a8000 0000017b 02147238 b015fd1b
9f20: 7fffffff 00000000 00000003 00000000 0001b4b8 d126f000 0001b4b8 00000000
9f40: d127f730 d127ff00 d126f000 0001b4b8 d1289ea0 d1289d14 d1283d38 00013000
9f60: 00014f90 00000000 00000000 00000000 000061e4 00000024 00000025 0000001b
9f80: 00000000 00000013 00000000 5e4e1a10 00000006 02147238 0213c3c8 f739ef00
9fa0: 0000017b b0101001 02147238 0213c3c8 00000006 a6f1bbb8 00000000 a6f1c398
9fc0: 02147238 0213c3c8 f739ef00 0000017b 00020000 004ef6c4 00000000 02147238
9fe0: ae813948 ae813938 a6f15aeb a6e25fc2 600f0030 00000006 00000000 00000000
[<af80ac2c>] (pcie_reset [mwlwifi]) from [<af80ac79>] (pcie_download_firmware+0x18/0x2fc [mwlwifi])
[<af80ac79>] (pcie_download_firmware [mwlwifi]) from [<af803875>] (mwl_init_hw+0xf8/0x8b0 [mwlwifi])
[<af803875>] (mwl_init_hw [mwlwifi]) from [<af80a52f>] (pcie_probe+0x18e/0x1e8 [mwlwifi])
[<af80a52f>] (pcie_probe [mwlwifi]) from [<b02629cf>] (pci_device_probe+0x6d/0xb6)
[<b02629cf>] (pci_device_probe) from [<b028e12f>] (really_probe+0x137/0x1c4)
[<b028e12f>] (really_probe) from [<b028e40b>] (driver_probe_device+0xef/0x10c)
[<b028e40b>] (driver_probe_device) from [<b028e473>] (__driver_attach+0x4b/0x6e)
[<b028e473>] (__driver_attach) from [<b028d1c9>] (bus_for_each_dev+0x31/0x5c)
[<b028d1c9>] (bus_for_each_dev) from [<b028d6a3>] (bus_add_driver+0xd7/0x130)
[<b028d6a3>] (bus_add_driver) from [<b028e8fd>] (driver_register+0x5b/0x8a)
[<b028e8fd>] (driver_register) from [<b0102601>] (do_one_initcall+0x45/0xfc)
[<b0102601>] (do_one_initcall) from [<b016020d>] (do_init_module+0x39/0x134)
[<b016020d>] (do_init_module) from [<b015f877>] (load_module+0x1141/0x1492)
[<b015f877>] (load_module) from [<b015fd1b>] (sys_finit_module+0x53/0x6c)
[<b015fd1b>] (sys_finit_module) from [<b0101001>] (ret_fast_syscall+0x1/0x62)
Exception stack(0xce3a9fa8 to 0xce3a9ff0)
9fa0:                   02147238 0213c3c8 00000006 a6f1bbb8 00000000 a6f1c398
9fc0: 02147238 0213c3c8 f739ef00 0000017b 00020000 004ef6c4 00000000 02147238
9fe0: ae813948 ae813938 a6f15aeb a6e25fc2
Code: 727c 6b43 589c 68e2 (f8d2) 2c14 
---[ end trace 57b500d8bc2ce2a5 ]---
Kernel panic - not syncing: Fatal exception
CPU0: stopping
CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D           4.20.3+ #11
Hardware name: Marvell Armada 380/385 (Device Tree)
[<b010a691>] (unwind_backtrace) from [<b0108017>] (show_stack+0xb/0xc)
[<b0108017>] (show_stack) from [<b043c073>] (dump_stack+0x5b/0x70)
[<b043c073>] (dump_stack) from [<b0109ba5>] (handle_IPI+0x95/0x118)
[<b0109ba5>] (handle_IPI) from [<b0250a85>] (gic_handle_irq+0x59/0x68)
[<b0250a85>] (gic_handle_irq) from [<b0101a65>] (__irq_svc+0x65/0x94)
Exception stack(0xb0801f20 to 0xb0801f68)
1f20: 00000000 00042a24 cfbd42a0 b010e121 00000000 b0800000 b0804c08 00000000
1f40: 00000001 b0801f78 b0804c68 00000000 00000008 b0801f70 b010636f b0106366
1f60: 40010033 ffffffff
[<b0101a65>] (__irq_svc) from [<b0106366>] (arch_cpu_idle+0xc/0x1e)
[<b0106366>] (arch_cpu_idle) from [<b0132d9b>] (do_idle+0xd7/0x17c)
[<b0132d9b>] (do_idle) from [<b0132fab>] (cpu_startup_entry+0x13/0x14)
[<b0132fab>] (cpu_startup_entry) from [<b0700937>] (start_kernel+0x25f/0x2dc)
```

Reverting the commit fixes the issue.  I don't know how to change the mwlwifi driver to cope with this change.

BrainSlayer, 2019-01-29:
you should report this to the mainline maintainer of this bridge driver. this is no bug in the mwlwifi driver

yuhhaurlin, 2019-01-29:
@BrainSlayer You are right. I close this one.

closed 2019-01-29 by yuhhaurlin

leighbb, 2019-01-29:
> @BrainSlayer [1] You are right. I close this one.

@BrianSlayer, @yuhhaurlin 

This out-of-tree driver no longer works with the latest version of
Linux. I don't know how the driver works or interfaces with Linux.
Simply closing the issue and asking the person who raised the issue to
go and persuade the Linux kernel developers to change the Linux kernel
to make it work again doesn't seem like the right course of action to
me. 

Regards, 

Leigh. 

Links:
------
[1] https://github.com/BrainSlayer
[2] https://github.com/kaloz/mwlwifi/issues/345#issuecomment-458483438
[3]
https://github.com/

BrainSlayer, 2019-01-29:
@leighbb 
the problem is that the linux kernel introduced a regression here. this bug is pci bus related and not driver related. this issue cannot be fixed with changing the mwlwifi driver in any way. the linux kernel must be fixed here

leighbb, 2019-01-29:
@BrainSlayer 
I have posted to linux-arm-kernel mailing list.

http://lists.infradead.org/pipermail/linux-arm-kernel/2019-January/628438.html

Also, it was pointed out that my kernel backtrace was incomplete so here is a full one:

```
[....] Waiting for /dev to be fully populated...sd 2:0:0:0: Attached scsi generic sg0 type 0
<<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.8.0-20181210>>
pci 0000:00:01.0: enabling device (0140 -> 0142)
ieee80211 phy0: priv->pcmd_buf = 26d7c639  priv->pphys_cmd_buf = 3c1b4384
Unhandled fault: external abort on non-linefetch (0x1008) at 0xd1480c14
pgd = cf4c3420
[d1480c14] *pgd=1ea6b811, *pte=e0100243, *ppte=e0100013
Internal error: : 1008 [#1] SMP THUMB2
Modules linked in: mwlwifi(+) sg
CPU: 0 PID: 421 Comm: systemd-udevd Not tainted 5.0.0-rc3 #2
Hardware name: Marvell Armada 380/385 (Device Tree)
PC is at pcie_reset+0xc/0x40 [mwlwifi]
LR is at pcie_download_firmware+0x19/0x2fc [mwlwifi]
pc : [<af815c0c>]    lr : [<af815c59>]    psr: 60070033
sp : ce2a5c58  ip : ce9ff0c0  fp : b0808c08
r10: 000029a8  r9 : ce9ff080  r8 : 000029a4
r7 : 00000000  r6 : cdcc0480  r5 : cdcc0e20  r4 : cdcc57e0
r3 : cdcc0e20  r2 : d1480000  r1 : cdcc0e20  r0 : cdcc0480
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment none
Control: 50c5387d  Table: 1620004a  DAC: 00000051
Process systemd-udevd (pid: 421, stack limit = 0x9ec239b5)
Stack: (0xce2a5c58 to 0xce2a6000)
5c40:                                                       cdcc57e0 af815c59
5c60: cf447070 000029a4 00000000 000029a8 00000000 cdcc0e20 af81e372 00000000
5c80: 00000000 000029a4 00000000 000029a8 b0808c08 af80e87d 1ea60000 ce87c940
5ca0: cdcc0480 b0808c08 cf447070 bf9649ae 00100000 cdcc57e0 cf447000 cdcc0480
5cc0: cdcc0e20 cf447000 af81d298 cf447070 b084adfc af815515 00019100 af820104
5ce0: af820104 cf447070 cf447000 00000000 af8200b4 af82007c 00000013 b026372b
5d00: b02636bf cf447070 00000000 00000000 b084adf0 00000000 af8200b4 b028f6b7
5d20: 00000000 cf447070 af8200b4 af8200b4 b028f9b1 00000000 00000024 00000028
5d40: ce9ffee4 b028f993 af8200b4 cf447070 00000000 cf447070 cf4470a4 af8200b4
5d60: b028f9b1 00000000 00000024 b028f9fb cf447070 b0808c08 af8200b4 b028e755
5d80: ce2a5da8 cf414358 ce99de34 bf9649ae 00000000 af8200b4 ce87c380 00000000
5da0: b081ac50 b028ec2f af81e0e3 af81e0e4 0000006d af8200b4 b0808c08 af826001
5dc0: ffffe000 b028fe89 af8200fc b0834680 b0808c08 b0102601 00000001 006000c0
5de0: b082ffc0 cfce6560 0000762b b017c98d 00000001 b0808c08 8040003f cfeaa3e4
5e00: cfce6564 bf9649ae cf6190c0 cffe6320 00000001 00000001 cffe6320 00000001
5e20: 00000001 bf9649ae af820580 af820580 ce9ff640 ce9ffec0 00000001 b0160e4d
5e40: af820580 b072f6fc ce2a5f40 af820580 00000001 b016049f af82058c 00007fff
5e60: af820580 b015e8a9 cf1415c0 af8205c8 b015e7e9 b08134d0 b05029fc d129a4bc
5e80: a6eecbb8 000001d6 ffffe000 ce2a4000 af000000 00000000 b0808c08 b01a42a3
5ea0: 00000000 00000000 00000000 00000000 00000000 00000000 6e72656b 00006c65
5ec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5ee0: 00000000 00000000 00000000 00000000 00000000 bf9649ae 7fffffff 00000000
5f00: b0808c08 a6eecbb8 0000000c b0101224 ce2a4000 0000017b 00a6b0d8 b0160943
5f20: 7fffffff 00000000 00000003 00000000 0001a534 d1280000 0001a534 00000000
5f40: d128ff78 d1290580 d1280000 0001a534 d1299fbc d1299e58 d1294088 00013000
5f60: 00014d70 00000000 00000000 00000000 00005ea8 00000020 00000021 00000017
5f80: 00000000 00000013 00000000 bf9649ae 0000000c 00a6b0d8 00a6a3c8 941a5c00
5fa0: 0000017b b0101001 00a6b0d8 00a6a3c8 0000000c a6eecbb8 00000000 a6eed398
5fc0: 00a6b0d8 00a6a3c8 941a5c00 0000017b 00020000 0051e6c4 00000000 00a6b0d8
5fe0: aeab5948 aeab5938 a6ee6aeb a6df6fc2 600f0030 0000000c 00000000 00000000
[<af815c0c>] (pcie_reset [mwlwifi]) from [<af815c59>] (pcie_download_firmware+0x19/0x2fc [mwlwifi])
[<af815c59>] (pcie_download_firmware [mwlwifi]) from [<af80e87d>] (mwl_init_hw+0x101/0x8b8 [mwlwifi])
[<af80e87d>] (mwl_init_hw [mwlwifi]) from [<af815515>] (pcie_probe+0x169/0x1bc [mwlwifi])
[<af815515>] (pcie_probe [mwlwifi]) from [<b026372b>] (pci_device_probe+0x6d/0xb6)
[<b026372b>] (pci_device_probe) from [<b028f6b7>] (really_probe+0x137/0x1c4)
[<b028f6b7>] (really_probe) from [<b028f993>] (driver_probe_device+0xef/0x10c)
[<b028f993>] (driver_probe_device) from [<b028f9fb>] (__driver_attach+0x4b/0x6e)
[<b028f9fb>] (__driver_attach) from [<b028e755>] (bus_for_each_dev+0x31/0x5c)
[<b028e755>] (bus_for_each_dev) from [<b028ec2f>] (bus_add_driver+0xd7/0x130)
[<b028ec2f>] (bus_add_driver) from [<b028fe89>] (driver_register+0x5b/0x8a)
[... 5 lines cut ...]
Exception stack(0xce2a5fa8 to 0xce2a5ff0)
5fa0:                   00a6b0d8 00a6a3c8 0000000c a6eecbb8 00000000 a6eed398
5fc0: 00a6b0d8 00a6a3c8 941a5c00 0000017b 00020000 0051e6c4 00000000 00a6b0d8
5fe0: aeab5948 aeab5938 a6ee6aeb a6df6fc2
Code: 727c 6b43 589c 68e2 (f8d2) 2c14 
---[ end trace e65bfe3eaf372676 ]---
Kernel panic - not syncing: Fatal exception
CPU1: stopping
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.0.0-rc3 #2
Hardware name: Marvell Armada 380/385 (Device Tree)
[<b010a6ad>] (unwind_backtrace) from [<b010800f>] (show_stack+0xb/0xc)
[<b010800f>] (show_stack) from [<b04408d7>] (dump_stack+0x5b/0x70)
[<b04408d7>] (dump_stack) from [<b0109ba5>] (handle_IPI+0x95/0x118)
[<b0109ba5>] (handle_IPI) from [<b0251655>] (gic_handle_irq+0x59/0x68)
[<b0251655>] (gic_handle_irq) from [<b0101a65>] (__irq_svc+0x65/0x94)
Exception stack(0xcf469f60 to 0xcf469fa8)
9f60: 00000000 00037908 cfbe43a0 b010e0e1 00000000 cf468000 b0808c08 00000000
9f80: 00000002 cf469fb8 b0808c68 00000000 00000008 cf469fb0 b0106367 b010635e
9fa0: 40070033 ffffffff
[<b0101a65>] (__irq_svc) from [<b010635e>] (arch_cpu_idle+0xc/0x1e)
[<b010635e>] (arch_cpu_idle) from [<b01330a7>] (do_idle+0xd7/0x17c)
[<b01330a7>] (do_idle) from [<b01332b7>] (cpu_startup_entry+0x13/0x14)
[<b01332b7>] (cpu_startup_entry) from [<001023b1>] (0x1023b1)
Rebooting in 1 seconds..
```

BrainSlayer, 2019-01-29:
@leighbb okay. keep me up to date. i hope you posted that this may be a regression since it was introduced with the new pci bridge patch. set also the original maintainer as CC