kaloz/mwlwifi #346: wrt1900acs v2

olimpiumarius, 2019-01-29:
Poor performance in speed test Realtek 8822BE Wireless LAN 802.11ac PCI-E NIC, i use wrt1900acs v2 openwrt Kernel Version 4.14.95 SNAPSHOT r8392+767-3dba852547 / LuCI Master (git-19.026.48777-c81ae86) wi fi driver version: 10.3.8.0-20181210, 1Gbps internet connection.country Romania channel 36.

AjkayAlan, 2019-01-30:
Enable flow offload (in Luci go to Network - Firewall - Software Flow Offloading, save and apply). Right now you are probably hitting software Nat theoretical limits.

With flow offload enabled, what do things look like?

thagabe, 2019-01-31:
Most developers have assured me that the hardware is plenty strong to handle gigabit speed without any offloading.

PalebloodSky, 2019-02-05:
Not sure why you say that most people recommend enabling flow offload at this point it's a huge performance benefit, think it's even enabled by default at this point. At least try it to be sure.

olimpiumarius, 2020-09-26:
Hostname swrt
Model Linksys WRT1900ACS
Architecture ARMv7 Processor rev 1 (v7l)
Firmware Version SuperWRT SNAPSHOT r14567-b284333b3a / LuCI Master git-20.267.53658-b17ec09
Kernel Version 5.4.67
 -----------------------------------------------------
root@swrt:~# cat /etc/config/wireless verifica wifi

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option disabled '0'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option country 'GB'
        option txpower '21'
        option channel '36'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option macaddr '58:10:d1'
        option encryption 'psk2'
        option ssid 'superwrt 5Ghz'
        option key ''

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option disabled '0'
        option noscan '1'
        option htmode 'HT40'
        list channels '1'
        list channels '6'
        list channels '11'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option channel 'auto'
        option country 'GB'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option macaddr '58:ef:10:d0'
        option encryption 'psk2'
        option ssid 'superwrt'
        option key ''

cat: can't open 'verifica': No such file or directory
cat: can't open 'wifi': No such file or directory
root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/sta

mac address: ac:1e
aid: 1
ampdu: true
amsdu: false
wds: false
ba_hist: disable
ht_cap: 0x1020, ampdu: 02, 06
rx_mask: 0xff, 00, 00, 00
rx_bw: 0, rx_nss: 1
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 00000001160d

root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: 00::16
2g: enable
5g: disable
antenna: 4 4
irq number: 74
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: 135f8804
iobase1: 0d0afa06
tx limit: 768
rx limit: 64
qe trigger number: 64922

Software flow offloading: active

It seems that even in this version of OpenWrt the speed in the speedtest oscillates a lot, it is not constant!
The maximum speed I managed to have with this versionon 5Ghz is 545Mbps download and  500Mbps upload, in some tests the speed also drops to 250Mbps!
I can say that the values ​​indicated by Bitrate are now indicated correctly, on the information display side everything seems to work well in the latest driver versions!
[image]

closed 2023-12-08 by olimpiumarius