kaloz/mwlwifi #347: Repeated ieee80211 phy1: Mac80211 start BA

michaelh99, 2019-02-08:
Since updating to OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.034.35162-a0ae6c9)  I'm seeing repeated messages like the following when setting up a wlan client on radio1

The connection is regularly going up and down as well.

This issue was previously reported on https://github.com/eduperez/mwlwifi_LEDE/issues/18 but it looks like it never got re-reported here.

[   97.661592] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[   97.721592] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[   97.771595] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[   97.811596] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[   97.871593] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[   97.921593] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.182577] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.231561] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.279707] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.321567] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.371555] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
[  111.431555] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41

Thu Feb  7 21:55:38 2019 kern.debug kernel: [  111.182577] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:55:38 2019 kern.debug kernel: [  111.231561] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:55:38 2019 kern.debug kernel: [  111.279707] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:55:38 2019 kern.debug kernel: [  111.321567] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:55:38 2019 kern.debug kernel: [  111.371555] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:55:39 2019 kern.debug kernel: [  111.431555] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:56:02 2019 kern.debug kernel: [  135.219678] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:56:02 2019 kern.debug kernel: [  135.261411] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:56:09 2019 kern.debug kernel: [  141.722090] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 21:58:36 2019 kern.debug kernel: [  289.164356] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41

yuhhaurlin, 2019-02-08:
This issue has been fixed by latest driver.

closed 2019-02-08 by yuhhaurlin

michaelh99, 2019-02-08:
Where can I find the commit for this? 18.06.02 was fairly recently released and the latest commit here is 2 months old

yuhhaurlin, 2019-02-08:
Sorry. The log shows BA is started by mac80211. It is the fixed version. I don’t know what the problem you have?

michaelh99, 2019-02-08:
It's being restarted over and over and my connection is unreliable.

Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.170589] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.224031] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.264026] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.304036] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.344032] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.384033] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.424033] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:52 2019 kern.debug kernel: [ 1565.464039] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:19:53 2019 kern.debug kernel: [ 1565.504040] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.777635] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.835126] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.875120] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.915119] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.955123] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1586.995124] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1587.035133] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1587.075134] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1587.115131] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1587.155145] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:20:14 2019 kern.debug kernel: [ 1587.195135] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:24:12 2019 kern.debug kernel: [ 1824.908947] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41
Thu Feb  7 22:24:47 2019 kern.debug kernel: [ 1859.556165] ieee80211 phy1: Mac80211 start BA f2:9f:c2:aa:a7:41

derosier, 2019-02-08:
I don't know for sure, but...

You state you're using a recent upgrade of OpenWRT. I assume that uses
an upgrade of the kernel. There's been changes in the mac80211 stack
recently that might cause this problem. My best guess is that mwlwifi
should be tested and fixed against the more recent mac80211 stack.

Note that you can't blindly upgrade part of the system and just expect
other code that depends on it to "just work".  While the kernel
development process tries to ensure that we never break userspace,
there are never any guarantees like that between mac80211 and a random
driver.  The development process tries to enforce the policy that if
you change something and it breaks an in-tree driver, that the person
who broke it is responsible for fixing the driver, but even sometimes
things get missed.  However, there is no way to do that with random
out-of-tree vendor drivers like this one.

In anycase, I'm making some assumptions as to what is in the OpenWRT
version you mention as you didn't mention the relevant version
information (ie the kernel info). So it's possible it has nothing to
do with the recent changes that went upstream in the mac stack.

- Steve

michaelh99, 2019-02-08:
I haven't updated any single firmware or kernel modules since applying .2.  I have updated software through luci.
Kernel 4.14.95

It may have been a problem between the keyboard and chair.

I updated the rom from .1 to .2 and kept my configs.  I didn't immediately notice that travelmate and openvpn had been nuked (I was at a location where I hadn't needed them for a few months).

There were two disabled client APs that had previously been managed by travelmate and were still in the config.  Once I deleted those, the highly regular start messages changed to a couple and hour stop/start pairs and my connection is much more reliable

danielkza, 2019-02-17:
@michaelh99 I think your improvement might be coincidental, because I have never had any disabled clients/APs or any of the software you mentioned, and I'm getting the same spam and seeing my WDS client (TP-Link WDR4300, also running OpenWRT 18.06.2) dropped multiple times per hour.

danielkza, 2019-05-06:
Still present on the latest OpenWRT nightlies. My WDR4300 repeater is dropped and can't reauth until I reset the Wi-fi device in the WRT1900ACS.

adam6806, 2019-05-22:
I'm also currently seeing this with my WRT1900ACS that I setup just a few days ago fresh out of the box using the latest firmware. OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d)

`[27637.892678] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27638.948261] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.000913] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.050894] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.166970] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.220903] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.302969] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.360899] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.426387] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.490882] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.608023] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.660886] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.739798] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27639.841659] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27645.899515] ieee80211 phy0: Mac80211 start BA 4c:32:75:91:f7:c7
[27646.841802] ieee80211 phy0: Stop BA b0:fc:0d:89:f5:7d
[27674.633585] ieee80211 phy1: Mac80211 start BA a4:77:33:26:a2:0f
[27696.458911] ieee80211 phy0: Mac80211 start BA b0:fc:0d:89:f5:7d
[27696.891353] ieee80211 phy0: Stop BA 4c:32:75:91:f7:c7`

jeolives, 2019-05-29:
Does this problem occur during multicasting services? e.g. NAT-PMP, Apple's Bonjour service, etc.?

BrainSlayer, 2019-05-29:
ignore it. thats debug output shit of the mwlwifi driver. it just says that a ampdu block ack session has been started / stopped. block ack is a form of packet aggregation which increases performance. so if you see that message all is running as expected. i dont know why openwrt enabled debug output for that driver. its useless and confusing, but no error. just a information whats currently going on

zuzzurro, 2022-02-25:
I'm running 21.02.1 on my wrt1900ac-v2, two years have passed and still it happen and when it happens I have big issues connecting with my S21 to the WiFi.
I understand @BrainSlayer  point of this being just debug output, but I also think that the fact that people have reported this happening while they are having problems connecting has been slightly overlooked. I am going to upgrade later today to 21.02.2 and I'm quite willing to help debugging this issue that is most annoying.

xaoniu, 2022-04-08:
I'm running OpenWrt 21.02.2 on my wrt1900acs-v2. It happened frequently.  
When this happens, my WiFi becomes very slow.
OpenWrt 19.07 have the same problem.

Fri Apr  8 09:41:44 2022 kern.debug kernel: [21279.552104] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 09:56:00 2022 kern.debug kernel: [22135.528068] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 09:56:00 2022 kern.debug kernel: [22135.831899] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 09:56:07 2022 kern.debug kernel: [22143.206863] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 09:56:08 2022 kern.debug kernel: [22144.084845] ieee80211 phy0: Stop BA 66:0d:37:69:5c:94
Fri Apr  8 10:09:04 2022 kern.debug kernel: [22920.267749] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:09:05 2022 kern.debug kernel: [22920.476846] ieee80211 phy0: Stop BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:14 2022 kern.debug kernel: [23349.645489] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:14 2022 kern.debug kernel: [23349.691467] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:14 2022 kern.debug kernel: [23349.751470] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.397957] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.451460] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.511466] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.571454] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.631463] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.691467] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23350.751457] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:15 2022 kern.debug kernel: [23351.307313] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:16 2022 kern.debug kernel: [23351.371459] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:17 2022 kern.debug kernel: [23352.690450] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:16:18 2022 kern.debug kernel: [23353.914319] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:17:19 2022 kern.debug kernel: [23414.401967] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94
Fri Apr  8 10:17:19 2022 kern.debug kernel: [23414.450817] ieee80211 phy0: Mac80211 start BA 66:0d:37:69:5c:94

tamsky, 2022-06-26:
I have a client (M1 macbook) that has this identical logged message against its MAC addr, and symptoms
- slowness
- or total lack of wifi for **minutes** at a time

persist until `ieee80211 phy0: Stop BA <client_mac_addr>` is logged.

Same equipment — wrt1900acs-v2, running
OpenWrt 21.02.1 r16325-88151b8303, on Kernel Version | 5.4.154

vishap, 2023-01-14:
The same is here. It was stable until recently. 

[   34.212222] ieee80211 phy1: change: 0x40
[   34.382223] ieee80211 phy1: change: 0x40
[   34.552223] ieee80211 phy1: change: 0x40
[   34.752226] ieee80211 phy1: change: 0x40
[   34.810258] ieee80211 phy1: change: 0x100
[   34.819253] ieee80211 phy1: change: 0x100
[   34.828273] ieee80211 phy1: change: 0x42
[   34.948262] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   34.954743] br-lan: port 5(wlan1) entered blocking state
[   34.960084] br-lan: port 5(wlan1) entered forwarding state
[   53.493543] ieee80211 phy1: Mac80211 start BA 38:e7:c0:f2:3f:68
[   83.116631] ieee80211 phy0: Mac80211 start BA 64:a2:f9:3c:e8:aa
[  159.339450] ieee80211 phy0: Mac80211 start BA 8e:84:5f:05:5e:3b
[  877.937844] ieee80211 phy1: Mac80211 start BA 60:6d:c7:29:ff:9d
[ 1469.270347] ieee80211 phy1: Mac80211 start BA 9e:9e:b0:14:4c:1d
[ 1871.623246] ieee80211 phy1: Mac80211 start BA 60:6d:c7:29:ff:9d
[ 2960.566736] ieee80211 phy0: Mac80211 start BA 3e:91:c7:9f:6b:c9
[ 2961.660859] ieee80211 phy0: Stop BA 8e:84:5f:05:5e:3b
root@OpenWrt:~# reboot
root@OpenWrt:~# Connection to 192.168.0.1 closed by remote host.
Connection to 192.168.0.1 closed.
$ ssh root@192.168.0.1
root@192.168.0.1's password: 

BusyBox v1.35.0 (2022-10-14 22:44:41 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.2, r19803-9a599fee93
 -----------------------------------------------------
root@OpenWrt:~# uname -a
Linux OpenWrt 5.10.146 #0 SMP Fri Oct 14 22:44:41 2022 armv7l GNU/Linux
root@OpenWrt:~#

gkvirkvia, 2023-01-16:
I have this same problem happening. My 5Ghz interface is effectively unusable and I see the above `dmesg` lines.

Some observations: The network becomes available for minutes after I manually change the channel assignment. It goes offline soon after.

FW version: `OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.361.69894-438c598`

vishap, 2023-01-16:
My impression is that it's not the wireless, but actually eth0/eth1 get down in my case. That's why network becomes unavailable. Sometimes, those are brought back automatically after awhile. Sometimes, not. But ping to my wireless devices is ok. But I cannot ping my wired devices. 
What I have noticed is that this happens when I connect my macbook pro through a usb-c dock. I am not sure what it sends there, probably there is a poison packet. It's definitely not an electrical problem, because it's connected to a netgear switch which then is connected to router. That is completely isolated. In addition when I connect my dell laptop (linux not windows) through the same usb-c dock, no problem is observed.

tamsky, 2023-01-17:
This is a good reminder — I have seen this poison/jabber packet behavior with a M1 MacBook Air and the first usb-c<>RJ45 gigabit adapter I had around to try at the time.
My network also has the M1 connected to a netgear1000bT switch first, before reaching the openwrt router.

I'll try to replicate it and report if I can.

tamsky, 2024-02-08:
Just wanted to point out an unofficial openwrt build with patches/fixes specific to the `wrt1900acs-v2`'s wifi driver... (TX-AMSDU and DSA/FDB fixes)

https://divested.dev/unofficial-openwrt-builds/mvebu-linksys/