kaloz/mwlwifi #349: After a radar is detected, connection is impossible

kubrickfr, 2019-02-16:
When the firmware detects a radar and reverts to channel 36, it's impossible to connect to 5GHz any more, I have to restart the wireless interface.

The is the kernel log output:
```
[53763.723739] ieee80211 phy0: staid 1 deleted
[54321.957156] ieee80211 phy0: radar detected by firmware
[54322.490301] ieee80211 phy0: channel switch is done
[54322.495128] ieee80211 phy0: change: 0x60
[56785.476372] ieee80211 phy0: staid 1 deleted
[65342.262148] ieee80211 phy0: staid 1 deleted
[65680.127772] ieee80211 phy0: staid 1 deleted
[69039.340264] ieee80211 phy0: staid 1 deleted
[76378.672345] ieee80211 phy0: staid 1 deleted
[76413.872415] ieee80211 phy0: staid 1 deleted
[76428.574732] ieee80211 phy0: staid 1 deleted
[76443.617523] ieee80211 phy0: staid 1 deleted
[76774.500522] ieee80211 phy0: staid 1 deleted
[81028.579937] ieee80211 phy0: staid 1 deleted
[81372.457272] ieee80211 phy0: staid 1 deleted
[82260.422416] ieee80211 phy0: staid 1 deleted
[86802.751344] ieee80211 phy0: staid 1 deleted
[87455.435111] ieee80211 phy0: staid 1 deleted
[87662.033426] ieee80211 phy0: staid 1 deleted
[87665.265959] ieee80211 phy0: staid 1 deleted
[87711.862078] ieee80211 phy0: staid 1 deleted
[138014.656767] ieee80211 phy0: staid 1 deleted
[138029.970419] ieee80211 phy0: staid 1 deleted
[138066.903053] ieee80211 phy0: staid 1 deleted
[138068.883698] ieee80211 phy0: staid 1 deleted
[138097.483496] ieee80211 phy0: staid 1 deleted
[138111.020130] ieee80211 phy0: staid 1 deleted
[138149.846510] ieee80211 phy0: staid 1 deleted
[138483.002302] ieee80211 phy0: staid 1 deleted
[141104.695353] ieee80211 phy0: staid 1 deleted
[141439.000044] ieee80211 phy0: staid 1 deleted
[142146.323882] ieee80211 phy0: staid 1 deleted
[142185.809608] ieee80211 phy0: staid 1 deleted
```
For each `staid 1 deleted`, I get the following system log:
```
Fri Feb 15 18:15:22 2019 daemon.info hostapd: wlan0: STA ac:37:43:51:xx:xx IEEE 802.11: associated (aid 1)
Fri Feb 15 18:15:22 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED ac:37:43:51:xx:xx
Fri Feb 15 18:15:22 2019 daemon.info hostapd: wlan0: STA ac:37:43:51:xx:xx WPA: pairwise key handshake completed (RSN)
Fri Feb 15 18:15:25 2019 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED ac:37:43:51:xx:xx
Fri Feb 15 18:15:25 2019 kern.debug kernel: [87711.862078] ieee80211 phy0: staid 1 deleted
``` 

On the client side, association is performed but the dhcp request fails, but it works from the wired interface.

kubrickfr, 2019-02-19:
Also, it's not just from one client, it doesn't work from my phone or my laptop.

In case it helps, I've got tow SSIDS, one of which is hidden, connection is impossible to either.

ValCher1961, 2019-02-19:
I encountered a similar phenomenon on WRT3200ACM when using HOSTAPD-2.7.
What version of HOSTAPD do you use?

davidc502, 2019-02-20:
Does it behave the same way if the width is set to 40Mhz?

ValCher1961, 2019-02-20:
In my case, Debian, hostapd-2.7, recommended settings hostapd.conf for Mwlwifi width 80 mHz. After detecting radar and switching AP to 36 channel, clients on Intel AC7260 cannot connect to the AP. After the allotted time to release the channel and the absence of new radar signals, the AP does not return to the previous channel. If you use hostapd-2.6, you do not have this problem.
It is difficult to say whether my problem is similar to the author's problem.

davidc502, 2019-02-21:
The issue I see is when a device can't connect to a specific DFS channel this is in the kernel log. This usually only happens after radar is detected and goes to another DFS channel. 

[196922.640147] marvell-cesa f1090000.crypto: dma_pool_free cesa_cache, dma 0x125fb000 already free

ValCher1961, 2019-02-21:
Unfortunately I have not kept the logs and I can not quickly introduce new, because the detection of work radar I happen very rarely, two-three times a week.
Is there any method of emulation of DFS activation?

davidc502, 2019-02-21:
If you introduce another wifi SSID, from another device, on that same channel, it will probably vacate that channel and produce the log when clients try to re-connect.

ValCher1961, 2019-02-23:
Today I configured two routers WRT3200ACM on 132 channel with different SSID names, but they did not react to each other, they worked in parallel.
[image]

davidc502, 2019-02-25:
Thanks for testing.... Yeah, that hasn't been my experience. I have a neighbor's remote TV box that roams around DFS, and consistently bumps me off.  So, I'm surprised it has not happened to you.

lantis1008, 2019-03-01:
I believe you can 
`echo 1 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/dfs_test`

yuhhaurlin, 2019-03-02:
Yes. This is the way to fake radar detection. I did not find reconnection problem.

closed 2019-03-02 by yuhhaurlin

kubrickfr, 2019-03-02:
@yuhhaurlin have you tried with the European version? Because the problem is definitely happening, at least for two people here.

yuhhaurlin, 2019-03-02:
Can you use above way to generate radar detection and check what happens?

kubrickfr, 2019-03-05:
@yuhhaurlin just tested
```
echo 1 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/dfs_test
```
Nothing happens at all. Nothing in the system or kernel log, channel stays on 116.

FYI:
```
root@lede:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x09030206
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b6:97:ca
2g: disable
5g: enable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e0e80000
iobase1: e1100000
tx limit: 1024
rx limit: 16384
```

kubrickfr, 2019-04-23:
This is still happening, at least once a week...