kaloz/mwlwifi #352: openwrt 18.06.2 wrt32x 2.4G wifo not stability

HZ89, 2019-02-25:
i have same issue but only on 2G interface

kmod-mwlwifi - 4.14.95+2018-11-14-81413aa9-1
mwlwifi-firmware-88w8964 - 2018-11-14-81413aa9-1
``` bash
cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info 

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.8.0-20181114
firmware version: 0x09030206
power table loaded from dts: yes
firmware region code: 0x0
mac address: 60:38:e0:c1:bc:01
2g: enable
5g: disable
antenna: 4 4
irq number: 50
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1280000
iobase1: e1500000
tx limit: 1024
rx limit: 16384
```
logs

``` bash
Mon Feb 25 22:12:43 2019 kern.debug kernel: [ 2399.978118] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:12:43 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:12:43 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:12:43 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:12:43 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:13:00 2019 kern.debug kernel: [ 2417.379680] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:13:00 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:13:00 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:13:00 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:13:00 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:13:15 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:13:15 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:13:15 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:13:17 2019 kern.debug kernel: [ 2434.780302] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:13:17 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:13:17 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:13:17 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:13:17 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:13:35 2019 kern.debug kernel: [ 2452.181263] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:13:35 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:13:35 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:13:35 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:13:35 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:13:52 2019 kern.debug kernel: [ 2469.571216] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:13:52 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:13:52 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:13:52 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:13:52 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:14:10 2019 kern.debug kernel: [ 2486.973011] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:14:10 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:14:10 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:14:10 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:14:10 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:14:14 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:14:14 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:14:14 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:14:14 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: authenticated
Mon Feb 25 22:14:27 2019 kern.debug kernel: [ 2504.374577] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:14:27 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:14:27 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:14:27 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:14:27 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:14:44 2019 kern.debug kernel: [ 2521.776124] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:14:44 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:14:44 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:14:44 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:14:44 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)
Mon Feb 25 22:15:02 2019 kern.debug kernel: [ 2539.176588] ieee80211 phy1: staid 1 deleted
Mon Feb 25 22:15:02 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 IEEE 802.11: associated (aid 1)
Mon Feb 25 22:15:02 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 78:11:dc:0a:16:d2
Mon Feb 25 22:15:02 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 RADIUS: starting accounting session 0E668E41C1A830B2
Mon Feb 25 22:15:02 2019 daemon.info hostapd: wlan1: STA 78:11:dc:0a:16:d2 WPA: pairwise key handshake completed (RSN)

```

yuhhaurlin, 2019-03-02:
This driver is not used for WRT32X.

closed 2019-03-02 by yuhhaurlin

BrainSlayer, 2019-03-02:
@yuhhaurlin. ahm the wrt32x is technically identical with the wrt3200acm and openwrt like dd-wrt works on it. so the driver "IS USED"