kaloz/mwlwifi #353: Will this driver be worked on any more?

tapper82, 2019-02-27:
There has not bin any updates to this driver for a long time. The last commit was 18 Dec 2018 by @yuhhaurlin Is that it now?

davidc502, 2019-03-01:
Good question.

PalebloodSky, 2019-03-01:
It would appear so, based on dramatic reduction of commits over the past year, and literately nothing at all in 2-3 months that development is abandoned. It's a huge setback for routers as a whole, because wrt3200acm / wrt32x could be the best router on the market and now nothing.

I don't know if there will be anywhere to turn with OpenWrt / DD-WRT routers if this continues.

davidc502, 2019-03-02:
In my opinion the 3200/32x is the best router, minus the wifi :)  
Anyone have recommendations on really good Wifi Access points that could be used in place of mwlwifi?

Chadster766, 2019-03-02:
@davidc502 I haven't read many wireless issues with the OpenWRT stable release.

I recommend you test with OpenWRT Stable's defaults for a comparison.

yuhhaurlin, 2019-03-02:
Busy on 11ax.

closed 2019-03-02 by yuhhaurlin

BrainSlayer, 2019-03-02:
so working on a new chipset. all other older chipsets are out of maintainance and out of support. just to make clear

ryan-dev, 2019-03-02:
Wow, that is a pretty bold statement.

Marvell hasn't even made any wifi chipset for any open source driver 100%
stable as of late but in the end, merely forgetting the past mistakes and
just make a new chip for 11ax that will obviously be just as unstable.

I won't be buying any further device that has a Marvell chip inside it.

yuhhaurlin, 2019-03-02:
Unless there is major issue which needed to be fixed immediately.

tapper82, 2019-03-02:
We don't cair about 11ax We cant even use 5 ghz radios we have unless we keep them on chan 36. What about DFS?

yuhhaurlin, 2019-03-02:
If your environment has radar signal, DFS will work and move the channel to safe one.

cilix-lab, 2019-03-02:
@tapper82 I’ve been running a custom Ubuntu on my WRT3200ACM for years now and I’ve had no problem with DFS. I have an EU router and it gave me some headaches, but I got it working eventually. MU-MIMO never worked nor was developed, though.

I bought this router because I’d owned other Linksys routers in the good old (Cisco) days, but, even though it’s great hardware, I won’t be buying routers from Linksys anymore because of this support-less chips they’re using on them. 

They (Linksys) continue selling this line of routers as “Open Source Ready”, with no support at all and far from ready. I can’t believe there’s no class action lawsuit or something of the sort against Linksys for claiming such a lie.

yuhhaurlin, 2019-03-02:
The driver can work except for interoperability issue with few clients. MU-MIMO is not ready for all open source AP routers.

PalebloodSky, 2019-03-02:
Had expected better support from Marvell, including feature and performance with mvebu that is consistent with the closed source drivers. You told us you were working on MU-MIMO almost a year ago and it never happened.  802.11ax / Wi-Fi 6 is pointless for the average user, and I wouldn't buy a Marvell based router again after this.

Appreciate you providing an update though @yuhhaurlin at least now we know what to expect.

davidc502, 2019-03-03:
Yes, thank you for the work you have put into this project.

BrainSlayer, 2019-03-03:
@yuhhaurlin: the interoperability issue alone is a major issue and a good reason not to stop any maintainance. marvell is the far only one chipset vendor with such issues right now and its the only vendor who constantly refuses to fix known issues. consider that these issues have been reported 2 years ago already. so your argument "unless there is a major issue" is really a bad joke. marvell is really working hard on loosing all trust and getting the brand of a "never buy" vendor.

danny30au, 2019-03-03:
@yuhhaurlin Thanks for your comment. I have reported Marvell to the ACCC that there product is not fit for purpose. Hopefully it get moved from Australian shops.

"Consumer guarantees applying to goods. Businesses that sell goods guarantee that those goods: ... are fit for any purpose that the consumer made known to the business before buying (either expressly or by implication), or the purpose for which the business said it would be fit for. have been accurately described." 

https://www.accc.gov.au/business/treating-customers-fairly/consumers-rights-obligations

I hope the ACCC takes legal action against Marvell for lieing to there customers.

PalebloodSky, 2019-03-04:
Marvell should also be reported to the Better Business Bureau (BBB):
https://www.bbb.org/

BrainSlayer, 2019-03-04:
@PalebloodSky its the same for most mobile phone vendors. you get the phone and 6 months later the support has been stopped just to sell you some new shit

PalebloodSky, 2019-03-04:
> @PalebloodSky its the same for most mobile phone vendors.

This problem is really bad not just for Marvell, but for Linksys too. There hasn't been a single oem firmware update for their flagship routers, the WRT3200ACM and WRT32X in over a year.

What about the interoperability issue though? Personally I haven't had problems at all on my WRT32X lately. Wondering if someone opened a ticket and carefully detailed the issue with devices, logs, etc. maybe someone would address it.

BrainSlayer, 2019-03-04:
@PalebloodSky: you can easily find it. all ESP chipset based IOT solutions simply wont work, except you disable wmm and 802.11n capabilties. . about the WRT1900/1200 series its even more dramatic. these router are reported to have unstable wifi. so wifi just stops working at some point. and linksys is responsible here to take some actions. they choose marvell as vendor and linksys/belkin has the power todo some actions here. but from the latest news foxconn wanted to aquire belkin, so finally its soon to be a chinese company with all the good benefits of a socialist country. (btw i'm from east germany)

PalebloodSky, 2019-03-05:
@BrainSlayer Then it's pathetic that hasn't been resolved by Marvell, sounds like a major issue. I don't have those devices so that's why mine has been working fine. If Linksys/Belkin is going to sell to Foxconn I'm totally done with them.

This just makes me want 1 router as powerful as the WRT32X that would receive years of open source support from OpenWrt or DD-WRT without any major issues. Would easily pay double the price of the device for something like that. Until then my WRT32X has been working great on OpenWrt, especially considering it's only $100 USD refurbished now. Still using my R7000 on DD-WRT for another network as well. My long-term worry about DD-WRT is you're only 1 person and you do so much with that firmware, if you decide to stop development it's OpenWrt or nothing.

p.s. from the US, but I visited Berlin last summer, great city and beer your people have 👍

BrainSlayer, 2019-03-05:
i'm not alone. its just looking like that. but yes. without me there would be likelly no future. but i'm doing it since 2005. so there is a statistical chance that it will go on. 

btw:
berlin is a hipster city :-) there is far more to see in germany

dengqf6, 2019-03-07:
And it was I who pushed that commit, only to enable kernel 4.15+ support.

Alas.
Here we go again

[image]
[image]
[image]

dengqf6, 2019-03-07:
@danny30au
> I hope the ACCC takes legal action against Marvell for lieing to there customers

FTFY: **lying** to **their** customers

BrainSlayer, 2019-03-07:
we are not the customer of marvell. we are the customer of linksys and linksys is the customer of marvell