kaloz/mwlwifi #355: Crash after Country Code change

tapper82, 2019-03-16:
Hi I changed my Country Code to GB and I get a log full of this:
WRT3200acm 
Firmware Version
OpenWrt SNAPSHOT r9589-2407b1edcc / LuCI Master (git-19.069.61084-db98461) 
Kernel Version
4.14.105
mwlwifi-firmware-88w8964
2018-12-18-c2c8244d-1

Sat Mar 16 20:14:38 2019 kern.err kernel: [107775.640305] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Mar 16 20:14:38 2019 kern.err kernel: [107775.646268] ieee80211 phy0: return code: 0x001d
Sat Mar 16 20:14:38 2019 kern.err kernel: [107775.650908] ieee80211 phy0: timeout: 0x001d
Sat Mar 16 20:14:38 2019 kern.debug kernel: [107775.655216] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Sat Mar 16 20:14:42 2019 kern.err kernel: [107779.661682] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Mar 16 20:14:42 2019 kern.err kernel: [107779.667644] ieee80211 phy0: return code: 0x001d
Sat Mar 16 20:14:42 2019 kern.err kernel: [107779.672285] ieee80211 phy0: timeout: 0x001d
Sat Mar 16 20:14:42 2019 kern.debug kernel: [107779.676583] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Sat Mar 16 20:14:46 2019 kern.err kernel: [107783.683059] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Sat Mar 16 20:14:46 2019 kern.err kernel: [107783.689020] ieee80211 phy0: return code: 0x001d
Sat Mar 16 20:14:46 2019 kern.err kernel: [107783.693659] ieee80211 phy0: timeout: 0x001d
Sat Mar 16 20:14:52 2019 kern.debug kernel: [107789.706204] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000

tapper82, 2019-03-16:
BTW my router works fine when the country code is set to FR

jeolivera, 2019-03-18:
Don't touch the country code. There isn't any support for changing it to any other one that isn't hardcoded on first boot. There were plans to fix it so that it would let you change it. But, the main developer stopped working on this driver and is now busy developing a driver for Marvell's 802.11ax (WiFi 6) chipset. 

GB (despite being under OFCOM regulation) still follows (most) of the EU (ETSI) regdomain rules. However, the name "EU" is a reserved country code (that is somehow hardcoded in the regulatory EEPROM in the Linksys AC series). Hence, we get the current hacks that we have in place in this driver.

_**Upstreamed Marvell mwifiex driver hack with their reasoning**_
[image]

_**Mwlwifi's copy of that hack in this Github**_
[image]

In honesty, I haven't tried it myself...but you could maybe change this: https://github.com/kaloz/mwlwifi/blob/c2c8244d8fea5d59762cb14438ded00bf6d5965c/core.c#L130

From "FR" to "GB" and it would allow you to set "GB" as a country without the kernel panicking because they have the same regdb values. 

[image]

Of course, you'd have to recompile this driver yourself though...

jeolivera, 2019-03-20:
If it is of interest to you, I did end up testing the modification I suggested. It works (no kernel errors!), but it is entirely cosmetic (the bands and frequencies available remain the same). 

[image]

[image]
_(Image shows my Mac connected to my wired repeater which has issues setting it's regdomain :/ )_

jeolivera, 2019-03-20:
And more proof:

[image]

yuhhaurlin, 2019-03-21:
WRT3200ACM is programmed with region code and power table. Please don't change country code and use the default one.

closed 2019-03-21 by yuhhaurlin