kaloz/mwlwifi #356: How to enable both bands

tiagogaspar8, 2019-03-17:
Hey!

I'd like to enable both bands in both radios. I saw the documentation on how to proceed.
Also, I'd like to enable both of them when building the system.

Thanks for the help!

eduperez, 2019-03-19:
There is one radio that can do 5GHz, one radio that can do 2.4GHz, and one radio that can do either but not both at the same time.

tiagogaspar8, 2019-03-19:
But for example, on my router, WRT3200ACM it has two Marvell 88W8964, can't they both do 2.4GHz and 5Ghz? I don't mean each chip do 2.4 and 5 at the same time, I mean both do 5GHz.
In the driver documentation it says this:

> DTS parameters for mwlwifi driver (pcie@X,0):
> 
> #Disable 2g band
> marvell,2ghz = <0>;
> 
> #Disable 5g band
> marvell,5ghz = <0>;
> 
> #Specify antenna number, default is 4x4. For WRT1200AC, you must set these values to 2x2.
> marvell,chainmask = <4 4>;
> 
> #Specify external power table. If your device needs external power table, you must provide the power table via this parameter, otherwise the Tx power will be pretty low.
> marvell,powertable

Also, on system logs I see a message saying that 2g got disabled in one radio and 5g got disabled in the other one.
So shouldn't it be possible to unlock both bands in both radios with a simple driver configuration?

yuhhaurlin, 2019-03-21:
Only one for 2.4 g and one for 5 g.

closed 2019-03-21 by yuhhaurlin