kaloz/mwlwifi #357: WRT1200AC,5GHz ac DFS broken

Massimo-B, 2019-03-21:
Hello,

I'm using Openwrt on a Linksys WRT1200AC. Using DFS channels no client can join the network. In the Openwrt forum I was told this is caused by a broken DFS on this hardware:

https://forum.openwrt.org/t/wrt1200ac-ac-wireless-stopped-working/33675

yuhhaurlin, 2019-03-21:
DFS should work. It needs to wait for CAC (60 seconds) before AP is working.

closed 2019-03-21 by yuhhaurlin

tapper82, 2019-03-21:
@yuhhaurlin  DFS does not work rite. eg on my wrt3200acm, when ever I switch to a DFS channel all trafic just stops. The same on my 1900ac v2. If it worked rite why would so menny people be complaning about it?

yuhhaurlin, 2019-03-21:
If you set channel to DFS channel, it needs CAC (60 seconds) before AP is working. You can check if SSID is showed up. After that, client can connect to AP.
If radar signal is detected, hostapd will switch channel to the save one. AP will notify client before channel is changed.

Massimo-B, 2019-03-21:
@tapper82, can we do something on Openwrt according to this information from @yuhhaurlin, to get this working?

jeolives, 2019-03-22:
In my testing with a 1900ACSv2-GB, DFS channels only work in channels 100-128 for 80MHz operation; Ch. 100-136 will work fine for 40Mhz; and Ch. 100-140 will work fine on 20MHz. This is because 80MHz will need to bond channels 132-144 (similar case with 140 needing to bond with 144 for 40MHz), BUT ch. 144 is unavailable in the EU/ETSI regions due to SRD @ 25mW. So, hostapd/mwlwifi errors out.

Follow these rules for GB/FR/DE/ETSI devices + not changing the country code on first boot and DFS will work. However, even if you get it working, DFS is incredibly sensitive and susceptible to false positives. It will try and attempt a channel switch usually 4-24 hours into operation - and even then it'd just throw you back to Ch. 36.

It's also a bit disheartening to see the third radio on the 3200ACM not being used for the intended purposes of DFS radar scanning (there is no special loop/function for the 3200ACM DFS operation) so that at least the 3200ACM retains one of it's core value propositions (DFS certification/stable DFS operation with proper noise calibration) on OpenWrt. But I do understand that that implementation may be proprietary for intellectual property or regulatory reasons.