kaloz/mwlwifi #359: WRT3200ACM no work 802.11s and ad-hoc

mikysal78, 2019-04-19:


PalebloodSky, 2019-07-03:
Adding that this is not just a WRT3200ACM issue but WRT32X, which would make sense given nearly the same hardware. It's a shame Marvel doesn't support mwlwifi with this fix, IOT fixes, and MU-MIMO this would be the best router on the market.

ratsputin, 2019-11-02:
This is why I'm ditching my two WRT3200ACMs and replacing them with Netgear Nighthawk X4S routers based on the Atheros chipset.  The WMM issues with ESP chips are causing me numerous problems and there has been no progress on this issue and not a peep out of the developer on GitHub for almost a year.  I'm done with Linksys and their abandoning the closed-source drivers.

davidc502, 2019-11-02:
The router is great.  Might have considered buying some nice AP's to complement.

ravenchan, 2019-11-02:
Agree.If we rip of the wireless card ,this thing is great.

mikysal78, 2019-11-02:
Il giorno sab 2 nov 2019 alle ore 01:25 davidc502 <notifications@github.com>
ha scritto:

> The router is great. Might have considered buying some nice AP's to
> complement.
>

Maybe you haven't tried applying OpenWRT!
Unsecured Mesh and Ad-Hoc!
A router that costs 30 euros works much better!

PalebloodSky, 2019-11-02:
Unfortunately there is no perfect OpenWrt router. Even the R7800 you mention has performance problems because so much relies on the NSS cores in the stock firmware and they don't work right in OpenWrt. I'd say the WRT3200ACM/WRT32X is the best router outside of Wi-Fi. It's a shame the driver support is dead. I wouldn't buy this product again but don't know where else to turn there is nothing really better.

ratsputin, 2019-11-03:
Along the lines of @mikysal78’s original issue, I was able to implement a WDS-based mesh network yesterday with zero issues using three Atheros-based routers.

@mikysal78, that may be a route to consider.  Implementation is simple.

mikysal78, 2019-11-03:
Il giorno dom 3 nov 2019 alle ore 12:13 ratsputin <notifications@github.com>
ha scritto:

> Along the lines of @mikysal78 <https://github.com/mikysal78>’s original
> issue, I was able to implement a WDS-based mesh network yesterday with zero
> issues using three Atheros-based routers.
>
> @mikysal78 <https://github.com/mikysal78>, that may be a route to
> consider. Implementation is simple.
>
Yes! with tp-link routers that cost 30 or 40 euros you make mesh or ad-hoc
without problems.
The WRT3200ACM which on the Amazon costs 195 euros does not work!
The anger is that an expensive router does not work as it should, a cheap
one is fine.

filu34, 2023-07-31:
Few years later, and still no available fix for that device?

closed 2026-10-29 by mikysal78