kaloz/mwlwifi #360: Is this driver still being worked on?

tapper82, 2019-05-13:
Hi are you still working on this driver? No updates for a long time. Is there any new fermwair?

thagabe, 2019-05-13:
firmware* and no this driver is "stable" and new features are only introduced for we products. Give up on mwlwifi and look for something better.

kubrickfr, 2019-05-15:
short answer: no.
According to its author, it works well enough and delivered what was promised. #delusional

danny30au, 2019-06-04:
@yuhhaurlin If your not going to work on this driver anymore. Can you please release the "firmware source code" I would class this driver and firmware as "abandoned"

BrainSlayer, 2019-06-04:
@danny30au : no he wont. its about patent and software license shit. its owned my marvell and marvell isnt interested in realeasing this firmware since it may increase software quality of old products and increasing quality of older products decreases sales of newer products of course

danny30au, 2019-06-04:
@BrainSlayer I know but you never know right! :) Hang on.... wait a min. Anyone with half a brain would go if we "release the REAL source code" WE might actually get more money/profit lol. They are incredible short sighted Marvell. They really are loosing customers and long term they will be loosing out, :)

BrainSlayer, 2019-06-04:
marvell isnt different from broadcom or other american vendors. its always the same shit. but from some of these its possible to get firmware codes like QCA but only under NDA

danny30au, 2019-06-05:
@BrainSlayer Little do the Yanks/America know that WE Australians invented Wi-Fi not the Yanks/America. :)

BrainSlayer, 2019-06-05:
@danny30au And we Germans try to make it perfect :-) for the invention if wifi i dont find much information about australia. the first wireless network was build in the 60s in hawaii and lucent sold wave lan in 1988 as product. thats all what i found. but teach me more if you have further informations about the invention of it.

aaron1959, 2019-06-05:
@danny30au  I invented the internet while I was living in American Samoa to avoid the high cost of foot messengers from village to village. We used spark gap transmitters at about 50 Baud.

BrainSlayer, 2019-06-05:
a peer to peer connection is surelly not the internet. there was a lot missing still. arp, tcp, udp, ip at all. services. and transmitting data over radio isnt new too. i mean the morse code is some sort of data transmission. and even in the second world war such technology was already used to transmit messages in electronic way

BrainSlayer, 2019-06-05:
@aaron1959 and btw. using a spark gap transmitter in the 60s was very old school even at that time

danny30au, 2019-06-05:
@BrainSlayer https://en.m.wikipedia.org/wiki/John_O'Sullivan_(engineer)

BrainSlayer, 2019-06-06:
so he invented a part of it to make it more robust. unfortunalty there isnt much written about his algorithm

tapper82, 2019-08-07:
@yuhhaurlin  any updates?

eduperez, 2019-09-13:
I did not see any development in that repo, other than the commit you posted; I guess @schm0 has just extracted the firmware from an update file or similar.

schm0, 2019-09-13:
eduperez is right. The only change in this repo is the firmware for 88W8864.
The firmware was extracted from the official Linksys firmware.
However, I'm trying to bring power adjustment for the V1 devices. (Testing on WRT1200)
But this is still work in progress.

smoe, 2020-02-21:
With 11s mesh networks not possible I do not consider this driver to be feature complete (https://github.com/kaloz/mwlwifi/issues/359).

SamJongenelen, 2021-04-26:
Another person stuck with a 3200acm without working 802.11s. I guess i should get Atheros hardware?

smoe, 2021-04-26:
@SamJongenelen I opted for the Netgear Nighthawk r7800 - meshes nicely with 802.11s under Gluon that is based on OpenWrt stable. Has more memory, too, and computes faster despite a slightly lower frequency.

filu34, 2023-08-01:
> With 11s mesh networks not possible I do not consider this driver to be feature complete (#359).

I agree. Mesh starts to become a standard. So basically driver is still broken.

kubrickfr, 2023-08-01:
I don't think that either Marvell or Linksys has ever claimed that this chip/router has mesh capabilities with their proprietary drivers. Their commitment was to get to feature parity between the two for the WRT routers. And it was a pretty vague commitment at that.

I've been using the WRT3200ACM  for many years, and performance and stability are excellent.

Also, AFAIK this chip and routers are EOL, so you'll get nothing by complaining here.

smoe, 2023-08-01:
My expectations are low. Should have sold my WRT3200ACM a long time ago :-) Also, I do not know where to complain. One should sue them and likely win - but for what - to ensure that no other company will ever care to promote an Open Source device? This would be a https://en.wikipedia.org/wiki/Pyrrhic_victory much like the PlayStation 3 one https://www.polygon.com/2016/10/12/13261780/ps3-otheros-linux-lawsuit-settlement-claims . I still hope for some "yeah, it is all EOL now, here are the details of our chips for you to program against"-kind of reaction.