kaloz/mwlwifi #361: Slow speed on 2.4 ghz N mode faster on g mode.

xromansx, 2019-05-17:
Hi,
 5Ghz working good, 2.4Ghz N not stable, connection shows 144Mbps during the test real speed max 5-10Mbps when I switch to Legacy connection drops to 54Mbps but I can get up to25Mbps, download speed, Upload speed is always excelent only download problem.
I have the similar issue with Original firmware and with Dd-wrt, did someone had same problem ? Thanks
Below is my router information.
Linksys WRT1900ACv2
OpenWrt 18.06.2 r7676-cddd7b4c77 / LuCI openwrt-18.06 branch (git-19.020.41695-6f6641d)
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181114
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:13:8d:f6
2g: enable
5g: disable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000007
radio: enable
iobase0: e1100000
iobase1: e1280000
tx limit: 768
rx limit: 64
qe trigger number: 281366

BrainSlayer, 2019-05-29:
mbit or mbyte? i mean with 54g its indeed possbible to get about 32 mbit download/upload. with 144 802.11n depending on the conditions you get about 90-100 mbit max. both values are just samples for really perfect conditions which do not exist in these days anymore

RomSand, 2019-05-29:
mbit, I think 802.11n has and issue, is  shown 144 in connection details but speed goes to 10mbit max almost for any device...

BrainSlayer, 2019-05-30:
from your description it seems like one of those typical compatiblity issue with your client device(s) so simpy bugs in the marvell firmware. the problem is just that marvell refuses to do any support for these chipsets. they do not fix bugs and ignore all problems since a very long time.

RomSand, 2019-05-30:
I wave this problem from all my devices almost... To bad, would be a nice router... I'll take wrt3200 from one of my friends and test speed with it...

RomSand, 2019-06-06:
Same thing on wrt3200. did someone get higher speeds on this routers on 2.4 Ghz ?

RomSand, 2019-06-07:
Was told by linksys that 2.4Ghz part of the chip is broken and to return it to store )

ratsputin, 2019-11-02:
For anyone encountering this issue, make sure you have WMM turned on (checked) on the wireless settings.  Unfortunately, this will break many ESP-based IoT devices; however...