kaloz/mwlwifi #362: 802.11w - "Optional PMF" doesn't work properly on 88w8864

jeolives, 2019-06-10:
Hello.

I've been facing some issues with 802.11w with PMF set to "Optional" (hostapd.conf with `ieee80211w=1` for a while now. I've tested several different configurations in both the hardware and the software.

When I make 802.11w optional, devices that don't support PMF and devices that do support it should still be able to communicate with each other whilst concurrently connected on the same access point. This does not happen. 

In an environment where all the clients/stations do not support PMF, they can connect to the AP with PMF-Optional, they pass the 4-way handshake and receive a DHCP address. They are visible in the router's ARP table and can be pinged from the router. Devices attached on the router's switch (via ethernet) can communicate with the printer. **HOWEVER**, as soon as a station that is **PMF CAPABLE** joins the _AP with `ieee80211w=1`_, all communication with the station that does **NOT** support PMF ceases. It cannot be pinged from other clients/stations, but the entry for it still exists on the router's ARP table and it is still on the associated stations list. 

### What I've tried: 
- OpenWrt Master, OpenWrt 18.06
- Hostapd 2.6, 2.7
- wpad, wpad-openssl
-- both with hostapd-utils

### Devices:
Macbook Pro Mid-2012 non retina, Macbook Air Early-2015 (PMF Capable)
iPhone 6s (PMF Capable)
**AFFECTED** Brother MFC-J200 Printer (Does not support PMF)
**AFFECTED** Techage Yoosee 1080p Camera (Does not support PMF)

### Protocols:
Seems to affect Apple's Airprint/Bonjour MLD & other more standard multicast protocols.

tiagogaspar8, 2019-10-04:
I've been having the same issue since I bought this router.
Now that WPA3 is coming and it REQUIRES 802.11w this router will stop in time just because of this bug.
We really need someone to fix this bug.

tiagogaspar8, 2019-12-23:
No updates... sadly...

zorxd, 2020-06-11:
Thanks, just saved a lot of time. I seem to have the same issue. WRT1900acs openwrt 19.07.3

phineasthecat, 2020-08-18:
Still no updates on this bug? Enabling ieee80211w=1 on either radio on my WRT1900AC v1 breaks ARP/broadcast for all wireless clients that _do not support PMF_ for me as well. Also breaks Bonjour, or anything else that uses mDNS, between wireless clients. This is in pure AP mode, no routing, no DNS, no DHCP, all LAN ports and wireless networks bridged.

davidc502, 2020-08-18:
Wifi drivers were abandoned over 1.5 years ago.

intgr, 2022-06-29:
I found a work-around that seems to work for me. Create two separate wireless interfaces -- one for legacy IoT devices without PMF support (using `ieee80211w=0` and WPA2 only), and another interface for modern clients (using `ieee80211w=2` and WPA3 only).

I only have legacy clients on 2.4 GHz, not sure if there's any use case for separate interfaces for 5 GHz radio as well.

[image]

tiagogaspar8, 2022-06-29:
Can you actually connect in wpa3 mode?

regulskimichal, 2022-06-29:
WPA3 works for me on MacOS 12.4 with optional Management Frame Protection
OpenWrt 21.02.3 r16554-1d4dea6d4f
mvebu/cortexa9
Linksys WRT1200AC

tiagogaspar8, 2022-06-29:
What if it is not optional?

regulskimichal, 2022-06-30:
With disabled Management Frame Protection it does not work.

With optional Management Frame Protection it looks like this on my Mac:
[image]

tiagogaspar8, 2022-06-30:
I was saying with mandatory, because the last time I tested wpa3 with this device it did not work

regulskimichal, 2022-06-30:
Sorry, I did not understand what you mean. With Management Frame Protection setting set as Required it also works for me but I can not say anything certain about the stability of the connection. I will use it for some time from now and I will try to share with you with a feedback.

regulskimichal, 2022-06-30:
I have 2 different SSIDs, one for each AP (2.4GHz and 5GHz)