kaloz/mwlwifi #364: WiFi drops and after it is back on, broadcast is not working

dolohow, 2019-06-23:
I am using ` OpenWrt 18.06.3, r7798-97ae9e0ccb` and when device is powered on, after about 15 minutes, WiFi drops and when it is back on, broadcast is not working.  Means I cannot reach other devices in the network.

```
[ 1186.813737] WARNING: CPU: 1 PID: 5429 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1414 0xbf0870e8 [mac80211@bf082000+0x5e
000]                                                                                                                               
[ 1186.825722] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time
 xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn x
t_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_C
T xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_t
able_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt 
sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingr
ess mwlwifi mac80211 cfg80211
[ 1186.897174]  compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb
 tun gpio_button_hotplug
[ 1186.910921] CPU: 1 PID: 5429 Comm: hostapd Not tainted 4.14.128 #0
[ 1186.917133] Hardware name: Marvell Armada 380/385 (Device Tree)
[ 1186.923086] Function entered at [<c010ecac>] from [<c010a980>] 
[ 1186.928949] Function entered at [<c010a980>] from [<c0627a88>] 
[ 1186.934812] Function entered at [<c0627a88>] from [<c01217f0>] 
[ 1186.940674] Function entered at [<c01217f0>] from [<c01218bc>] 
[ 1186.946536] Function entered at [<c01218bc>] from [<bf0870e8>] 
[ 1186.952410] Function entered at [<bf0870e8>] from [<bf09df24>] 
[ 1186.958276] Function entered at [<bf09df24>] from [<bf065604>] 
[ 1186.964142] Function entered at [<bf065604>] from [<c055d050>] 
[ 1186.970005] Function entered at [<c055d050>] from [<c055c230>]                                                         [63/9796]
[ 1186.975867] Function entered at [<c055c230>] from [<c055cd58>] 
[ 1186.981729] Function entered at [<c055cd58>] from [<c055ba04>] 
[ 1186.987590] Function entered at [<c055ba04>] from [<c055bea8>] 
[ 1186.993451] Function entered at [<c055bea8>] from [<c050b2f8>] 
[ 1186.999313] Function entered at [<c050b2f8>] from [<c050b82c>] 
[ 1187.005175] Function entered at [<c050b82c>] from [<c050c634>] 
[ 1187.011037] Function entered at [<c050c634>] from [<c01076e0>] 
[ 1187.016912] ---[ end trace 1c7c2cbf6fb9e31a ]---
[ 1187.021829] ------------[ cut here ]------------
[ 1187.026489] WARNING: CPU: 0 PID: 5429 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1414 0xbf0870e8 [mac80211@bf082000+0x5e
000]
[ 1187.038483] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time
 xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn x
t_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_C
T xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_t
able_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt 
sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingr
ess mwlwifi mac80211 cfg80211
[ 1187.109985]  compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb
 tun gpio_button_hotplug
[ 1187.123756] CPU: 0 PID: 5429 Comm: hostapd Tainted: G        W       4.14.128 #0
[ 1187.131190] Hardware name: Marvell Armada 380/385 (Device Tree)
[ 1187.137142] Function entered at [<c010ecac>] from [<c010a980>] 
[ 1187.143004] Function entered at [<c010a980>] from [<c0627a88>] 
[ 1187.148867] Function entered at [<c0627a88>] from [<c01217f0>] 
[ 1187.154729] Function entered at [<c01217f0>] from [<c01218bc>] 
[ 1187.160591] Function entered at [<c01218bc>] from [<bf0870e8>] 
[ 1187.166457] Function entered at [<bf0870e8>] from [<bf09df24>] 
[ 1187.172323] Function entered at [<bf09df24>] from [<bf065604>] 
[ 1187.178189] Function entered at [<bf065604>] from [<c055d050>]                                                         [33/9796]
[ 1187.184050] Function entered at [<c055d050>] from [<c055c230>] 
[ 1187.189912] Function entered at [<c055c230>] from [<c055cd58>] 
[ 1187.195773] Function entered at [<c055cd58>] from [<c055ba04>] 
[ 1187.201635] Function entered at [<c055ba04>] from [<c055bea8>] 
[ 1187.207497] Function entered at [<c055bea8>] from [<c050b2f8>] 
[ 1187.213359] Function entered at [<c050b2f8>] from [<c050b82c>] 
[ 1187.219221] Function entered at [<c050b82c>] from [<c050c634>] 
[ 1187.225083] Function entered at [<c050c634>] from [<c01076e0>] 
[ 1187.231045] ---[ end trace 1c7c2cbf6fb9e31b ]---
[ 1187.235856] ------------[ cut here ]------------
[ 1187.240524] WARNING: CPU: 0 PID: 5429 at backports-2017-11-01/net/mac80211/ieee80211_i.h:1414 0xbf0870e8 [mac80211@bf082000+0x5e
000]
[ 1187.252512] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time
 xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn x
t_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_C
T xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_t
able_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt 
sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingr
ess mwlwifi mac80211 cfg80211
[ 1187.324010]  compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb
 tun gpio_button_hotplug
[ 1187.337762] CPU: 0 PID: 5429 Comm: hostapd Tainted: G        W       4.14.128 #0
[ 1187.345194] Hardware name: Marvell Armada 380/385 (Device Tree)
[ 1187.351146] Function entered at [<c010ecac>] from [<c010a980>] 
[ 1187.357008] Function entered at [<c010a980>] from [<c0627a88>] 
[ 1187.362869] Function entered at [<c0627a88>] from [<c01217f0>] 
[ 1187.368731] Function entered at [<c01217f0>] from [<c01218bc>] 
[ 1187.374593] Function entered at [<c01218bc>] from [<bf0870e8>] 
[ 1187.380460] Function entered at [<bf0870e8>] from [<bf09df24>] 
[ 1187.386326] Function entered at [<bf09df24>] from [<bf065604>]                                                          [3/9796]
[ 1187.392191] Function entered at [<bf065604>] from [<c055d050>] 
[ 1187.398053] Function entered at [<c055d050>] from [<c055c230>] 
[ 1187.403915] Function entered at [<c055c230>] from [<c055cd58>] 
[ 1187.409777] Function entered at [<c055cd58>] from [<c055ba04>] 
[ 1187.415639] Function entered at [<c055ba04>] from [<c055bea8>] 
[ 1187.421501] Function entered at [<c055bea8>] from [<c050b2f8>] 
[ 1187.427361] Function entered at [<c050b2f8>] from [<c050b82c>] 
[ 1187.433223] Function entered at [<c050b82c>] from [<c050c634>] 
[ 1187.439085] Function entered at [<c050c634>] from [<c01076e0>]
[ 1187.445057] ---[ end trace 1c7c2cbf6fb9e31c ]---
[ 1188.959552] ieee80211 phy0: change: 0x100
[ 1188.972645] ieee80211 phy0: change: 0x60
[ 1189.159683] ieee80211 phy0: change: 0x40
[ 1189.339857] ieee80211 phy0: change: 0x60
[ 1189.407945] ieee80211 phy0: change: 0x100
[ 1189.419951] ieee80211 phy0: change: 0x100
[ 1189.433010] ieee80211 phy0: change: 0x62
[ 1189.503026] ieee80211 phy0: change: 0x40
[ 1220.534290] ieee80211 phy0: radar detected by firmware
[ 1220.539477] ieee80211 phy0: change: 0x40
[ 1220.607667] ieee80211 phy0: change: 0x100
[ 1220.620377] ieee80211 phy0: change: 0x100
[ 1220.633437] ieee80211 phy0: change: 0x42
[ 1220.771858] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1220.778345] br-lan: port 2(wlan0) entered blocking state
[ 1220.783708] br-lan: port 2(wlan0) entered forwarding state
[ 1220.872375] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[ 1220.975100] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
```

yzhelezko, 2019-10-14:
I have the same problem
OpenWrt 18.06.4 r7808-ef686b7292
Linksys WRT32X

davidc502, 2019-10-16:
Known issues with using DFS channels. I recommend staying off of them to avoid this problem.

[ 1220.534290] ieee80211 phy0: radar detected by firmware     <<<<< You will only see this in the log if a DFS channel is selected.

dolohow, 2020-02-29:
> Known issues with using DFS channels. I recommend staying off of them to avoid this problem.
That worked, but it is a workaround.  Closing...

Thanks

closed 2020-02-29 by dolohow