kaloz/mwlwifi #365: Ubuntu?

jrosetto, 2019-06-24:
Is there any way to make this work on Ubuntu?  The stock driver doesn't support monitoring mode and this seems like the only other option I can find.

aaron1959, 2019-06-25:
Ubuntu? how wouls an operating language magically change the driver?

Chadster766, 2019-06-25:
https://github.com/Chadster766/McDebian

jrosetto, 2019-06-26:
> 
> 
> Ubuntu? how would an operating language @ the UI level magically change the driver?

This is source code.  Theoretically if I add all the required dependencies and cross-compile from arm arch I should be able to make this work.

Thanks for the asinine response though.  The @ symbol and abbreviation for user interface really sold it.

If you don't have anything to contribute... Don't respond!

jrosetto, 2019-06-26:
> 
> 
> https://github.com/Chadster766/McDebian

Thanks, I will look further into this.

colzin, 2026-10-30:
Hi,
I recently installed Linux Mint on a Microsoft Surface Book. I ran 'sudo airmon-ng start wlp3s0', which responded with this message:
phy0    wlp3s0    mwifiex_pcie    Marvell Technology Group Ltd. 88W8897 [AVASTAR] 802.11ac wireless
ERROR adding monitor mode interface: command failed: Operation not supported (-95)
I would like to have monitor mode in this built-in adapter. Googling brought me to this repo. It looks like this may be a driver for DD/Open WRT linux. 
Does anyone have any guidance on porting to Linux Mint? I believe it is built on Ubuntu...
Thanks!