kaloz/mwlwifi #366: 8997 set region not working

borislevin1984, 2019-06-30:
Hello,

I have tried to change region with iw application and hostapd.

In hostapd attempt to change region fails after timeout.

With iw application you can see below there is just no effect for the industrial module:

Please see iw output:

**iw reg get**
global
country 98: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), NO-OUTDOOR, DFS, AUTO-BW

phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

# **iw reg set GB**

iw reg get
global
country 98: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), NO-OUTDOOR, DFS, AUTO-BW

phy#0
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

Is there any issue with changing region with this chipset?

Thanks.

ValCher1961, 2019-06-30:
I think this driver does not allow to change the region as well as on chips 8864/8964.

borislevin1984, 2019-06-30:
I have another Marvell chipset 8897 and with mwlwifi driver region change works perfectly.
Only in 8997/8864/8964 region change will not work?

ValCher1961, 2019-06-30:
Try this patch - https://github.com/ValCher1961/McDebian_WRT3200ACM/blob/master/kernel-4.19.X/regfree.patch . I use it to change the region to 8964. I do not know if it is suitable for 8997.

borislevin1984, 2019-06-30:
Thanks! i will try it

borislevin1984, 2019-06-30:
After going over your patch it will not solve the problem for 8997 chip...

ValCher1961, 2019-06-30:
Sorry, I do not have a card with 8997 chip

porsche613, 2019-07-07:
The same issue happened at 88w8864 chipset in WRT1900ACSv2, but I may have to install OpenWRT Snapshot to try the patch.

ValCher1961, 2019-07-07:
I can't verify this patch, since I don't have 8864, but you can try.

porsche613, 2019-07-09:
Another way can be tried, reghack (and v2), can be found at [this reposity](https://github.com/Luminger/reghack). There are discussing in OpenWrt fourm, [thread 1](https://forum.openwrt.org/t/wrt1900acs-v2-how-to-update-mwlwifi-driver/1491/16), [thread 2](https://forum.openwrt.org/t/linksys-wrt1900ac-v2-reghack/18394).

borkra, 2019-09-04:
Sounds like you purchased router for US market. And yes it will not work anywhere else.
Modern wireless certification especially US one, require AP manufacturer to make sure that router transmit only on allowed frequencies and with power values for the country, and prevent users from altering them.

borislevin1984, 2019-09-04:
With mwifiex driver activated in AP mode region change working as expected so i'm pretty sure this is mwlwifi limitation.

borkra, 2019-09-04:
Yes, mwiflex is client driver it allowed to change region using 802.11d. 
But for AP it is prohibited. I can refer you to appropriate FCC standards if you are interested..

borislevin1984, 2019-09-04:
Sorry, what do you mean "for AP it is prohibited"?

borkra, 2019-09-04:
Per FCC regulations (and if I am not mistaken now days ETSI as well), it is illegal for AP manufacturer to permit user selection of transmission frequency or power level outside permitted. 

Permits are issued by the government agency on individual basis. Usually most actual compliance work outsourced to 3rd party labs qualified by the government, but the final permit is issued by the government agency.

Permitted transmission and power level are validated during device certification to meet standards and other regulation and listed on the certificate for the device. This certification is per country as each country have different allocation for Wireless spectrum.

So yes it is driver "issue", and it was done this way to meet government regulations.

borislevin1984, 2019-09-04:
I'm not sure i understand you, i have 2 Marvell chip sets 8997 and 8897:
1. with mwlwifi driver i can change region on 8897 but not on 8997.
2. with mwifiex driver i can change region on both.

What i understand from you that the reason i can change region on

derosier, 2019-09-04:
The 8997 (and 8897) are client-mode chips and mwlwifi was specifically
intended for the 8864 and 8964 AP mode chips. The fact this driver works
with the two client mode chips is incidental and this driver doesn't really
_support_ them. You should be using mwifiex for these chips.

Or you should be using the driver your module manufacturer requires
(assuming they're not being lazy and recommending this driver - if they do
that they're wrong and you should find a new vendor). For example, I know
the Laird ST60 series modules (based on 8997) are supported by Laird's own
driver (based on Marvell's vendor driver but heavily modified). Also, I
know that in the 89xx chips, there's been lots of work done to make it
difficult or impossible to change the power settings. On the newer APs that
this driver supports, the power tables are hardwired in eeprom on the
device. I know with the mentioned Laird modules, said power tables are
likewise locked in hardware and this was due in part to Marvell pushing
them to this design decision.

You aren't going to get any movement on this issue with this driver. If you
go back and read the closed issues, eventually you'll stumble upon the
maintainer (who is from Marvell) saying he won't/can't do anything about it.

- Steve

borkra, 2019-09-04:
For 8997 you need to contact your module manufacturer for proper driver and firmware, and EEPROM configuration and AP certification for your country.

And yes it's a custom work that your Wi-Fi module provider together with Marvell can do. And likely be expensive.