kaloz/mwlwifi #367: Incorrect channel 5ghz display

olimpiumarius, 2019-08-09:
Hello,
Router wrt1900acs v2
Kernel Version	4.19.62
driver version to 10.3.8.0-20181210.
From what you can see if i choose channel 40, which i use too, he continues to show me channel 36.
Has anyone else encountered this problem?
[image]
[image]

lantis1008, 2019-08-11:
Channel 40 @ 80MHz can be labelled as a few different things depending on the implementation.
To be honest, 40 is not a label i would consider "valid". I would consider 36 (first channel) and 42 (middle channel) the correct label for that part of the spectrum.

You can also see in the following code how your channel input is translated to 42, which is actually what hostapd expects.
```
    case "$(( ($channel / 4) % 4 ))" in
        1) idx=$(($channel + 6));;
        2) idx=$(($channel + 2));;
        3) idx=$(($channel - 2));;
        0) idx=$(($channel - 6));;
    esac
```
```
40 / 4 = 10
10 % 4 = 2

:. 40 + 2 = 42
```

olimpiumarius, 2019-08-11:
@lantis1008 
I appreciate you responding to this message, i tried to check what you recommended to me.
It seems that channel 42 does not appear in the 5Ghz menu, but i checked channel 44, 48, 52 and the problem does not persist!
I am currently running OpenWrt with Kernel Version 4.19.65 and the problem persists on channel 40!

lantis1008, 2019-08-11:
As long as the Wi-Fi us working, this is a visual bug at best, and not related to mwlwifi. 
I don't know what behaviour LUCI tried to give.

BrainSlayer, 2019-08-12:
> Capture10
> Capture9

ht40 / vht80?

if yes. check the side channel and center channels

olimpiumarius, 2019-08-12:
@BrainSlayer : 
Hello,
Sorry to delay the message, only now did I see the message, i worked at night... 
The problem only appears on 40Mhz and 80Mhz, it looks like 20Mhz is working correctly!

olimpiumarius, 2019-08-12:
I also found a problem with channel 136 watching the pictures!
[image]
https://media.discordapp.net/attachments/590608888504647706/609803554915614723/Capture.JPG?width=929&height=821

olimpiumarius, 2020-09-26:
It seems that this problem no longer persists in the latest drive versions!

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a

closed 2020-09-26 by olimpiumarius