kaloz/mwlwifi #370: Problem changing the signal power wi-fi

olimpiumarius, 2019-08-20:
Hello,

Router Linksys WRT1900ACS
Kernel Version	4.19.66 SNAPSHOT r10815-fd4804ce33 / LuCI Master (git-19.228.36069-f3bcb91)

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address
2g: enable
5g: disable
antenna: 4 4
irq number: 49
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: 58325fdd
iobase1: 2f18e395
tx limit: 768
rx limit: 64
qe trigger number: 5607907

If I want to change the signal power transmission from the Luci web interface, it has no effect, the value remains still set on Auto.
The problem with channel display also persists!

[image]

kubrickfr, 2019-08-30:
1) Unfortunately this driver seems to be dead
2) No #332, no #291 and no #315. An many more times no.

borkra, 2019-09-04:
Max power values are fixed at the certification time, any change to them even though it is technically a software change, require re-certification of the product, which is expensive and time consuming. I doubt Linksys will go for it on a legacy product. But you can ask Linksys support if they are interested.

tiagogaspar8, 2019-10-04:
This power limit is related to the channel number, non DFS channels have low power.
If you issue the command `iw phy0 info` (I believe it's this one, but it won't vary much from that) you can see each channel and it's power limit.

borkra, 2019-10-04:
In addition 
power limits depend on a country one is in.

power limits often reduced by manufacturers when they have hardware design issues most common ones excessive signal bleed into nearby channels and excessive antenna gain.

olimpiumarius, 2020-01-31:
root@swrt:~# iw phy0 info
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x106f
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-23, 32
                VHT Capabilities (0x33837930):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        SU Beamformer
                        SU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (radar detection)
                        * 5280 MHz [56] (20.0 dBm) (radar detection)
                        * 5300 MHz [60] (20.0 dBm) (radar detection)
                        * 5320 MHz [64] (20.0 dBm) (radar detection)
                        * 5500 MHz [100] (27.0 dBm) (radar detection)
                        * 5520 MHz [104] (27.0 dBm) (radar detection)
                        * 5540 MHz [108] (27.0 dBm) (radar detection)
                        * 5560 MHz [112] (27.0 dBm) (radar detection)
                        * 5580 MHz [116] (27.0 dBm) (radar detection)
                        * 5600 MHz [120] (27.0 dBm) (radar detection)
                        * 5620 MHz [124] (27.0 dBm) (radar detection)
                        * 5640 MHz [128] (27.0 dBm) (radar detection)
                        * 5660 MHz [132] (27.0 dBm) (radar detection)
                        * 5680 MHz [136] (27.0 dBm) (radar detection)
                        * 5700 MHz [140] (27.0 dBm) (radar detection)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (13.0 dBm)
                        * 5765 MHz [153] (13.0 dBm)
                        * 5785 MHz [157] (13.0 dBm)
                        * 5805 MHz [161] (13.0 dBm)
        valid interface combinations:
                 * #{ AP } <= 16, #{ mesh point } <= 1, #{ managed } <= 1,
                   total <= 16, #channels <= 1, radar detect widths: { 20 MHz (n                                  o HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ RRM ]: RRM
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
root@swrt:~#

borkra, 2020-01-31:
Do not see the problem here, max allowed power (per router certification) is matching current power

olimpiumarius, 2020-09-26:
It seems that this problem no longer persists in the latest drive versions

closed 2020-09-26 by olimpiumarius