kaloz/mwlwifi #373: Linksys WRT32X (radio2)almost always ends up in the "Wireless is disabled or not associated" state

Einic, 2019-10-17:
I tried installing OpenWrt Reboot v18.06.4 on my WRT3200ACM. This alone has made the router go from shutting off the wireless interfaces after a few hours to bringing them back up on its own after a bit (but not before causing all the devices on wifi to choke for a few minutes.) I can share/pm logs.
`Thu Oct 17 10:25:32 2019 daemon.notice netifd: radio2 (25183): Could not find PHY for device 'radio2'
Thu Oct 17 10:25:32 2019 daemon.notice netifd: radio2 (25201): WARNING: Variable 'data' does not exist or is not an array/object`

eduperez, 2019-10-30:
Disable radio2 and forget about it, see the "Note for DFS of WRT3200ACM" on the README.

davidc502, 2019-10-30:
Agreed..  disable radio2.  Not recommended.

Einic, 2019-10-31:
> Disable radio2 and forget about it, see the "Note for DFS of WRT3200ACM" on the README.

Radio2 (5g) and radio0 (5g) are very unstable, often unable to connect, unable to connect to the Internet

howl, 2020-01-30:
https://github.com/kaloz/mwlwifi/issues/280#issuecomment-580008509

zmetzing, 2022-01-12:
Radio2 works fine for me. I'm using it as DMZ "guest" wireless on 5.2 GHz.

OpenWrt 19.07.8 r11364-ef56c85848 / LuCI openwrt-19.07 branch git-21.189.23240-7b931da

toughdeep, 2025-09-16:
Hi @zmetzing, It has been long time you seem active here- I also happen to have this router - Linksys WRT3200ACM ARMv7 Processor rev 1 (v7l)- not able to use radio2, but you seem to have managed to use it.
me not being knowledgeable like you in software and insights like you, would you help me understand- If I use this build you pointed out, Radio2 will work or are there any other settings to consider ?

> Radio2 works fine for me. I'm using it as DMZ "guest" wireless on 5.2 GHz.
> 
> Linksys WRT3200ACM ARMv7 Processor rev 1 (v7l)
> 
> OpenWrt 19.07.8 r11364-ef56c85848 / LuCI openwrt-19.07 branch git-21.189.23240-7b931da
> 
> > config wifi-device 'radio2'
> > option type 'mac80211'
> > option channel '36'
> > option hwmode '11a'
> > option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
> > option htmode 'VHT20'
> > config wifi-iface 'default_radio2'
> > option device 'radio2'
> > option mode 'ap'
> > option network 'dmz'
> > option key '12345'
> > option ssid 'Luggage'
> > option encryption 'psk2'

Hi @zmetzing , It has been long time you since you have been active here- but any way I will ask, any others knowledgeable please help me understand. 

I also happen to have this router - Linksys WRT3200ACM ARMv7 Processor rev 1 (v7l)- not able to use radio2, but you seem to have managed to use it.
me not being knowledgeable like you in software and insights like you, would you help me understand- If I use this build you pointed out, Radio2 will work or are there any other settings I should consider ?