kaloz/mwlwifi #374: Exception happened

tiagogaspar8, 2019-11-06:
```
[28761.391342] ieee80211 phy0: radar detected by firmware
[28772.677320] ieee80211 phy0: radar detected by firmware
[28782.627198] ieee80211 phy0: radar detected by firmware
[28792.573202] ieee80211 phy0: radar detected by firmware
[28820.516810] ieee80211 phy0: radar detected by firmware
[28832.530519] ieee80211 phy0: radar detected by firmware
[28842.511155] ieee80211 phy0: radar detected by firmware
[28900.572897] ieee80211 phy0: radar detected by firmware
[28906.564195] ieee80211 phy0: radar detected by firmware
[28948.834058] ------------[ cut here ]------------
[28948.838784] WARNING: CPU: 1 PID: 0 at backports-5.4-rc2-1/net/mac80211/rx.c:4581 ieee80211_rx_napi+0x168/0xa80 [mac80211]
[28948.849798] Rate marked as a VHT rate but data is invalid: MCS: 12, NSS: 2
[28948.856705] Modules linked in: nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir_ipv4 nft_redir nft_quota nft_numgen nft_nat nft_masq_ipv4 nft_masq nft_log nft_limit nft_flow_offload nft_ct nft_counter nft_chain_route_ipv6 nft_chain_route_ipv4 nft_chain_nat_ipv4 nf_tables nf_conntrack_netlink iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY rfcomm nfnetlink nf_reject_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conncount
[28948.928586]  iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hidp hci_uart btusb btmrvl_sdio btmrvl btintel bnep bluetooth sch_cake nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred hid evdev input_core mwlwifi mac80211 cfg80211 compat cryptodev ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ecdh_generic md5 kpp ecb cmac authenc gpio_button_hotplug
[28948.983460] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.19.81 #0
[28948.989491] Hardware name: Marvell Armada 380/385 (Device Tree)
[28948.995445] [<c010ebdc>] (unwind_backtrace) from [<c010ab34>] (show_stack+0x10/0x14)
[28949.003229] [<c010ab34>] (show_stack) from [<c065aa7c>] (dump_stack+0x88/0x9c)
[28949.010487] [<c065aa7c>] (dump_stack) from [<c0125f50>] (__warn+0xd4/0xf0)
[28949.017393] [<c0125f50>] (__warn) from [<c0125bf8>] (warn_slowpath_fmt+0x38/0x48)
[28949.024958] [<c0125bf8>] (warn_slowpath_fmt) from [<bf146648>] (ieee80211_rx_napi+0x168/0xa80 [mac80211])
[28949.034628] [<bf146648>] (ieee80211_rx_napi [mac80211]) from [<bf1c1b40>] (pcie_rx_recv_ndp+0x824/0xd94 [mwlwifi])
[28949.045046] [<bf1c1b40>] (pcie_rx_recv_ndp [mwlwifi]) from [<c012a0e0>] (tasklet_action_common.constprop.3+0x7c/0xdc)
[28949.055705] [<c012a0e0>] (tasklet_action_common.constprop.3) from [<c01021c8>] (__do_softirq+0xe0/0x25c)
[28949.065229] [<c01021c8>] (__do_softirq) from [<c012a650>] (irq_exit+0x9c/0xac)
[28949.072485] [<c012a650>] (irq_exit) from [<c0162eb0>] (__handle_domain_irq+0x9c/0xac)
[28949.080355] [<c0162eb0>] (__handle_domain_irq) from [<c039dee0>] (gic_handle_irq+0x5c/0x90)
[28949.088744] [<c039dee0>] (gic_handle_irq) from [<c0101a0c>] (__irq_svc+0x6c/0x90)
[28949.096258] Exception stack(0xdf46bf68 to 0xdf46bfb0)
[28949.101332] bf60:                   00000000 06948b28 dfbe12b0 c01144e0 ffffe000 c0a03c6c
[28949.109546] bf80: 00000002 c0a03ca8 c09464e8 00000000 00000000 00000000 c0a0e080 df46bfb8
[28949.117759] bfa0: c0108358 c010835c 60000013 ffffffff
[28949.122835] [<c0101a0c>] (__irq_svc) from [<c010835c>] (arch_cpu_idle+0x34/0x38)
[28949.130268] [<c010835c>] (arch_cpu_idle) from [<c014b044>] (do_idle+0xe4/0x1c8)
[28949.137611] [<c014b044>] (do_idle) from [<c014b384>] (cpu_startup_entry+0x18/0x1c)
[28949.145214] [<c014b384>] (cpu_startup_entry) from [<001023ec>] (0x1023ec)
[28949.152042] ---[ end trace 6adef90d29266056 ]---
```

This happens from time to time in my router WRT3200ACM with the 88W8964.

davidc502, 2019-11-06:
Doesn't surprise me. Wifi doesn't recover after radar detected, and especially doesn't recover if on 160hz width.
The work around is to keep wifi off of a DFS channel.

tiagogaspar8, 2019-11-07:
Thanks for the workaround, but I need to keep it in a DFS channel so that I can get a power boost.
The wifi never stops working just to clarify.
On another note, I see that this driver is unmaintained, I have no idea how to code in C but can't someone fix these issues in a fork? Or does this need to be fixed in the firmware (.bin file)?

kubrickfr, 2019-11-10:
> can't someone fix these issues in a fork? Or does this need to be fixed in the firmware (.bin file)?
In theory yes, someone could, however in practice this is going to be very hard considering that the firmware is closed source and has no public documentation.

tiagogaspar8, 2019-11-11:
Bu couldn't we by reading the code understand it and make up our own documentation?
I'm up for it but I have no knowledge of C and kernel related driers, but I have some knowledge of wireless technologies and that could help. Also since yuhhaurlin tried to merge this driver to the kernel, there he explains a lot of code. One of the ways to try and revive this driver is to try to get it to merge upstream.
I just need help of someone who knows some off that stuff.

eduperez, 2019-11-11:
@tiagogaspar8 The drivers that are added to the kernel are open sourced, but the firmware that runs on the wireless chipset is closed source.

tiagogaspar8, 2019-11-11:
I see waht you mean, but are the issues from the firmware? or are they from the driver? Cause for example in the stock Linksys firmware the firmware used in the proprietary driver is the same, actually it is an older version, but the issues we (at least that I see) aren't in the Linksys firmware so that leaves me to assume the bug is in the driver and not the firmware, see what I mean?

jeolives, 2019-11-13:
This open source version of mwlwifi is missing the proprietary implementation wherein radio2 (mwifiex) is used as a calibrator/DFS radar detector.

tiagogaspar8, 2019-11-26:
I guess it will never work, did yuhhaurlin maybe stop because Marvell Wireless was bought by NXP?