kaloz/mwlwifi #378: WRT3200ACM New Firmware Available

Chadster766, 2020-02-01:
Linksys has released a new firmware for this WRT. Below is the release notes for it:

```
===========================================================================

Firmware version:	1.0.8.198828
Release date:		January 8, 2020

- Integrated WLAN driver v9.1.2.6 and FW v9.3.2.8
- Added support for the Marvell Armada 385 Rev. B0 CPU
- Resolved multiple security vulnerabilities
- General bug fixes

---------------------------------------------------------------------------
```
Would mwlwifi work well with this 88W8964 v9.3.2.8 firmware?

eduperez, 2020-02-03:
I have privately received a confirmation that the new firmware works on OpenWrt 19.07.

BrainSlayer, 2020-02-03:
it works. i also added it to dd-wrt already. now the question is if it solves the problem with the esp chipsets

Chadster766, 2020-02-04:
Awesome thanks for the info guys :-)

rmandrad, 2020-02-04:
noob question - how one unpacks the bin file from the img file ?

tapper82, 2020-02-04:
How do we get this in to OpenWrt?

BrainSlayer, 2020-02-04:
use binwalk. and then just copy it from /lib/modules/...
its stored at the kernel modules directory.

1Jeff1, 2020-02-04:
> How do we get this in to OpenWrt?

I updated my fork with  the new firmware, for OpenWrt here is the patch [https://github.com/1Jeff1/openwrt/commit/993f98fc332fe5fc3322e0765858c6eaa3588320.patch](url)

I testing the new firmware on the wrt32x WiFi on 5Ghz 160MHz is more stable now , has frequently disconnects before , also internet speed has improved, I getting gigabit speed on WiFi.
```-----------------------------------------------------
 OpenWrt SNAPSHOT, r12171-ae61d21ca3
 -----------------------------------------------------
root@WRT32X:~# iw wlan0 info
Interface wlan0
        ifindex 21
        wdev 0x2
        addr 00:u8:00:70:t1:11
        ssid _5GHz
        type AP
        wiphy 0
        channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
        txpower 23.00 dBm
root@WRT32X:~# iperf3 -c 192.168.1.190
Connecting to host 192.168.1.190, port 5201
[  5] local 192.168.1.1 port 44562 connected to 192.168.1.190 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   101 MBytes   844 Mbits/sec    0   1.53 MBytes       
[  5]   1.00-2.00   sec   122 MBytes  1.03 Gbits/sec    0   1.53 MBytes       
[  5]   2.00-3.00   sec   118 MBytes   985 Mbits/sec    0   1.53 MBytes       
[  5]   3.00-4.01   sec   121 MBytes  1.01 Gbits/sec    0   1.53 MBytes       
[  5]   4.01-5.00   sec   116 MBytes   983 Mbits/sec    0   1.53 MBytes       
[  5]   5.00-6.01   sec   111 MBytes   927 Mbits/sec    0   1.53 MBytes       
[  5]   6.01-7.00   sec   119 MBytes   998 Mbits/sec    0   1.53 MBytes       
[  5]   7.00-8.00   sec   119 MBytes  1.00 Gbits/sec    0   1.53 MBytes       
[  5]   8.00-9.00   sec   116 MBytes   975 Mbits/sec    0   1.53 MBytes       
[  5]   9.00-10.00  sec   106 MBytes   892 Mbits/sec    0   1.53 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.12 GBytes   964 Mbits/sec    0             sender
[  5]   0.00-10.05  sec  1.12 GBytes   958 Mbits/sec                  receiver

iperf Done.```

Chadster766, 2020-02-05:
@BrainSlayer couldn't you make a merge request in this repo and ask @kaloz to approve it?

lantis1008, 2020-02-05:
Someone should test PMF and WPA3 with this new bin. 
You never know your luck in the big city

BrainSlayer, 2020-02-05:
@Chadster766 i could, but kaloz is known to be a strange guy and he doesnt like me at all. i dont think he will care about anything i do. but i'm sure openwrt will integrate this newer firmware soon

kaloz, 2020-02-05:
> @Chadster766 i could, but kaloz is known to be a strange guy and he doesnt like me at all. i dont think he will care about anything i do. but i'm sure openwrt will integrate this newer firmware soon

@BrainSlayer : <offtopic>I have no problems with you, I have problems about your business practises and shortcuting the GPL and other licenses..</offtopic> Feel free to submit a PR.

solidus1983, 2020-02-05:
@1Jeff1 your patch is it a patch -p 1 -i nameofpatch.patch or is it placing the .patch file into the package/kernel/mwlwifi/patches folder ?

1Jeff1, 2020-02-05:
> @1Jeff1 your patch is it a patch -p 1 -i nameofpatch.patch or is it placing the .patch file into the package/kernel/mwlwifi/patches folder ?

@solidus1983 is patch -p 1 -i 993f98fc332fe5fc3322e0765858c6eaa3588320.patch

> Someone should test PMF and WPA3 with this new bin.
> You never know your luck in the big city

@lantis1008 No luck with WAP3, I was expecting the new firmware fix it.

ninjanoir356, 2020-02-06:
> > How do we get this in to OpenWrt?
> 
> I updated my fork with the new firmware, for OpenWrt here is the patch https://github.com/1Jeff1/openwrt/commit/993f98fc332fe5fc3322e0765858c6eaa3588320.patch
> 
> I testing the new firmware on the wrt32x WiFi on 5Ghz 160MHz is more stable now , has frequently disconnects before , also internet speed has improved, I getting gigabit speed on WiFi.
> 
> ```
>  OpenWrt SNAPSHOT, r12171-ae61d21ca3
>  -----------------------------------------------------
> root@WRT32X:~# iw wlan0 info
> Interface wlan0
>         ifindex 21
>         wdev 0x2
>         addr 00:u8:00:70:t1:11
>         ssid _5GHz
>         type AP
>         wiphy 0
>         channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
>         txpower 23.00 dBm
> root@WRT32X:~# iperf3 -c 192.168.1.190
> Connecting to host 192.168.1.190, port 5201
> [  5] local 192.168.1.1 port 44562 connected to 192.168.1.190 port 5201
> [ ID] Interval           Transfer     Bitrate         Retr  Cwnd
> [  5]   0.00-1.00   sec   101 MBytes   844 Mbits/sec    0   1.53 MBytes       
> [  5]   1.00-2.00   sec   122 MBytes  1.03 Gbits/sec    0   1.53 MBytes       
> [  5]   2.00-3.00   sec   118 MBytes   985 Mbits/sec    0   1.53 MBytes       
> [  5]   3.00-4.01   sec   121 MBytes  1.01 Gbits/sec    0   1.53 MBytes       
> [  5]   4.01-5.00   sec   116 MBytes   983 Mbits/sec    0   1.53 MBytes       
> [  5]   5.00-6.01   sec   111 MBytes   927 Mbits/sec    0   1.53 MBytes       
> [  5]   6.01-7.00   sec   119 MBytes   998 Mbits/sec    0   1.53 MBytes       
> [  5]   7.00-8.00   sec   119 MBytes  1.00 Gbits/sec    0   1.53 MBytes       
> [  5]   8.00-9.00   sec   116 MBytes   975 Mbits/sec    0   1.53 MBytes       
> [  5]   9.00-10.00  sec   106 MBytes   892 Mbits/sec    0   1.53 MBytes       
> - - - - - - - - - - - - - - - - - - - - - - - - -
> [ ID] Interval           Transfer     Bitrate         Retr
> [  5]   0.00-10.00  sec  1.12 GBytes   964 Mbits/sec    0             sender
> [  5]   0.00-10.05  sec  1.12 GBytes   958 Mbits/sec                  receiver
> 
> iperf Done.```
> ```

Hi, 

I've installed openwrt last build, so dont know why we need to install the new update from linksys , cause we have installed an open firmware instead? And if we need, is it possible to give me exactly how to do it. I'm a novice.
Thank you very much for your help and work.

solidus1983, 2020-02-06:
@ninjanoir356 simply git clone OpenWRT like normal for building then ever the patch to the root of the git cloned OpenWRT directory then run patch -p 1 <nameofpatch.patch> then build

anomeome, 2020-02-06:
@ninjanoir356 , one way to kick it around while keeping things, ssh into your device and:

```
wifi down
cd /lib/firmware/mwlwifi
mv 88W8964.bin 88W8964.bin.ORG
wget https://github.com/1Jeff1/mwlwifi/raw/master/bin/firmware/88W8964.bin -O 88W8964.bin.OEM
ln -s 88W8964.bin.OEM 88W8964.bin
wifi up
```

and if you want to go back to the original FW, change the link back.

ninjanoir356, 2020-02-06:
Hi.
Thanks.
To be sure, with putty, I log in to my router,  and put it the script you
provide me , and sorry for the question, but it includes " wifi down and
wifi up" ?

🤔... I'm really a novice and english is not my la language....
Thanks , really.

anomeome, 2020-02-06:
That is correct, although you might want to just copy and paste lines one at a time in your _putty_ ssh window(third mouse button to paste ???). It is probably best to bounce the wifi to try and ensure modules are not in use. When you restart the radios with _wifi up_ you may want to run _logread -f_ in another ssh window to see that things load up correctly.

ninjanoir356, 2020-02-06:
This command is only for openwrt right?

p3x-robot, 2020-02-06:
it is faster!!!

solidus1983, 2020-02-06:
@1Jeff1

iw wlan0 info
Interface wlan0
        ifindex 41
        wdev 0x8
        addr 60:38:e0:ce:37:5a
        ssid TNCAP41C27_5G
        type AP
        wiphy 0
        channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
        txpower 23.00 dBm

iperf3 -c 192.168.1.103
Connecting to host 192.168.1.103, port 5201
[  5] local 192.168.1.3 port 59010 connected to 192.168.1.103 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  25.1 MBytes   210 Mbits/sec    0    214 KBytes
[  5]   1.00-2.00   sec  30.5 MBytes   256 Mbits/sec    0    214 KBytes
[  5]   2.00-3.00   sec  30.7 MBytes   256 Mbits/sec    0    214 KBytes
[  5]   3.00-4.00   sec  29.6 MBytes   249 Mbits/sec    0    214 KBytes
[  5]   4.00-5.00   sec  31.0 MBytes   260 Mbits/sec    0    214 KBytes
[  5]   5.00-6.00   sec  30.4 MBytes   255 Mbits/sec    0    214 KBytes
[  5]   6.00-7.00   sec  30.4 MBytes   255 Mbits/sec    0    214 KBytes
[  5]   7.00-8.00   sec  29.4 MBytes   246 Mbits/sec    0    214 KBytes
[  5]   8.00-9.00   sec  29.4 MBytes   247 Mbits/sec    0    214 KBytes
[  5]   9.00-10.00  sec  30.4 MBytes   255 Mbits/sec    0    214 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   297 MBytes   249 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   296 MBytes   249 Mbits/sec                  receiver

iperf Done.

Not sure how i am getting different results.

p3x-robot, 2020-02-06:
i think it is your wifi card is wrong. i have the same issue, with intel wifi card and i work good on 80mhz, but on 160mhz i get slow connections:
https://github.com/kaloz/mwlwifi/issues/280#issuecomment-581009794

solidus1983, 2020-02-06:
> i think it is your wifi card is wrong. i have the same issue, with intel wifi card and i work good on 80mhz, but on 160mhz i get slow connections:
> [#280 (comment)](https://github.com/kaloz/mwlwifi/issues/280#issuecomment-581009794)

Killer 1650 is my wireless card.

jeolives, 2020-02-06:
@kaloz - I've submitted a PR.

SHA-256 of the new FW: ca23f5bb730fde399359a716481651c120e09c299d4e6bbec6fe718c6e87e751

Confirmed by @hnyman:
https://forum.openwrt.org/t/new-official-firmware-wrt3200-acm/54313/8

kaloz, 2020-02-06:
@jeolives : merged, thanks :)

closed 2020-02-06 by kaloz

1Jeff1, 2020-02-06:
> Not sure how i am getting different results.

@solidus1983 I was using a Intel AX200 card on the test, when conected to 160mhz on the GUI it look like this:
[image]

solidus1983, 2020-02-06:
Strange, I was on 108 channel 160hz was fine for a while but it dropped back to 36 channel. Will see how it goes it's.more stable either way.

1Jeff1, 2020-02-06:
> Strange, I was on 108 channel 160hz was fine for a while but it dropped back to 36 channel. Will see how it goes it's.more stable either way.

@solidus1983 That is normal on DFS channels, I had set to channel 44 and changes between 100 and 36

lucenera, 2020-02-06:
I write here, because you will have done it before me. How to compile the drivers with the new firmware? I compiled the whole image for the WRT3200ACM, replacing the commit string in the Makefile in package /kernel/mwlwifi/ and renaming the patches folder to patces.tmp.
If I flash the image, however, everything works except the mwlwifi card and LuCI installed, I find only the mwifiex card in the Wireless card.
What procedure should I follow?

anomeome, 2020-02-06:
Grab the [PR2756](https://patch-diff.githubusercontent.com/raw/openwrt/openwrt/pull/2756.patch) patch and apply to your tree.

lucenera, 2020-02-06:
Eg: git apply https://patch-diff.githubusercontent.com/raw/openwrt/openwrt/pull/2756.patch in /openwrt/? Can it work?

anomeome, 2020-02-06:
You can use _git_ if you want the semantics around your version number indicating _+x_ for number of patches above and beyond your pull.

eduperez, 2020-02-07:
If you can wait untill the weekend, I will try to build updated packages for latest OpenWrt release.

lucenera, 2020-02-07:
Ok, thanks @eduperez but I use snapshot.  In the meantime I have tried both manually changing the Makefile she by applying the patch as suggestion, but the result is the same: the image is built, starts and mwlwifi is also in the list of loaded modules (lsmod), but it is as if the card  wifi was not there and even in LuCi it does not appear.

eduperez, 2020-02-07:
@lucenera Check that the correct firmware file gets copied to the correct location. Look for something relevant in the logs.

lucenera, 2020-02-07:
@eduperez I don't find any relevant data. Everything looks correct. If I compile the master image without patches, everything works and WiFi works perfectly. If I apply the patch by executing the commands suggested by @anomeome then the image starts, the mwlwifi module and the corresponding firmware are loaded, but the wifi interface does not appear or rather only the one relating to mwifiex appears. In the end the command of the patch seems to me quite simple. So I don't know what it can depend on. Can you check my steps?
* git clone https://github.com/openwrt/openwrt.git
* cd openwrt
*. /scripts/feeds update -a
*. /scripts/feeds install -a
* make menuconfig
* wget https://github.com/openwrt/openwrt/pull/2756.patch
* patch -p1 <2756.patch
* make -j 10 downloads
* make -j 10

anomeome, 2020-02-07:
Does generating a new default config change behaviour:
```
wifi down
mv /etc/config/wireless /tmp
wifi config
wifi up
```

lucenera, 2020-02-07:
Now I can't try, but first I replaced the file /etc/config/network with the one on the partition with working OpenWrt. Nothing changes. Can anyone tell me how he did it? My procedure seems correct to you. Once the patch is applied, I receive confirmation and no errors.

jeolives, 2020-02-07:
@lucenera & @anomeome 

I've updated the pull request since there is @borkra's commit doesn't properly detect the Linux kernel version found in OpenWrt's backports package for mac80211. 

Right now the code is only pulling the version ("4.14.y") or (4.19.y"), whilst it should be pulling in ("5.4-rc8") from backports.

Do a fresh clone of openwrt and apply the updated PR (including the last bit of whitespace at the bottom).

```
From f05311a4d07bf3490bc443b6f89fa9783b026abb Mon Sep 17 00:00:00 2001
From: Jose Olivera <oliverajeo@gmail.com>
Date: Fri, 7 Feb 2020 00:06:31 +0800
Subject: [PATCH] mwlwifi: Update the 88W8964's firmware to 9.3.2.12

Updates the 88W8964 firmware used in the Linksys WRT3200ACM and WRT32X
[v9.3.2.6 -> v9.3.2.12]

Removes 0c43219 ("mwlwifi: Fix loading with backports v5.3")
as it has been merged upstream.

Unfortunately, there is a bug where in Kaloz's repo, the version
detection mechanism for fixing vendor commands doesn't work.

It pulls in the Linux kernel version, which as of this time is
"4.14.y" or "4.19.y"

However, the proper behaviour is that it should  pull in the mac80211
backports version which as of now is "5.4-rc8"

The included patch fixes this.

---
 package/kernel/mwlwifi/Makefile                      |  8 ++++----
 ...1-Fix-compile-with-mac80211-backports-5_3+.patch} | 12 +++++++-----
 2 files changed, 11 insertions(+), 9 deletions(-)
 rename package/kernel/mwlwifi/patches/{001-vendor_command_policy.patch => 001-Fix-compile-with-mac80211-backports-5_3+.patch} (57%)

diff --git a/package/kernel/mwlwifi/Makefile b/package/kernel/mwlwifi/Makefile
index aa8eea0450a..37232fdb3d4 100644
--- a/package/kernel/mwlwifi/Makefile
+++ b/package/kernel/mwlwifi/Makefile
@@ -8,16 +8,16 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=mwlwifi
-PKG_RELEASE=2
+PKG_RELEASE=1
 
 PKG_LICENSE:=ISC
 PKG_LICENSE_FILES:=
 
 PKG_SOURCE_URL:=https://github.com/kaloz/mwlwifi
 PKG_SOURCE_PROTO:=git
-PKG_SOURCE_DATE:=2019-03-02
-PKG_SOURCE_VERSION:=31d9386079b91cc699658c19294e139b62b512bc
-PKG_MIRROR_HASH:=7bdd05765d8215a9c293cdcb028d63a04c9e55b337eaac9e8d3659bd86218321
+PKG_SOURCE_DATE:=2020-02-06
+PKG_SOURCE_VERSION:=a2fd00bb74c35820dfe233d762690c0433a87ef5
+PKG_MIRROR_HASH:=0eda0e774a87e58e611d6436350e1cf2be3de50fddde334909a07a15b0c9862b
 
 PKG_MAINTAINER:=Imre Kaloz <kaloz@openwrt.org>
 PKG_BUILD_PARALLEL:=1
diff --git a/package/kernel/mwlwifi/patches/001-vendor_command_policy.patch b/package/kernel/mwlwifi/patches/001-Fix-compile-with-mac80211-backports-5_3+.patch
similarity index 57%
rename from package/kernel/mwlwifi/patches/001-vendor_command_policy.patch
rename to package/kernel/mwlwifi/patches/001-Fix-compile-with-mac80211-backports-5_3+.patch
index 1f06d55fc8a..00de0ddee59 100644
--- a/package/kernel/mwlwifi/patches/001-vendor_command_policy.patch
+++ b/package/kernel/mwlwifi/patches/001-Fix-compile-with-mac80211-backports-5_3+.patch
@@ -1,19 +1,21 @@
-mac80211 from kernel 5.3 and later checks the new policy attribute.
-
 --- a/vendor_cmd.c
 +++ b/vendor_cmd.c
-@@ -92,12 +92,14 @@ static const struct wiphy_vendor_command
+@@ -92,18 +92,14 @@ static const struct wiphy_vendor_command
  			  .subcmd = MWL_VENDOR_CMD_SET_BF_TYPE},
  		.flags = WIPHY_VENDOR_CMD_NEED_NETDEV,
  		.doit = mwl_vendor_cmd_set_bf_type,
-+		.policy = mwl_vendor_attr_policy,
+-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,3,0))
+ 		.policy = mwl_vendor_attr_policy,
+-#endif
  	},
  	{
  		.info = { .vendor_id = MRVL_OUI,
  			  .subcmd = MWL_VENDOR_CMD_GET_BF_TYPE},
  		.flags = WIPHY_VENDOR_CMD_NEED_NETDEV,
  		.doit = mwl_vendor_cmd_get_bf_type,
-+		.policy = mwl_vendor_attr_policy,
+-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5,3,0))
+ 		.policy = mwl_vendor_attr_policy,
+-#endif
  	}
  };
  
```

anomeome, 2020-02-07:
There is an issue [see chatter](https://github.com/openwrt/openwrt/pull/2756)

lucenera, 2020-02-07:
@jeolives Now I try, but sorry if I'm really noob. What does that backport refer to? I use master which is on 4.19.101. Should it work already? You refer to the 5.x kernel, but I don't know about OpenWrt for now.

borkra, 2020-02-07:
OpenWRT compiles mwlwifi driver out of backports tree, so no automatic detection possible.
 mwlwifi driver is written for compilation without backports so all Linux version macros correct
Patch inside OpenWRT need to change to accommodate new code.

borkra, 2020-02-07:
backports is the project that ports wi-fi stack from the newer kernel into older ones.

jeolives, 2020-02-07:
> @jeolives Now I try, but sorry if I'm really noob. What does that backport refer to? I use master which is on 4.19.101. Should it work already? You refer to the 5.x kernel, but I don't know about OpenWrt for now.

OpenWrt bases their mac80211 (wifi subsystem which `mwlwifi` interacts with) closer to the latest Linux kernel, see `package/kernel/mac80211/Makefile`:

```
PKG_NAME:=mac80211

PKG_VERSION:=5.4-rc8-1 <---- mac80211/wifi interface is actually this Linux version
PKG_RELEASE:=1
PKG_SOURCE_URL:=@KERNEL/linux/kernel/projects/backports/stable/v5.4-rc8/ <---- See this
PKG_HASH:=707ffc50d0f6fe3398bcc3a7abd44ffa088d9309255a3a71d014620c20d3bbf9
```

So whilst your device boots with a kernel with `v4.19.y` or `v4.14.y`, mwlwifi actually interacts with a driver interface that is `v5.4.y`. 

What I pasted will work on a fresh `git clone` of the OpenWrt sources.

jeolives, 2020-02-07:
> backports is the project that ports wi-fi stack from the newer kernel into older ones.

It's not just the WiFi stack anymore (it also covers display drivers and others now), but that's out of scope for OpenWrt...

lucenera, 2020-02-07:
Thank you all!

borkra, 2020-02-07:
Unfortunately you are behind times, everything except Wi-Fi was pulled out of backports as of last year, even Bluetooth.

The only thing backport ports now days is Wi-Fi.

jeolives, 2020-02-07:
> Unfortunately you are behind times, everything except Wi-Fi was pulled out of backports as of last year, even Bluetooth.
> 
> The only thing backport ports now days is Wi-Fi.

Sort of makes sense given that Hauke is now managing the repo, overlooked that detail.

kaloz, 2020-02-07:
@jeolives : instead of submitting a patch to parch the driver in openwrt, submit that change here ;)

jeolives, 2020-02-07:
@kaloz 

I'm not sure, this could break operations for `mwlwifi` users with older kernel versions who don't use a backports package. But I haven't tried testing this out myself :/

kaloz, 2020-02-07:
@jeolives : given neither Marvell or Linksys is interested in the driver anymore, I doubt there would be a userbase outside openwrt..

lucenera, 2020-02-07:
@kaloz while I try, I also want to take away a curiosity. In the project guide you say to modify the Makefile to compile the new drivers. There is an entry in the Makefile of the OpenWrt versions I use, which I don't know where it can be retrieved: PKG_MIRROR_HASH
What is that? Where do I find the correct value for the last commit?

kaloz, 2020-02-07:
@lucenera : as long as you download from git openwrt ignores that value, it's useful if you need to grab the archive from a mirror

p3x-robot, 2020-02-07:
@kaloz isnt the mwlwifi sold to npx.com? not Marvell or Linksys right?

borkra, 2020-02-07:
@kaloz The patch must be in OpenWRT.

OpenWRT when compiles mwlwifi does not know the version of backported kernel, only version of the main kernel (because mwlwifi compiles out backports tree) As such it is impossible to setup ##ifdefs for backported kernel version.

And backported kernel version != main kernel version.

borkra, 2020-02-07:
> @kaloz isnt the mwlwifi sold to npx.com? not Marvell or Linksys right?

IKind of, it is NXP driver and firmware for Linksys routers.

p3x-robot, 2020-02-07:
@kaloz says none is interested in this driver, but at least the firmware is upgraded and it is more stable as ever! linksys wrt rulez!

CHKDSK88, 2020-02-07:
Does anybody test how radius work on new firmware? Is still broken?

Chadster766, 2020-02-07:
I've used radius on the wrt1900ac v1 and wrt1900acs successfully. I didn't know of a radius issue with the wrt3200/32x.

CHKDSK88, 2020-02-07:
Rekeying problem affect radius in wrt32x.

[40 more replies]

PR #379 (2020-02-06): Upgrade the 88W8964's firmware to 9.3.2.12
Firmware version: 0x903020c (from 0x9030206)

Note: 0c is hex for decimal 12

Devices affected: WRT3200ACM; WRT32X

Closes: #378