kaloz/mwlwifi #382: last 2 commits - Operating Frequncy and Width missing - No SSID Broadcast

davidc502, 2020-02-14:
One of the last 2 commits is breaking the Wifi Operating frequency when compiling from trunk.  Removing these 2 patches fixes the issue. No SSID is broadcast either.

I removed both of the last commits, and wifi compiled normally, and works properly. Of the 2 patches below breaks things, I don't know which one is the actual culprit. 

1. Merge pull request #379 from jeolives/master
2. Merge pull request #369 from borkra/vendor_cmd_kernel_5.3

Chadster766, 2020-02-14:
I don't see these issues with those commits:
```
root@MCDEBIAN:~# iw dev
phy#2
        Interface wwlan
                ifindex 15
                wdev 0x200000002
                addr 06:a8:5b:be:4d:f1
                ssid Wemo-24
                type AP
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
                txpower 30.00 dBm
        Interface wlp2s0
                ifindex 11
                wdev 0x200000001
                addr 06:a8:5b:be:4d:f0
                ssid McDebian-24
                type AP
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
                txpower 30.00 dBm
phy#0
        Interface wlp1s0
                ifindex 10
                wdev 0x1
                addr 60:38:e0:bd:31:92
                ssid McDebian_AC
                type AP
                channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 30.00 dBm
phy#1
        Interface mlan0
                ifindex 9
                wdev 0x100000001
                addr 60:38:e0:bd:31:93
                type managed
                txpower 0.00 dBm
```

anomeome, 2020-02-15:
Using OpenWrt Github PR2756 things are fine
```
OpenWrt SNAPSHOT, r12220-dba431d8ab

iw dev
phy#1
	Interface wlan1-1
		ifindex 17
		wdev 0x100000003
		addr 62:38:e0:b8:9e:b0
		ssid KabuliGuest
		type AP
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 30.00 dBm
	Interface wlan1
		ifindex 16
		wdev 0x100000002
		addr 60:38:e0:b8:9e:b1
		ssid Kabuli
		type AP
		channel 1 (2412 MHz), width: 40 MHz, center1: 2422 MHz
		txpower 30.00 dBm
phy#0
	Interface wlan0-1
		ifindex 18
		wdev 0x3
		addr 62:38:e0:b8:9e:b3
		ssid ChanaGuest
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 23.00 dBm
	Interface wlan0
		ifindex 15
		wdev 0x2
		addr 60:38:e0:b8:9e:b2
		ssid Chana
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 23.00 dBm

```
there were issues around the kernel policy, and there is new PR opened here offering an alternative method.

davidc502, 2020-02-15:
Very interesting..  Well, I'll go ahead and close it.

closed 2020-02-15 by davidc502