kaloz/mwlwifi #383: Any chance you could make this Debian compatible?

professorIsaac, 2020-02-14:


anomeome, 2020-02-14:
https://github.com/kaloz/mwlwifi/issues/365#issuecomment-505510174

professorIsaac, 2020-02-16:
I am asking you to make compatible or help me

professorIsaac, 2020-02-16:
The make file is not working

ValCher1961, 2020-02-16:
Take a look here - https://github.com/cilix-lab/ubuntu-wrt

professorIsaac, 2020-02-16:
Thank you for trying but what I want you is a downloadable driver that I can run make command

ValCher1961, 2020-02-16:
I make the driver mwlwifi in the kernel linux. I take the current driver and embed it in the core.
Example.
1. Download the driver.
```
git clone https://github.com/kaloz/mwlwifi.git
mkdir drivers/net/wireless/marvell
cp -r mwlwifi ./drivers/net/wireless/marvell/mwlwifi
```
2. Download the desired version of the kernel.
```
wget https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.4.20.tar.xz && tar xvf linux-5.4.20.tar.xz
```
3. Copy the driver to the kernel.
cp -r drivers linux-5.4.20

4. I build the driver into the core with a patch - https://github.com/ValCher1961/McDebian_WRT3200ACM/blob/master/kernel-4.19.X/mvdrv.patch
```
cp -rf mvdrv.patch linux-5.4.20
cd linux-5.4.20 && patch -p1 < mvdrv.patch
```
5. Next configuration of the for kernel and inclusion driver (for module)
```
make menuconfig
make ARCH=arm CROSS_COMPILE=arm-none-eabi- modules
```

professorIsaac, 2020-02-16:
I see but is this assuming that I have a wrt device

bmork, 2020-03-11:
I just tried building the current master on Debian buster, and cannot see any issue.
```
bjorn@miraculix:/usr/local/src/git/mwlwifi$ make KDIR=/lib/modules/`uname -r`/build
make -C /lib/modules/4.19.0-8-amd64/build M=/usr/local/src/git/mwlwifi
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-8-amd64'
  CC [M]  /usr/local/src/git/mwlwifi/core.o
  CC [M]  /usr/local/src/git/mwlwifi/mac80211.o
  CC [M]  /usr/local/src/git/mwlwifi/utils.o
[..]
  CC [M]  /usr/local/src/git/mwlwifi/mu_mimo.o
  CC [M]  /usr/local/src/git/mwlwifi/vendor_cmd.o
  LD [M]  /usr/local/src/git/mwlwifi/mwlwifi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/local/src/git/mwlwifi/mwlwifi.mod.o
  LD [M]  /usr/local/src/git/mwlwifi/mwlwifi.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-8-amd64'
```
I believe this has been working for ages.  I've certainly test-built the driver on Debian many times in the past without doing anything extra-ordinary.

You'll probably get better feedback if you show us why you believe that "The make file is not working"....

professorIsaac, 2020-03-11:
Can you put it in to monitor mode?

bmork, 2020-03-11:
I'm finished here unless you decide to share what you are trying to do.  As it is now, it looks mostly like trolling.  Which I guess must be fun.

professorIsaac, 2020-03-11:
I just want to use my surface pro to hack my WiFi network

closed 2020-04-16 by kaloz

midozalouk, 2021-12-15:
Hello Kaloz
thanks for the great work
can you please help us to provide driver for chipset 88W8897 that support monitor mode to be used in linux 
Thanks

midozalouk, 2021-12-16:
I have the following error when i tried your command
No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/include/generated/asm/syscalls_32.h'. Stop.

how can i proceed from there ?

bmork, 2021-12-16:
> No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/include/generated/asm/syscalls_32.h'. 

And you successfully can build other kernel modules on your system?  If not, then I don't think the problem is related to this project/driver at all.

tapper82, 2021-12-16:
The driver is ded anyway!

midozalouk, 2021-12-17:
> > No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/include/generated/asm/syscalls_32.h'.
> 
> And you successfully can build other kernel modules on your system? If not, then I don't think the problem is related to this project/driver at all.

sorry for the silly question
i try to build it for using in my chromebook on surface device

what is the essential packages needed to by able to build that kernel ?

will it work if i used it in other system than openwrt

thanks sir

midozalouk, 2021-12-17:
I have successfully compiled it
But when I load it with insmod it gives me error unknown symbol in module
How can I fix it

midozalouk, 2021-12-17:
I have successfully build it and managed to load it with insmod
But I don't show as active wireless!!!
Is there anything wrong?
How can I activate the interface?

midozalouk, 2021-12-18:
Hello @bmork 
any help is appreciated sir ?

midozalouk, 2022-06-26:
I have managed to compile it after removing part of the code that checks the firmware signature in `hif/pcie/fwdl.c` file

`	if (curr_iteration == 0) {
		wiphy_err(hw->wiphy,
			  "Exhausted curr_iteration for fw signature\n");
		goto err_download;
	}
`

the problem now that it loads with no mac address 

and i can't  activate it with either ifconfig or ip set

```
localhost / # sudo ip link set dev wlan0 up
RTNETLINK answers: Input/output error 
```

```
localhost / # sudo ifconfig wlan0 up
SIOCSIFFLAGS: Input/output error
```

this is dmesg
```

[ 2020.455368] ieee80211 phy1: cmd 0x9169=LedCtrl timed out
[ 2020.455379] ieee80211 phy1: return code: 0x1169
[ 2020.455384] ieee80211 phy1: timeout: 0x1169
[ 2020.456911] ieee80211 phy1: 2G enabled, 5G enabled
[ 2020.456916] ieee80211 phy1: 2 TX antennas, 2 RX antennas
[ 2020.512965] udevd[15384]: Process '/usr/sbin/iw dev wlan0 set power_save on' failed with exit code 161.
[ 2020.710225] arc_wlan0: port 1(vmtap3) entered blocking state
[ 2020.710241] arc_wlan0: port 1(vmtap3) entered disabled state
[ 2020.710475] device vmtap3 entered promiscuous mode
[ 2020.710523] arc_wlan0: port 1(vmtap3) entered blocking state
[ 2020.710525] arc_wlan0: port 1(vmtap3) entered forwarding state
[ 2020.810230] IPv6: ADDRCONF(NETDEV_UP): veth3: link is not ready
[ 2020.812619] IPv6: ADDRCONF(NETDEV_CHANGE): veth3: link becomes ready
[ 2021.146385] IPv6: ADDRCONF(NETDEV_UP): lo: link is not ready
```