kaloz/mwlwifi #384: linksys WTR32X wireless not work

zcracker, 2020-03-12:
Thu Mar 12 01:07:19 2020 kern.err kernel: [  867.497650] ieee80211 phy1: cmd 0x9122=UpdateEncryption timed out
Thu Mar 12 01:07:19 2020 kern.err kernel: [  867.503795] ieee80211 phy1: return code: 0x1122
Thu Mar 12 01:07:19 2020 kern.err kernel: [  867.508353] ieee80211 phy1: timeout: 0x1122
Thu Mar 12 01:07:19 2020 kern.err kernel: [  867.512555] wlan1: failed to remove key (0, b8:5d:0a:b5:bf:c8) from hardware (-5)
Thu Mar 12 01:07:19 2020 kern.debug kernel: [  867.520177] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:07:39 2020 kern.err kernel: [  887.511069] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:07:39 2020 kern.err kernel: [  887.516943] ieee80211 phy0: return code: 0x001d
Thu Mar 12 01:07:39 2020 kern.err kernel: [  887.521492] ieee80211 phy0: timeout: 0x001d
Thu Mar 12 01:07:39 2020 kern.debug kernel: [  887.525700] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:07:59 2020 kern.err kernel: [  907.526192] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:07:59 2020 kern.err kernel: [  907.532055] ieee80211 phy0: return code: 0x001d
Thu Mar 12 01:07:59 2020 kern.err kernel: [  907.536614] ieee80211 phy0: timeout: 0x001d
Thu Mar 12 01:07:59 2020 kern.debug kernel: [  907.540816] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:07:59 2020 authpriv.info dropbear[15618]: Child connection from 10.0.0.121:60292
Thu Mar 12 01:08:00 2020 cron.info crond[4407]: USER root pid 15619 cmd /usr/share/clash/clash-watchdog.sh
Thu Mar 12 01:08:00 2020 cron.info crond[4407]: USER root pid 15620 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 01:08:01 2020 authpriv.notice dropbear[15618]: Password auth succeeded for 'root' from 10.0.0.121:60292
Thu Mar 12 01:08:07 2020 user.err ddns-scripts[10389]: myddns_ipv4: ip Error: 1
Thu Mar 12 01:08:19 2020 kern.err kernel: [  927.516900] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:08:19 2020 kern.err kernel: [  927.522768] ieee80211 phy0: return code: 0x001d
Thu Mar 12 01:08:19 2020 kern.err kernel: [  927.527318] ieee80211 phy0: timeout: 0x001d
Thu Mar 12 01:08:19 2020 daemon.info hostapd: wlan1: STA b8:5d:0a:b5:bf:c8 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Mar 12 01:08:19 2020 kern.debug kernel: [  927.537608] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:08:39 2020 kern.err kernel: [  947.512350] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
Thu Mar 12 01:08:39 2020 kern.err kernel: [  947.518212] ieee80211 phy1: return code: 0x1111
Thu Mar 12 01:08:39 2020 kern.err kernel: [  947.522770] ieee80211 phy1: timeout: 0x1111
Thu Mar 12 01:08:39 2020 daemon.info hostapd: wlan0: STA f4:60:e2:84:0f:d2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Mar 12 01:08:39 2020 kern.debug kernel: [  947.529337] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:08:59 2020 kern.err kernel: [  967.521226] ieee80211 phy0: cmd 0x9111=SetNewStation timed out
Thu Mar 12 01:08:59 2020 kern.err kernel: [  967.527105] ieee80211 phy0: return code: 0x1111
Thu Mar 12 01:08:59 2020 kern.err kernel: [  967.531668] ieee80211 phy0: timeout: 0x1111
Thu Mar 12 01:08:59 2020 kern.debug kernel: [  967.536131] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:09:00 2020 cron.info crond[4407]: USER root pid 15735 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 01:09:07 2020 user.err ddns-scripts[10389]: myddns_ipv4: ip Error: 1
Thu Mar 12 01:09:19 2020 kern.err kernel: [  987.550172] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:09:19 2020 kern.err kernel: [  987.556058] ieee80211 phy1: return code: 0x001d
Thu Mar 12 01:09:19 2020 kern.err kernel: [  987.560620] ieee80211 phy1: timeout: 0x001d
Thu Mar 12 01:09:19 2020 kern.debug kernel: [  987.564838] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:09:39 2020 kern.err kernel: [ 1007.594940] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:09:39 2020 kern.err kernel: [ 1007.600811] ieee80211 phy1: return code: 0x001d
Thu Mar 12 01:09:39 2020 kern.err kernel: [ 1007.605376] ieee80211 phy1: timeout: 0x001d
Thu Mar 12 01:09:39 2020 kern.debug kernel: [ 1007.609594] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 01:09:59 2020 kern.err kernel: [ 1027.572230] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 01:09:59 2020 kern.err kernel: [ 1027.578096] ieee80211 phy1: return code: 0x001d
Thu Mar 12 01:09:59 2020 kern.err kernel: [ 1027.582653] ieee80211 phy1: timeout: 0x001d
Thu Mar 12 01:09:59 2020 kern.debug kernel: [ 1027.587053] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000

[  687.479415] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  707.436575] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[  707.442465] ieee80211 phy1: return code: 0x1111
[  707.447032] ieee80211 phy1: timeout: 0x1111
[  707.485678] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  727.465330] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[  727.471198] ieee80211 phy1: return code: 0x001d
[  727.475751] ieee80211 phy1: timeout: 0x001d
[  727.479961] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  747.442018] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[  747.447889] ieee80211 phy1: return code: 0x001d
[  747.452438] ieee80211 phy1: timeout: 0x001d
[  747.456670] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  767.421303] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[  767.427171] ieee80211 phy1: return code: 0x001d
[  767.431727] ieee80211 phy1: timeout: 0x001d
[  767.437659] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  787.444255] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[  787.450129] ieee80211 phy1: return code: 0x001d
[  787.454679] ieee80211 phy1: timeout: 0x001d
[  787.458890] ieee80211 phy1: MACREG_REG_INT_CODE: 0x0000
[  807.463756] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[  807.469634] ieee80211 phy1: return code: 0x001d
[... 108 lines cut ...]
[  968.175070] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
[  987.820517] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out

Thu Mar 12 14:29:12 2020 daemon.info hostapd: wlan1: STA ac:d5:64:a2:d6:97 IEEE 802.11: authenticated
Thu Mar 12 14:30:00 2020 cron.info crond[4446]: USER root pid 6672 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 14:30:00 2020 cron.info crond[4446]: USER root pid 6673 cmd /usr/share/clash/clash-watchdog.sh
Thu Mar 12 14:31:00 2020 cron.info crond[4446]: USER root pid 6713 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 14:32:00 2020 cron.info crond[4446]: USER root pid 6927 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 14:32:00 2020 cron.info crond[4446]: USER root pid 6928 cmd /usr/share/clash/clash-watchdog.sh
Thu Mar 12 14:32:49 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED ac:1f:74:ab:35:38
Thu Mar 12 14:32:49 2020 daemon.info hostapd: wlan0: STA ac:1f:74:ab:35:38 IEEE 802.11: disassociated due to inactivity
Thu Mar 12 14:33:00 2020 cron.info crond[4446]: USER root pid 6943 cmd /usr/share/clash/kill_watchdog.sh
Thu Mar 12 14:33:09 2020 kern.err kernel: [  450.631832] ieee80211 phy0: cmd 0x9122=UpdateEncryption timed out
Thu Mar 12 14:33:09 2020 kern.err kernel: [  450.637979] ieee80211 phy0: return code: 0x1122
Thu Mar 12 14:33:09 2020 kern.err kernel: [  450.642541] ieee80211 phy0: timeout: 0x1122
Thu Mar 12 14:33:09 2020 kern.err kernel: [  450.646742] wlan0: failed to remove key (0, ac:1f:74:ab:35:38) from hardware (-5)
Thu Mar 12 14:33:09 2020 kern.debug kernel: [  450.654421] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 14:33:29 2020 kern.err kernel: [  470.661457] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 14:33:29 2020 kern.err kernel: [  470.667466] ieee80211 phy0: return code: 0x001d
Thu Mar 12 14:33:29 2020 kern.err kernel: [  470.672015] ieee80211 phy0: timeout: 0x001d
Thu Mar 12 14:33:29 2020 kern.debug kernel: [  470.676218] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000
Thu Mar 12 14:33:49 2020 kern.err kernel: [  490.674192] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
Thu Mar 12 14:33:49 2020 kern.err kernel: [  490.680082] ieee80211 phy0: return code: 0x001d
Thu Mar 12 14:33:49 2020 kern.err kernel: [  490.684650] ieee80211 phy0: timeout: 0x001d
Thu Mar 12 14:33:49 2020 kern.debug kernel: [  490.688854] ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000

kubrickfr, 2020-03-12:
Don't change the region, see https://github.com/kaloz/mwlwifi/issues/355

closed 2020-03-17 by zcracker

thagabe, 2020-03-19:
I'ma just put this here: [https://github.com/EXSERENS/mwlwifi/commit/97e051a562c57e61a0404f09d64132a3abbfcc91](url)

monokrome, 2021-04-01:
The region isn't the issue here. The issue happens even on the expected region.

gitthangbaby, 2021-04-20:
i also have a normal region set on all interfaces, and all of a sudden, for the first time in years, i get this
`ieee80211 phy0: MACREG_REG_INT_CODE: 0x0000`
no timeouts or other errors usually displayed with this. wifi up and not accepting clients.
i do frequently run command "wifi" to restart wifi (so i don't need to restart router), if it helps.  i must have restarted the wifi/router by my own watchdog script 10.000 times, it's so unreliable:o

andyinsf, 2023-04-30:
I fixed mine by shutting off WPA3 per https://openwrt.org/toh/linksys/wrt_ac_series#wpa380211w_may_hang_kernel