kaloz/mwlwifi #386: a2fd00b breaks 802.11r AP roaming

vpngine, 2020-04-11:
Hi,

Thanks for your efforts with this code.

I've been testing the a2fd00b drivers on wrt32x (mlwifi being compiled into 19.07.2) and it seems that with this latest version 802.11r roaming breaks. 

Test scenario:
router/AP1. wrt32x 19.07.2 compiled with new a2fd00b drivers
AP2. wrt1900acsv2 19.07.2 compiled with standard release drivers 
In the above config, logs show hosts re-authenticate to a new AP when roaming, but after authentication clients fail to be handed over to the new AP as any date transfer stops and the original AP remaining connected.

exact relevant config on both devices is as follows:
         option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option pmk_r1_push '1'
	option mobility_domain '75b9'
	option nasid '0049E0AE8DDA'

With the exact same gear and config, only with both running same 19.07.2 compiled with the release version mlwifi drivers, 802.11r works perfectly .

kaloz, 2020-04-16:
try to use the same chipset firmware on both devices - there has been no promises from Marvell to keep them compatible (or promises at all)

kubrickfr, 2020-04-27:
For me compatibility between devices and even brands has been good. I'm using the latest version of mwlwifi on a WRT3200ACM and I can roam betwen it and an Archer c7, no problem

thagabe, 2020-07-21:
@kaloz How can you even use the same chipset firmware on both devices if they are different architectures?