kaloz/mwlwifi #387: Inconsistent Wi-Fi - WRT3200ACM

larivierec, 2020-06-11:
Hello,

I have to spam reboot my router to get a stable state. I don't think this is intended.
Everytime, you see a message like this.
[image]

closed 2020-06-12 by larivierec

davidc502, 2020-07-12:
it's been my experience that 160Mhz has been unreliable and I have never recommended anyone use it. 
BTW -- No 'real' driver updates in over a year and a half, so take that as you will.

larivierec, 2020-07-13:
This is true. Believe it or not it wasn't the fault of the driver but how OpenWRT handles LEDs that caused that issue.

It's still shameful that they are no more driver updates though...

Thanks for your comment