kaloz/mwlwifi #391: Poor performance and poor stability wrt1900acs v2

olimpiumarius, 2020-11-22:
Below i will leave some information about the latest version of OpenWrt with the latest wi-fi drive.
Performance continues to be poor as well as stability on the wifi side!
[image]
[image]
[image]
[image]

BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 SuperWRT SNAPSHOT, r14984-d369993898
 -----------------------------------------------------
root@swrt:~# cat /etc/config/wireless verifica wifi

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci00000.0'
        option htmode 'VHT80'
        option disabled '0'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option country 'RO'
        option channel '48'
        option noscan '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option macaddr '5d1'
        option ssid 'superwrt 5Ghz'
        option key ''
        option encryption 'psk2+ccmp'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci00000.0'
        option disabled '0'
        option htmode 'HT40'
        list channels '1'
        list channels '6'
        list channels '11'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option country 'RO'
        option channel '9'
        option noscan '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option macaddr ''
        option ssid 'superwrt'
        option key ''
        option encryption 'psk2+ccmp'

cat: can't open 'verifica': No such file or directory
cat: can't open 'wifi': No such file or directory
root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: 0
2g: enable
5g: disable
antenna: 4 4
irq number: 74
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: d4e9dd33
iobase1: 34894f17
tx limit: 768
rx limit: 64
qe trigger number: 2014154

root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/sta

mac address:
aid: 2
ampdu: true
amsdu: false
wds: false
ba_hist: disable
ht_cap: 0x1020, ampdu: 02, 06
rx_mask: 0xff, 00, 00, 00
rx_bw: 0, rx_nss: 1
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 00000003eb64

mac address: e
aid: 1
ampdu: true
amsdu: false
wds: false
ba_hist: disable
ht_cap: 0x002c, ampdu: 03, 00
rx_mask: 0xff, 00, 00, 00
rx_bw: 0, rx_nss: 1
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 00000000bd4b

BrainSlayer, 2020-11-23:
500 mbit does not look poor

gimbleguy, 2021-01-17:
Oli I too am disappointed with the openwrt/dd-wrt wifi drivers. The ping latency on wifi is too high for me and yes the wifi is very unstable now :( . The openwrt/dd-wrt wifi developers can't seem to fix this issue or are struggling with it for many years. May be they don't even have a wrt1900acs V2??? For the record AMSDU is still enabled on the wrt1900acs firmware. ugh

In any case, I'll be selling my wrt1900acs V2 and hopefully I'll be getting this in the mail soon
https://www.aliexpress.com/item/4001054259808.html?spm=a2g0s.9042311.0.0.54324c4dzYg64Z

I ordered the high end version of it. Hopefully run OpenWRT on it.

Do yourself a favour Oli sell your WRT1900acs V2, get a router the is properly and fully supported.

Good Luck buddy!

olimpiumarius, 2021-01-18:
@gimbleguy I recently found out that I can somehow change the whole experience with this router...
http://s.go.ro/6z1eoy9u
for settings I was inspired from here:
https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1219835#1219835
https://forum.dd-wrt.com/wiki/index.php/QCA_wireless_settings#TurboQAM_.28QAM256.29_support
https://wiki.dd-wrt.com/wiki/index.php/Atheros/ath_wireless_settings
```
BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 SuperWRT SNAPSHOT, r15520-54bfebdca0
 -----------------------------------------------------
root@swrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option disabled '0'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option country 'RO'
        option cell_density '0'
        option frag '2347'
        option rts '2346'
        option noscan '1'
        option beacon_int '113'
        option channel '48'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option macaddr ''
        option encryption 'psk2'
        option ssid 'superwrt 5Ghz'
        option key ''
        option dtim_period '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option disabled '0'
        option noscan '1'
        option htmode 'HT40'
        list channels '1'
        list channels '6'
        list channels '11'
        list ht_capab 'SHORT-GI-20'
        list ht_capab 'SHORT-GI-40'
        option country 'RO'
        option cell_density '0'
        option frag '2346'
        option rts '2347'
        option channel '11'
        option legacy_rates '1'
        option beacon_int '127'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option macaddr ''
        option encryption 'psk2'
        option ssid 'superwrt'
        option key ''
        option dtim_period '1'

root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: 
2g: enable
5g: disable
antenna: 4 4
irq number: 74
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: c6e3edb5
iobase1: f9691a9a
tx limit: 768
rx limit: 64
qe trigger number: 1369735

root@swrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/sta

mac address:
aid: 1
ampdu: true
amsdu: false
wds: false
ba_hist: disable
ht_cap: 0x1020, ampdu: 02, 06
rx_mask: 0xff, 00, 00, 00
rx_bw: 0, rx_nss: 1
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000080722

mac address:
aid: 2
ampdu: true
amsdu: true
wds: false
ba_hist: disable
amsdu cap: 0x01
ht_cap: 0x106f, ampdu: 02, 00
rx_mask: 0xff, ff, 00, 00
rx_bw: 1, rx_nss: 2
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 00000001a970
```

olimpiumarius, 2021-01-18:
From the tests I did I saw a changed behavior for the better when I played with the advanced settings: Fragmentation Threshold, RTS/CTS Threshold,Beacon Interval, DTIM Interval, Channel
Short Preamble&Disassociate On Low Acknowledgement:they can improve stability and speed in the speed test when disabled
Each change can influence each device in a positive or negative status...

BrainSlayer, 2021-01-18:
> Oli I too am disappointed with the openwrt/dd-wrt wifi drivers. The ping latency on wifi is too high for me and yes the wifi is very unstable now :( . The openwrt/dd-wrt wifi developers can't seem to fix this issue or are struggling with it for many years. May be they don't even have a wrt1900acs V2??? For the record AMSDU is still enabled on the wrt1900acs OPENWRT firmware. ugh
> 
> In any case, I'll be selling my wrt1900acs V2 and hopefully I'll be getting this in the mail soon
> https://www.aliexpress.com/item/4001054259808.html?spm=a2g0s.9042311.0.0.54324c4dzYg64Z
> 
> I ordered the high end version of it. Hopefully run OpenWRT on it.
> 
> Do yourself a favour Oli sell your WRT1900acs V2, get a router that is properly and fully supported by OpenWRT.
> 
> Good Luck buddy!

dont blame dd-wrt or openwrt for the wifi drivers marvell provided (and never finished) major parts of this driver are based on a binary only firmware blob. we have no choice and possibility to enhance them. and even if you ordered the "high end" version of it. this will not change anything. the driver remains the same

BrainSlayer, 2021-01-18:
> From the tests I did I saw a changed behavior for the better when I played with the advanced settings: Fragmentation Threshold, RTS/CTS Threshold,Beacon Interval, DTIM Interval, Channel
> Short Preamble&Disassociate On Low Acknowledgement:they can improve stability and speed in the speed test when disabled
> Each change can influence each device in a positive or negative status...
> If you have other recommendations in using other settings I can test them just leave here directions and I can provide you other information

short preamble has only effect for legacy (g/b) networks. disassoc in low ack may be a positive effects for kickouts. because this may be related to data the driver reports (or wrong reports to the network stack). so disabling this makes absolutelly sense

gimbleguy, 2021-01-20:
Oli you shouldn't need to do anything! It should be just set and go! I gave up on the WRT1900acs wifi. I just plug in a usb ALFA Awus1900 at the back of the WRT1900acs usb port.

Just for comparison, both are stock setting on 5Ghz wifi, no tweaking/tuning or any funny business.

WRT1900acs V2 @ 5Ghz wifi

https://www.speedtest.net/result/10782159874

ALFA Awus1900 USB @ 5Ghz wifi

https://www.speedtest.net/result/10782128413

Check the pings!

the WRT wifi use to be really good, but now days, it has dropped in performance (ping and speed) so badly. :(

olimpiumarius, 2021-01-20:
I'm not saying this router is the best, but neither is it the worst on the wifi side, I'm saying it has some problems!
The problem maybe the settings on the network card....
just follow my test!
http://s.go.ro/san7eivl

gimbleguy, 2021-01-20:
Other than the wifi issues, the WRT1900ACS V2 is an awesome router.

There is no problem on the wifi network cards oli. I run them stock with a Draytek 2926Vac router, and a ZTE router. They work really well stock settings. I use an Intel 7260ac on all my systems stock settings!

As pointed out by Brainslayer, Linksys/Marvell didn't provide the correct information needed for the Openwrt/DD-WRT developers to make a "Proper" wifi drivers for the WRT1900ACS V2.

Anyways, it's up to you Oli, just telling you that you have other options to get the most optimum performance in wifi. Also depends on the size of your house or property, adding a wifi mesh system to your WRT1900ACS V2 router is another alternative.

BrainSlayer, 2021-01-30:
> Other than the wifi issues, the WRT1900ACS V2 is an awesome router.
> 
> There is no problem on the wifi network cards oli. I run them stock with a Draytek 2926Vac router, and a ZTE router. They work really well stock settings. I use an Intel 7260ac on all my systems, stock settings! For the record the stock Linksys Firmware works really well, again stock wifi settings.
> 
> As pointed out by Brainslayer, Linksys/Marvell didn't provide the correct information needed for the Openwrt/DD-WRT developers to make a "Proper" wifi drivers for the WRT1900ACS V2.
> 
> Anyways, it's up to you Oli, just telling you that you have other options to get the most optimum performance in wifi. Also depends on the size of your house or property, adding a wifi mesh system to your WRT1900ACS V2 router is another alternative. This wifi mesh is a growing market.

more important is that marvell sold all wifi chipset development to nxp/philips and there it ends. all development stopped at that time

olimpiumarius, 2021-01-30:
> more important is that marvell sold all wifi chipset development to nxp/philips and there it ends. all development stopped at that time

Too bad because these routers are very solid, but do not have the necessary support!
I will make this public petition here as well...
https://www.change.org/p/belkin-linksys-to-release-the-drivers-for-their-wrt-routers-for-better-open-source-support?utm_content=cl_sharecopy_24769284_en-GB%3A2&recruited_by_id=8c753d60-fdd0-11ea-a226-4163ed15bf1e&utm_source=share_petition&utm_medium=copylink&utm_campaign=psf_combo_share_initial&utm_term=psf_combo_share_abi

Until then, some tests with DDWRT, access the link below...
http://s.go.ro/q8dv8w1p

thekiefs, 2021-02-12:
Is there any settings on OpenWRT to fix this? I disabled legacy networks and short preamble but still getting "Wireless is not associated" error. Anything else?

gimbleguy, 2021-02-12:
> Until then, some tests with DDWRT, access the link below...
> http://s.go.ro/q8dv8w1p

That's what I'm talking about! Gigabit speeds and a very low ping! This just shows the Openwrt wifi drivers have the potential to be better.

BrainSlayer, 2021-02-13:
@gimbleguy since i'm the developer of dd-wrt. maybe some clarification. openwrt has the very same wifi driver as dd-wrt. so this isnt the reason. maybe just the tunings in dd-wrt are different (wifi settings etc.) so best start is to compare the hostapd config files to check whats different. from the driver side i dont expect much difference. except that dd-wrt does support qam256 and openwrt does not for 2.4 ghz (the so called turboqam feature)

shauny1, 2021-03-18:
> For the record the stock Linksys Firmware works really well, again stock wifi settings.

Just to check. Is the Linksys Firmware any different to the mwlwifi firmware? That is, should the performance and stability be the same, or is it to be expected that the stock firmware will be better owing to.... what @BrainSlayer says here:

>.. more important is that marvell sold all wifi chipset development to nxp/philips and there it ends.

So, does this mean the drivers/firmware diverged at this point whilst linksys stock got further improvements? Or did they both die at that point and are *both* no longer updated/developed?

I'm really just trying to work out if I'd get better WiFi experience with the stock image. My main issue other than poor signal strength on both 2GHZ and 5GHZ is that the 5GHZ just disappears completely sometimes and I have to restart the radio. I can live with the sub-par signal strength (it's good enough) but the 5GHZ disappearing randomly all the time is a show stopper.

I have an AP on order so I'll probably use my wrt1900acs as a router only. I like @gimbleguy's idea of using an Alfa USB WiFi. I'd have to build my own img for that presumably? Is there a USB device that will work with the acs by simply installing the right package?

micpro7, 2021-07-01:
> @gimbleguy since i'm the developer of dd-wrt. maybe some clarification. openwrt has the very same wifi driver as dd-wrt. so this isnt the reason. maybe just the tunings in dd-wrt are different (wifi [...]

Hi Brian , I was wondering if you could share info on enabling the 256-QAM feature on a WRT1900ACS, is it as simple as adding option vendor_vht '1' to etc/config/wireless ?

Thanks

olimpiumarius, 2021-08-30:
Greetings to all,

Honestly, the problem I reported in the following configuration is solved!

```
BusyBox v1.33.1 (2021-08-24 17:41:47 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02-SNAPSHOT, r16273-378769b555
 -----------------------------------------------------
root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option country 'RO'
        option cell_density '0'
        option noscan '1'
        option beacon_int '100'
        option frag '2346'
        option rts '2347'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option macaddr ''
        option key '1986wrt27'
        option ssid 'superwrt 5Ghz'
        option dtim_period '2'
        option encryption 'psk2+ccmp'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option htmode 'HT40'
        option legacy_rates '1'
        option country 'RO'
        option cell_density '0'
        option noscan '1'
        option beacon_int '100'
        option channel '9'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option macaddr ''
        option ssid 'superwrt'
        option key '1989wrt27'
        option dtim_period '2'
        option encryption 'psk2'

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: no
[... 134 lines cut ...]
                   total <= 16, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
root@OpenWrt:~# iw phy0 info
```
Watch the video below link with the test!

http://s.go.ro/4fk0f6ej

micdudikov, 2021-09-21:
To me it seems that wrt1900ac v1 with  20.02.0 on 5GHz performs perfectly out of the box , at least for me . 
Even better then stock linksys firmware. however on 2.4Ghz spectrum which you did not test , it is another matter.

On 2.4 GHz i am getting 15-30 MBps on average and as slow as 10 and sometime up to 60 but rarely.
While one stock  linksys firmware i would get 50-70 MBps .

olimpiumarius, 2021-09-21:
> To me it seems that wrt1900ac v1 with 20.02.0 on 5GHz performs perfectly out of the box , at least for me .
> Even better then stock linksys firmware. however on 2.4Ghz spectrum which you did not test [...]

some tests with wrt1900acs v2

[image]
[image]
[image]
[image]

olimpiumarius, 2022-06-30:
Hostname | OpenWrt
-- | --
Model | Linksys WRT1900ACS
Architecture | ARMv7 Processor rev 1 (v7l)
Target Platform | mvebu/cortexa9
Firmware Version | OpenWrt SNAPSHOT r19958-aae3a8a254 / LuCI Master git-22.167.28356-8effea5
Kernel Version | 5.10.125

[image]
[image]
[image]
[image]
[image]
[wifi config.txt](https://github.com/kaloz/mwlwifi/files/9021932/wifi.config.txt)

closed 2023-12-08 by olimpiumarius