kaloz/mwlwifi #393: Embed wifi PCB unable to connect to any AP , openwrt,ddwrt etc.

putout, 2021-03-10:
Issue here: https://security.stackexchange.com/questions/245951/wi-fi-station-wont-connect-to-ap-wpa2-psk

pasted what i can 

Hardware: ELM327 station mode AP DD-wrt, openwrt, n66u Asus, Tried several others Legacy 802.11G
Stations disconnects after a few minutes , tried to reset the WiFi station board - see below)

enter image description here
[image]

OpenWRT AP 

[image]
[image]

Log from OpenWRT

Wed Mar 10 22:24:23 2021 daemon.info hostapd: wlan1-1: STA 00:25:09XXXXXX IEEE 802.11: deauthenticated due to local deauth request
Wed Mar 10 22:24:31 2021 daemon.info hostapd: wlan1-1: STA 00:25:09XXXXXX  IEEE 802.11: associated (aid 1)
Wed Mar 10 22:24:31 2021 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH 00:25:09XXXXXX 
Wed Mar 10 22:24:32 2021 daemon.notice hostapd: wlan1-1: AP-STA-POSSIBLE-PSK-MISMATCH 00:25:09XXXXXX 
Picture of Wifi Module, ( did short GND TO RST) to no avail why powering 3.3v, that stuff u see on the RS232 Header came like that from the factory: 

[image]

eduperez, 2021-03-11:
Do you know which chipset is used by that device? Could it be an ESP8266, perhaps?

BrainSlayer, 2021-03-11:
in dd-wrt is a ldpc workaround option at wireless basic. use this to get around this issue with ESP8266 chipsets

putout, 2021-03-11:
Hi Eduardo 
I added two images if the SOC (is that whats its called?) and PCB which was on the reverse side of the board

I can not find any info on it . I looked up the MAC it came to one company, they didnt reply, I looked up the firmware copyright info at the bottom, they didnt reply.

putout, 2021-03-11:
> in dd-wrt is a ldpc workaround option at wireless basic. use this to get around this issue with ESP8266 chipsets. you see this option only if you switch to n modes (all modes excluding ac). if ac is involved, ldpc is mandatory.

thanks i use the brain slayer ddwrt all the time,  i believe the the firmware/GUI in the wifi client allowed invalid settings (HEX 10 digit + wpa2-psk) 

Is there a way i can tell DDWRT to accept the station/client whatever passkey/ hex its using ?

BrainSlayer, 2021-03-11:
this is called "unencrypted". if its encrypted you always need a valid key, otherwise encryption cannot work by definition since the key is used for authentication and encrypting/decrypting the data. but you can run multiple vap's with different encryption ciphers  and also unencrypted of course

putout, 2021-03-11:
> this is called "unencrypted". if its encrypted you always need a valid key, otherwise encryption cannot work by definition since the key is used for authentication and encrypting/decrypting the data. but you can run multiple vap's with different encryption ciphers and also unencrypted of course

so there is no way that station can connect to a AP...

BrainSlayer, 2021-03-11:
of course it can. just configure the station to use the correct key. and on esp chipset based clients consider to activate the "ldpc workaround" on the basic wifi settings in dd-wrt. . consider that a wpa key can also consist of hex digits which may be simply ansi encoded hex values. then just convert them to plaintext. but its mainly just about setting the correct encryption method and the correct key. dd-wrt does support all encryption methods available on the market.

putout, 2021-03-11:
> of course it can. just configure the station to use the correct key. and on esp chipset based clients consider to activate the "ldpc workaround" on the basic wifi settings in dd-wrt. . consider that [...]

i can not access the station because its trying to connect to WPA2-psk using a 10 digit hex key.   so my OP was to try to reset teh embed chip.................

putout, 2021-03-11:
per your post here https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1218233&sid=a7b55f01ed3bfd5c4dd26d955bc7ef28 
 i dont have LDPC workaround on anywhere of my dd-wrt APs i have tried 4 of them . thank you

putout, 2021-03-11:
if i convert a 10 digit number to ASCII only gives me  6 characters, dd-wrt wpa2psk  field says it needs to be more . 
"Invalid Key, must be between 8 and 63 ASCII characters or 64 hexadecimal digits"