kaloz/mwlwifi #395: McDebian connection rate so high

Chadster766, 2021-05-21:
Why am I seeing 802.11ax connection rates (>500mbps) with McDebian on a Samsung S20?

kubrickfr, 2021-05-22:
Hello
1. This project is dead
2. McDebian also seems dead
3. I don't see a problem with high connection rates, what is your issue here?

I frequently see > Gbps connection rates on 160MHz with this router, and the bandwidth is actually usable, not just theoretical

Chadster766, 2021-05-22:
What's strange is the WRT is configured for 80Mhz and the S20 is showing a connection rate of 780mbps.

```
root@MCDEBIAN:~# vim /etc/hostapd/wlp1s0.conf
interface=wlp1s0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=xxxxxxxxxxxx
ignore_broadcast_ssid=0
hw_mode=a
channel=149
auth_algs=1
wmm_enabled=1
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40]
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMER][SU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155

wpa=2
wpa_passphrase=xxxxxxxx
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
country_code=US
```

ValCher1961, 2021-05-22:
Hi, Chad! 
Maybe it shows the speed of the connection, and the transfer speed will be lower. What does Speedtest show?

Chadster766, 2021-05-22:
Hi @ValCher1961 

It's coming in at 619mbps.

ValCher1961, 2021-05-22:
Wow, that's a very high speed.
I have the same hostapd settings, but the speed is half as fast.

kubrickfr, 2021-05-23:
So what is the problem you need help with?

anazz20, 2021-05-23:
I can't help with your problem, but I just wanted to say hello Chad, it's been a long time!

Chadster766, 2021-05-23:
> So what is the problem you need help with?

This a discussion to find out why the 80Mhz connection rate is at 802.11ax instead of 802.11ac between the WRT3200ACM and the Samsung S20.

There is no problem.

Chadster766, 2021-05-23:
Hi @anazz20,

Thank you :-)

Chadster766, 2021-05-23:
It looks like the WRT is broadcasting at 160Mhz like my Velop WiFi6 system.
[image]

Chadster766, 2021-05-24:
Could the Samsung S20 trigger the mwlwifi driver to start using 160Mhz?

Chadster766, 2021-05-24:
root@MCDEBIAN:~# iw dev wlp1s0 info
Interface wlp1s0
        ifindex 10
        wdev 0x1
        addr 60:38:e0:bd:31:92
        ssid McCorp50
        type AP
        wiphy 0
        channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
        txpower 30.00 dBm

Chadster766, 2021-05-24:
iw phy phy0 info

valid interface combinations:
                 * #{ AP } <= 16, #{ managed } <= 1,
                   total <= 16, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

tapper82, 2021-05-24:
I don't know why people are even talking about this driver? It's Ded as a dodo! If it will not work the way you want it to then get a new router or new AP for WIFI.

jukkapirinen, 2021-05-25:
I don't understand the issue. +500Mbps is normal 80MHz 802.11ac rate. There's multiple streams so total is more than just 433Mbps.

Chadster766, 2021-05-25:
> I don't understand the issue. +500Mbps is normal 80MHz 802.11ac rate. There's multiple streams so total is more than just 433Mbps.

You are right and I had forgotten that. 

It's seemed like something unusual was occurring but I guess not.

closed 2021-05-25 by Chadster766

anazz20, 2021-05-28:
Chadster, I'm glad it's not a problem!
You have any clue with what's going on with the linksys community, or are you off of there? I was in the hospital, then stuck on the couch for a month and a half.

Chadster766, 2021-06-05:
Hi @anazz20,

AFAIK the Official Linksys Community will not be coming back online.

I have started a new forum to hopefully help those Linksys owners 😄 

http://chadster766.com/forums/index.php

anazz20, 2021-06-11:
So all that helpful stuff is gone, great...
I'll check out your forums, good for you for trying to help them! I'll chat with you there.

Chadster766, 2021-06-11:
That would be great.

I have a lot of the info recorded and in my head 😃